emsesp/EMS-ESP32 #635: Add new boiler model - Worcester Bosch Greenstar 30Ri Compact ErP

ChrisHomewood, 2022-09-23:
**Is your feature request related to a problem? Please describe.**

I have a Worcester Bosch Greenstar 30Ri Compact ErP boiler manufactured in 2016 connected using the service jack via BBQKEES interface board. It is recognised as an unknown EMS device, clicking for details shows no recognised entities for the unknown device.

**Describe the solution you'd like**

Please add this boiler to the  list of recognised EMS devices so that it is usable in EMS-ESP. I have collected a log file showing the EMS messages generated by the boiler some of which appear to be standard EMS message types that should be decodable by EMS-ESP

**Describe alternatives you've considered**

Unfortunately I do not have the skills to create a device profile myself so am making a feature request for it to be done for me.

**Additional context**

The boiler is wired as a heat only system boiler with on/off control, the central heating controls do not control the boiler via the EMS bus so in the attached log file you will see it is the only device on the EMS bus and is only posting status messages.

Ideally I would like to control the boiler flow temperature setpoint using EMS-ESP (accepting this may not be possible without an EMS thermostat on the bus, but being able to send the command from EMS-ESP interface would be a good start), as a minimum I would like EMS-ESP to correctly interpret bus messages so they are available to be used in Homeassistant (etc).

The attached log file runs from the moment the boiler is powered, at around line 150 it turns on for a heat demand, the remainder of the file is the boiler starting, running up to full power, modulating down and then turning off.

[EMS log boiler switch start run stop.txt](https://github.com/emsesp/EMS-ESP32/files/9635074/EMS.log.boiler.switch.start.run.stop.txt)

config JSON file:

{
  "System Status": {
    "version": "3.4.1",
    "uptime": "000+00:10:39.158",
    "freemem": 157,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -57,
    "IPv4 address": "192.168.1.173/255.255.255.0",
    "IPv4 gateway": "192.168.1.1",
    "IPv4 nameserver": "192.168.1.101",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tz label": "Europe/London"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 323,
    "bus reads (tx)": 3,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 4,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "CUSTOM",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 16,
    "tx gpio": 17,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Unknown",
      "name": "unknown",
      "device id": "0x08",
      "product id": 154,
      "version": "06.03",
      "entities": 0
    }
  ]
}

proddy, 2022-09-23:
We can add it. Are you on v3.5.0 yet?

ChrisHomewood, 2022-09-23:
I am using the v3.4.2 build, however I can download and flash the development build v3.5.0 onto the ESP32

proddy, 2022-09-24:
It's merged into v3.5.0 (dev branch)

ChrisHomewood, 2022-09-25:
Thanks, just tried flashing with new v3.5.0 bin file (using ESP-Flasher) got a bunch of errors and the board refused to boot. Tried to revert back to the V3.4.1 build that was on there and got a different error. So I have not been able to test. Also any clues how to fix the ESP 32 now it has a file system error??

V3.5.0 goes into a boot loop, serial port log:

[15:09:13]
[15:09:13]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:09:13]configsip: 0, SPIWP:0xee
[15:09:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:09:13]mode:DIO, clock div:2
[15:09:13]load:0x3fff0030,len:1184
[15:09:13]load:0x40078000,len:12784
[15:09:13]load:0x40080400,len:3032
[15:09:13]entry 0x400805e4
[15:09:14]ets Jun  8 2016 00:22:57
[15:09:14]
[15:09:14]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:09:14]configsip: 0, SPIWP:0xee
[15:09:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:09:14]mode:DIO, clock div:2
[15:09:14]load:0x3fff0030,len:1184
[15:09:14]load:0x40078000,len:12784
[15:09:14]load:0x40080400,len:3032
[15:09:14]entry 0x400805e4
[15:09:14]

V3.4.2 now boots but has a file system error (worked fine on this board before trying the v3.5.0 firmware), serial port log:
Showing logs:
[15:11:03]
[15:11:03]┌──────────────────────────────────────┐
[15:11:03]│ EMS-ESP version 3.4.2                │
[15:11:03]│ https://github.com/emsesp/EMS-ESP32  │
[15:11:03]│                                      │
[15:11:03]│ type help to show available commands │
[15:11:03]└──────────────────────────────────────┘
[15:11:03]
[15:11:03]ems-esp:$ ./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
[15:11:03]E (52) esp_littlefs: mount failed,  (-84)
[15:11:03]E (56) esp_littlefs: Failed to initialize LittleFS
[15:11:05][15:11:05]

Full log of flashing v3.4.1 complete with detected board specs (its a generic NodeMCU ESPDevkit V1 from aliexpress):
Using 'COM3' as serial port.
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: EC:62:60:9A:6A:A0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 12.0s
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
[... 81 lines cut ...]
Writing at 0x001b31aa... (95 %)
Writing at 0x001b93ef... (97 %)
Writing at 0x001c2ff3... (98 %)
Writing at 0x001c8c20... (100 %)
Wrote 1816928 bytes (1153714 compressed) at 0x00010000 in 27.0 seconds (effective 537.5 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[15:14:34]
[15:14:34]┌──────────────────────────────────────┐
[15:14:34]│ EMS-ESP version 3.4.1                │
[15:14:34]│ https://github.com/emsesp/EMS-ESP32  │
[15:14:34]│                                      │
[15:14:34]│ type help to show available commands │
[15:14:34]└──────────────────────────────────────┘
[15:14:34]
[15:14:34]ems-esp:$ lib/LITTLEFS/src/lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
[15:14:34]E (49) esp_littlefs: mount failed,  (-84)
[15:14:34]E (52) esp_littlefs: Failed to initialize LittleFS
[15:14:36][15:14:36]

ChrisHomewood, 2022-09-25:
I did try using this upload method as it looks like my problem with uploading may be to do with this open issue, however it still does not work:

https://github.com/emsesp/EMS-ESP32/issues/629#issuecomment-1253275150

proddy, 2022-09-25:
This is a known issue I'm afraid. It's the way the Gateway boards are pre-loaded with the firmware. We're still trying to find the best way forward. What you need to do is wipe the flash and re-load. The easiest way is using https://github.com/MichaelDvP/EMS-ESP_Flasher/releases 

Make sure you download the settings and customizations so you can re-upload those afterwards

ChrisHomewood, 2022-09-25:
Sorry, just checked the serial monitor output and it turns out the upload method I used in my previous message did actually work, I just forgot that the thing creates a wifi hotspot when it doesn't know your wifi details!!!

Will check if the boiler changes work later tonight

ChrisHomewood, 2022-09-25:
Can confirm that the Boiler now appears in EMS-ESP with 49 entities (see attached file for example output), quite a few are not relevant as this is a system boiler (no direct hot water mode). Entities that are relevant appear to be returning data consistent with how the boiler is operating.

The following will be useful for anyone trying to interface with the boiler as part of automations:

- entity "Heating Temperature" is the temperature set via the control knob on the front of the boiler.
- entity "Selected flow temperature" can be written to by EMS-ESP. It has some defined behaviours: when the boiler is running it will accept this written value as the target flow temperature and modulate or cycle accordingly. When the boiler is off then it will default to 5c, writing a value to it is immediately overwritten to 5c, however the next time the boiler cycles on it will use this new value. The value persists over multiple boiler demands for heat. If the value written is higher than the "Heating Temperature" entity then "Selected Flow Temperature" will be set to "Heating Temperature", if "Heating Temperature" is increased using the control knob then "Selected Flow Temperature" will increase upwards until it gets to the value that was written to it. I have not yet checked if values written persist after a power cycle, there is the potential for the boiler to work in unexpected ways if it doesn't because in normal non EMS mode the control knob has master control over "selected flow temperature", but if "selected flow temperature" was written to over EMS with a low value and then the boiler disconnected from EMS control then it would be effectively bricked until a new value could be written.

I have not tried writing to the other entities (e.g. to turn the boiler on or off, change pump mode or burner max power) as I don't need to for my planned purpose, the boiler is used in a conventional system boiler configuration with on/off relay control mode and no EMS thermostat on the bus, all I want to do with the EMS bus is set the flow temperature as an offset of external temperature so that the boiler can be run in a more efficient lower temperature condensing mode when heating demand is lower and then set to a higher temperature for hot water tank heating and during colder weather. - Basically what a more sophisticated opentherm or EMS thermostat would do if it had external temperature compensation (ironically the Nest thermostat I use does have this opentherm mode (set it to "true radiant" mode in the menu) but the boiler and 3 way valve are all wired for on/off relay control and of course the boiler only speaks EMS without an expensive adapter)

[Greenstar 30ri device_entities.xlsx](https://github.com/emsesp/EMS-ESP32/files/9641868/Greenstar.30ri.device_entities.xlsx)
[Greenstar 30ri device_entities.csv](https://github.com/emsesp/EMS-ESP32/files/9641869/Greenstar.30ri.device_entities.csv)

proddy, 2022-10-29:
Thanks Chris for the extra information. Closing this enhancement for now and will copy your notes to the Wiki's FAQ

closed 2022-10-29 by proddy

kingosticks, 2023-12-14:
@ChrisHomewood this is really, really useful information. Sorry to comment here on this old issue but I don't have a better way to ask Chris a couple of questions, if you don't mind.

I have the exact same use-case as you with a WB 30Ri. I feared this wasn't going to be possible without an actual EMS thermostat also connected, so this is great news. Are you still using this system, would you recommend it? Did you find any problems with making changes and frequent boiler power-cycling? I can see how mixing EMS control alongside relay on/off isn't how it's designed to be used.

ChrisHomewood, 2023-12-16:
@kingosticks I have been using the EMS-ESP ever since my original post, it works faultlessly. To answer your questions;

1. I have the EMS-ESP configured in HomeAssistant, I then use an automation which runs every 15 minutes to set the "EMS Boiler Heating Temperature" variable based on the outside temperature that HomeAssistant receives from an outdoor temperature sensor. This performs the function of setting the flow temperature based on the outdoor temperature and works very well as a weather compensation control. I can provide the automation code if you want it but its quite clunky because I also need a separate automation to boost the flow temperature when the hot water tank is being heated.
2. You can use other MQTT automation software to control EMS-ESP, however I already use HomeAssistant so that is what I used. EMS-ESP is automatically recognised by HomeAssistant when it connects to the same MQTT server as HomeAssistant.
3. Careful choice of flow temperature means I actually get less boiler cycling because the boiler can now run for longer at a lower flow temperature. This is the behaviour I wanted because now the boiler is operating in its condensing temperature range which is more efficient (return temperature below 50c). Setting the target flow temperature too low will cause the boiler to start cycling because it can only reduce its output down to 23%, this will be system dependent but in my case a target flow temperature lower than 52c will cause cycling, in the winter its operating in the low 60's
4. Because the only variable I am setting is the Boiler Heating temperature it doesn't matter that the system is controlled by on/off relay. This particular variable is the one that gets changed by the temperature knob on the control panel, in fact if you move the knob it will override whatever is set in EMS-ESP until you send EMS-ESP a new value. Don't change the "Boiler Selected Temperature" variable, that is the one the boiler is currently trying to reach and gets overwritten every time the on/off relay is triggered (it defaults to 5c when the heat demand relay is off).
5. Although using EMS bus alongside the on/off relay isn't the intended use, altering the flow temperature via the EMS bus is supported and as far as the boiler controller is concerned its just someone moving the knob on the front panel. Its also worth bearing in mind this is using the service port on the boiler so it is mimicking what a service technician might do and therefore is a mode the manufacturer has designed for. My boiler actually has the terminals for an external temperature compensation sensor, but no means of setting it up (or specifications for what to connect) so even temperature compensation was designed in.
6. The boiler itself still controls safety critical aspects of its operation. So you cannot command it to heat beyond its maximum set temperature. You can use EMS-ESP to turn the boiler on directly, but any central heating system will be designed to allow the boiler to circulate water on its own without command by the heating controls, so whilst you might not get the desired result (because you don't have control of the zone valves) this won't damage the boiler either.

kingosticks, 2023-12-18:
Brilliant! Thanks for such a detailed reply. I am going to crack on with getting an EMS-ESP board.
Regarding 1, I was planning the same (to start with at least, but I agree with the downside of needing two automations).

Regarding point 5, the servicing instructions have those connections marked as "unused" so I assume they don't do anything and it's just cheaper for them to fit their standard low-voltage terminal block across all products. Either way, a shame.

Regarding 6, with the boiler still wired to the switched-live coming from the normal DHW/CH controller, it would only be able to turn on when the pump and valve were also active. If they are not, then the boiler is always off, no matter what EMS-ESP is trying to do. Maybe I misunderstand my wiring!

But anyway, I think I have more than enough to get started. Thanks again.

ChrisHomewood, 2023-12-22:
If you are getting the EMS-ESP board make sure to get the service jack connector cable, then you don't need to externally power the board, it just plugs in and powers off the boiler, much neater.

The boiler controls the circulating pump not the CH controller. The switched-live to the boiler is just a request line, it tells the boiler to deliver heat, the boiler will then switch the pump on and start heating water, when the switched-live turns off the boiler continues to run the pump until the heat exchanger in the boiler has cooled down. The boiler will also start by itself if the water temperature sensor detects a temperature below 5c, this anti-freezing feature will circulate water through the boiler, the path of that water will be whatever the position of the valves in the system are set to, it is designed to protect the boiler from freezing not the house or central heating pipework. An S or Y plan heating system will always have a flow path for the water but the boiler is unable to control that path because the valve(s) are moved by the CH/DHW controller. It is possible to start the boiler using a command sent from EMS-ESP, however without control of the valve position it is not a useful thing to do.

Remote control of the heating system is best accomplished by replacing the CH/DHW controller with one that allows remote control, EMS-ESP is not for that purpose. Automatic control of flow temperature is the purpose of the EMS bus, a CH/DHW controller that connects into the EMS bus would do this, along with on/off control that replaces the switched live connections (so that all the controls are on a single cable). You can get replacement CH/DHW controllers that work on S or Y plan heating systems and use EMS to talk to the boiler and do temperature compensated setback and other features. I didn't bother because I already had a Nest thermostat which I wanted to continue using and an EMS-ESP device is much cheaper than a EMS CH/DHW controller that works on Y-plan valves. The likely savings from temperature compensation are not enough to justify ripping out all my boiler controls for new ones, and I would lose the occupancy detection feature of the Nest which saves far more money. Plus I plan on getting a heat pump when this boiler dies, and that would work best with new controls suited to a heat pump, so doing anything with EMS is a temporary measure (although hopefully it lasts for a good few more years).

I would not recommend making a DIY boiler controller, everything runs at mains voltage and most micro controller relay boards have poor isolation to the low voltage side. Closest I would get to that is if your existing controls have a low voltage input designed for safe use of external control devices..

kingosticks, 2023-12-22:
I didn't realise that about the pump but makes sense, thanks for that explanation.

I think we are on the same page. Entirely agree regarding DIY controller, that's not interesting to me. I couldn't find much in the way of compatible CH/DHW S-plan controllers that talked EMS other than using a Greenstar Wiring Centre alongside their antiquated controller that fits into the boiler itself. Those are both expensive and while they are functionally interesting, they are a step backwards in usability. I was hoping to continue using my (semi-smart) Honeywell controller for now and just play with the flow temps using EMS-ESP depending on if the controller is doing CH or DHW.

MossyTC, 2024-05-11:
Hi @ChrisHomewood,

Can you provide me with your flow temperature automation code? I'm quite happy to declunk as necessary.

Thanks.

ChrisHomewood, 2024-05-12:
> Hi @ChrisHomewood,
> 
> Can you provide me with your flow temperature automation code? I'm quite happy to declunk as necessary.
> 
> Thanks.

See below homeassistant yaml code. There are two temperature sensors, one mounted outside in the shade and one in the kitchen to monitor indoor temperature. There is also an input boolean which is set to TRUE whenever the boiler is heating hot water according to the hot water timer, a separate automation increases the boiler flow temperature target high enough to heat the hot water tank, whilst that is running this automation will not set the flow temperature.

The values were arrived at by experimenting with the minimum flow temperature that would keep the house warm. Any boiler return temperature below 57c will have the boiler in its more efficient condensing mode, the lower the temperature the more efficient it will be be, but too low and it won't put out enough heat to keep the house warm. This script aims to keep the boiler below that temperature as much as possible without it cycling on and off all the time.

alias: Boiler - Choose Flow Set Temperature
description: Increase boiler flow temperature according to exterior ambient temperature
trigger:
  - platform: time_pattern
    minutes: /15
    seconds: "5"
condition:
  - condition: state
    entity_id: input_boolean.hot_water_running
    state: "off"
    for:
      hours: 0
      minutes: 0
      seconds: 10
action:
  - if:
      - condition: numeric_state
        entity_id: sensor.kitchen_temperature_2
        above: 5
        below: 17.5
    then:
      - device_id: e7d7c506c6d4b16470ff336681967366
        domain: number
        entity_id: number.boiler_heating_temperature
        type: set_value
        value: 70
      - stop: Kitchen cold boost
  - choose:
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            above: 10
            below: 20
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 50
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            above: 6
            below: 10
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 53
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            above: 4
            below: 6
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 55
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            above: -1
            below: 4
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 62
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            below: -1
            above: -3
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 68
      - conditions:
          - condition: numeric_state
            entity_id: sensor.outside_ambient
            below: -3
            above: -20
        sequence:
          - device_id: e7d7c506c6d4b16470ff336681967366
            domain: number
            entity_id: number.boiler_heating_temperature
            type: set_value
            value: 75
mode: single