emsesp/EMS-ESP32 #636: 3.5.0 (Dev) language support changes MQTT/HA entity_ids and breaks existing automations and visualizations

mvjt, 2022-09-23:
**Bug description**
When upgrading to 3.5.0 (dev), entity_names in HA break all automations and visualizations because:
A) ems-esp is added as prefix to all entity names
B) Entity ID (not only friendly name) is localized with the new language support.

**Steps to reproduce**
Install 3.5.0 (dev) and change to a different language. I used Swedish.

**Expected behavior**
Entity_IDs to stay the same so all my HA automations and visualizations don´t break.

**Screenshots**
[image]

MichaelDvP, 2022-09-24:
Please check if [this](https://github.com/MichaelDvP/EMS-ESP32/releases) is a solution. 
If changed the object_id back to device_english-fullname without prefix. 
Also changed the friendly name a bit, The first characte was always capitalized, but this was for tagged names the first char of the tag. e.g. `Selected temperature`, but `Dhw selected temperature`, now it shold show `dhw Selected temperature`.

proddy, 2022-09-24:
checking now....

proddy, 2022-09-24:
I like the 'kb' change you did, thanks! Could you rename it 'kB'? (The internationally recommended unit symbol for the kilobyte is kB.[[1]](https://en.wikipedia.org/wiki/Kilobyte#cite_note-IEC80000-1)" https://en.wikipedia.org/wiki/Kilobyte#cite_note-IEC80000-1)

proddy, 2022-09-24:
I checked in HA, it works great! You can change the entity language and the names change in real-time in the HA dashboards without rebooting or resetting. This can be pushed to dev so @mvjt can test. And it'll need a mention in the Change Log under "Breaking Changes' that the Home Assistant entity name will always be in English (e.g. number.boiler_selected_flow_temperature).

MichaelDvP, 2022-09-24:
One thing, i think you had a reason to add the prefix. In this step i've removed it again to keep the entity compatible to v3.4. If we add it again as basename this is a breaking change, but people can leave base empty and have compatible entity names.

MichaelDvP, 2022-09-24:
> Could you rename it 'kB'?

Sure, but i've used 1024 divider (better readable results), should i better use `KB` (legacy) or `KiB`(IEC)?

proddy, 2022-09-24:
KB then.

Also I guess this also fixes https://github.com/emsesp/EMS-ESP32/issues/634 ?

proddy, 2022-09-24:
> One thing, i think you had a reason to add the prefix. In this step i've removed it again to keep the entity compatible to v3.4. If we add it again as basename this is a breaking change, but people can leave base empty and have compatible entity names.

yes this is no longer needed. In earlier versions the prefix was hard coded to 'ems-esp' which messed up MQTT when multiple Gateways are used. Then I changed it to use the hostname as the prefix. And we learned that that broke when it had a dot (.) in it. Now I think we've solved all those issues!

MichaelDvP, 2022-09-24:
In actual main [there is the optional emsesp_ prefix](https://github.com/emsesp/EMS-ESP32/blob/main/src/mqtt.cpp#L1116-L1117) commented out. For multiple gateways now the ' object_id'  payload will be identical, but the the config-topic is different basename with `homeassistant/sensor/basename/name/config`. Does  this work or do HA need a different payload object_id for second gateway?

proddy, 2022-09-24:
yes that comment was added by another user (can't remember who). but it's not needed. For multiple gateways the best practice would be to use a unique base name on each

mvjt, 2022-09-24:
Just tested dev. The ems_esp prefix is still there so it is a breaking change for all HA entities. Is it not better to add a system setting "Add host name as prefix" which can be enabled by those few with multiple EMS-GWs instead of breaking for everyone else?

MichaelDvP, 2022-09-24:
It's not merged yet, try dev from [my repo](https://github.com/MichaelDvP/EMS-ESP32/releases), or wait for merge.

proddy, 2022-09-24:
it's merged now. Thanks Michael.

mvjt, 2022-09-25:
Thanks a lot nice work! Works fine.

closed 2022-09-25 by mvjt