emsesp/EMS-ESP32 #637: WebUI firmware upload - improve error handling and add MD5 check

proddy, 2022-09-24:
From https://github.com/emsesp/EMS-ESP32/issues/629#issuecomment-1256927475_

proddy, 2022-09-24:
For the MD5 check I was talking to Rick last year about this. See https://github.com/rjwats/esp8266-react/issues/260#issuecomment-996046241. The steps would be:
1. generate a sha512 signature for the artefact and sent as an optional header on the file upload, and verified if supplied.
2. in the upload loop call `md_update()` for each segment of the payload, and at the end verifying before calling the final `Update.end(true)`

something you'd like to look at Michael?

MichaelDvP, 2022-10-14:
I've made a [md5-branch](https://github.com/MichaelDvP/EMS-ESP32/tree/md5) to test, also [firmwares and md5-files](https://github.com/MichaelDvP/EMS-ESP32/releases) for all boards. Please check.

proddy, 2022-10-15:
I'll check today

MichaelDvP, 2022-10-15:
I'll change the way of md5 sending to web, in first appoach i added a hppt_get, but it's better to use direct respond of the post.

proddy, 2022-10-15:
ok. it would good to validate the md5 on the server side too (at least the # chars). Another thing I noticed is the dropzone drag&drop box shows read when dragging in an md5 file, so the .md5 needs to be added to the list of accepted extensions.

Can I help anywhere?

MichaelDvP, 2022-10-15:
Yes it's already done, md5-file is accepted, post to ems-esp, checked for 32 char (only length), stored as 32 char, send back to web and shown on web. On bin-upload the stored md5 is checked.
I've updated the md5 branch and merged the latest dev. Now building the bin/md5 files.

proddy, 2022-10-16:
do we also need to change the WebUI 'version check' to pick up the right platform and board?

MichaelDvP, 2022-10-16:
Yes, simple way to direct to the `html_url`, so the right bin and md5 can be downloaded, or enhance to right chip version and add an extra link to the md5?

proddy, 2022-10-22:
yes!

closed 2022-11-11 by proddy