emsesp/EMS-ESP32 #638: Unexpected behaviour when adjusting room temp value FR100

RonaldRegister, 2022-09-24:
FR100
v3.4.2 and previously 3.4.0b8
Gateway E32
Bosch Proline

I was trying to explain my issue but it got too complicated. So here’s version 2, with an image which should enhance my poor way of explaining myself. 

When I adjust ‘hc1 selected room temperature’ (1) from within the web-interface it also changes the value ‘hc heat temperature’ (2). This overwrites the mode temperature, and stays this way. Shouldn’t this value only change when I adjust (2)?

[image]

From my experience with other systems, selected room temperature is the result of a desire, be it a program or a temp setpoint, which later is overruled by the resuming of a program or mode change.

When I walk to the thermostat and adjust it by turning the temperature knob -like a caveman!- it only changes (1) like it should, and it doesn’t touch (2).

I've searched existing and closed issues, but couldn't find my exact problem, although some with overlap, not sure! 

If needed I can supply more info, logs, and I would love to contribute by testing or helping otherwise.

Thank you
Ronald

MichaelDvP, 2022-09-24:
Yes this a known issue, in #400 we have tried a lot to find out how it can be changed, but no luck. 
Seem this thermostat does not support changing the temporary setpoint via bus. Sorry.

RonaldRegister, 2022-09-25:
Ah that's a shame. This was the main reason for me to start using the E32 with EMS-ESP. 

I've got a spare FR10, which I am willing to sacrifice, I will post my findings in #400 maybe I can do some hardware modification to it. Found a 4 pin header on the pcb, who knows.  
 
Thanks all of you working on this project, and for the fast response!

RonaldRegister, 2022-12-11:
Well this did not get me very far. The 4 pin header has a ground and 3 times 3.3Volt. Unable to see if there is data on these lines and I'll have to borough a oscilloscope to see which line is which and at which baudrate or protocol it is (if it even is what I think it is)

Furthermore I tried to add this thermostat to my system but EMS-ESP, myself, and the boiler/thermastat manual combined couldn't figure out how to configure it all to work together. The manual is very unclear and written in a incredibly complicated manner. 

If there is anyone who can send me in the right directions, please let me know. Or maybe tag people who are more experienced with these systems?

[image]

rickdeck, 2022-12-18:
Your exploration of the 4-pin connector would be quite interesting for my purpose actually, did you have any luck in that direction?

I have a FR50 (which has the same connector) but found my boiler is not using EMS but the legacy analog 1-2-4 protocol, and so I'm wondering if I could maybe at least read information directly from the Thermostat (or at best even control it) using this interface.
Unfortunately I have only this one device so I'm hesitant to risk breaking it, especially during winter season ;)

RonaldRegister, 2022-12-19:
Haha, no don't do this in winter time on a device which is not a spare. What boiler do you have?

I'm waiting on someone to borrow a scope. But even then I'm not expecting to be capable enough to read/write or gain control via this interface (if it even is).

I am tagging @norberts1 (if thats ok?) I've seen his knowledge regarding this bus on other topics, and perhaps he can point us in the right direction. Any help would be very much appreciated.

Btw: if needed I'm willing to completely sacrifice this device (FR10) if it's needed.

rickdeck, 2022-12-19:
I have an old Junkers boiler, from the time before EMS bus. It uses the analog 1-2-4 interface, which is a 3-wire connection to the thermostat. FR50 and FR100 support it (but FR10 doesn't).
I found out about this detail while looking into adding the boiler to home assistant, so without EMS bus I now wonder if I could maybe read/control the thermostat directly instead.

I am considering to buy a cheap USB logic analyzer to hook up and trace the internal connector to the PC (they're just ~15 Euro), but so far my suspicion was that it's just a connector to reflash the firmware...

closed 2023-02-19 by proddy