emsesp/EMS-ESP32 #641: Unknown source of current room temp when no room sensor connected

moinmoin-sh, 2022-09-24:
**Bug description**
EMS_ESP 3.4.2

My problem is, that my Junkers installation doesn't have a room temperatur sensor connected. I don't know where the shown value is collected from. May be an open input to my Junkers? 
In an earlier version it was possible to configure what should be shown (faked) for the current room themperature if no sensor available. I used the "selected room temperature" . In the recent version I can't find this anymore and I can't remember were it was ... 
For me it would also be possible to feed in an other room sensor value that I have available via mqtt.

**Steps to reproduce**

**Expected behavior**
being able to override the auto generated value or replace with own sensor to make the thermostat looking consistant

**Screenshots**
[image]

**Device information**
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.07",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26",
      "handlers ignored": "0x05 0x22 0x01 0x03 0x04"
    },
    {
      "type": "Thermostat",
      "name": "FW100",
      "device id": "0x10",
      "product id": 105,
      "version": "13.07",
      "entities": 11,
      "handlers received": "0x06",
      "handlers fetched": "0x016F 0x0165 0x23",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0166 0x0167 0x0168 0xBB 0x0123",
      "handlers ignored": "0x35 0x1A 0xBE 0x01D3 0xC3 0x011E 0x01C9"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.07",
      "entities": 0
    }
  ]

**Additional context**
*Add any other context about the problem here.*

proddy, 2022-09-24:
Thanks for reporting. We need to figure out where that 33.5 value for currtemp is coming from.

MichaelDvP, 2022-09-25:
@moinmoin-sh Please make a log (all) which contains the telegrams 0x165 and 0x16F and post it here. 
Afaik all thermostats have a integrated temperature sensor. If mounted on the boiler a special contact on the backside of the thermostat deactivates this sensor, there is no setting to configure it. (On Norberrts1 list there is a setting in offset 9, maybe wrth a try)
 Is your thermostat mounted on boiler?
The master thermostats with more heatingcircuits allows to connect a room thermostat as remote. For RC35 we can emulate a roomthermostat RC20 and send only the current roomtemperature. But FW100 does not have this option, only FW400 and it's not yet implemented. So there is no way to feed a external roomtemperature to the themostat.

moinmoin-sh, 2022-09-25:
@MichaelDvP thanks for taking care of this.

Yes, the FW100 is mounted at the boiler.
I think I can't follow your note on Norberts1 list and what to do there?

I didn't intend to feed an external room temperature into the thermostat itselft, just thought about to feed in a value for the thermostat display on the lovelace card in HA, so cosmetic only.

The attached log should have the telegrams you're asking for.

[log .txt](https://github.com/emsesp/EMS-ESP32/files/9640227/log.txt)

MichaelDvP, 2022-09-25:
For the Junkers/HT3 NorbertS1 have a [very good list](https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf) of the telegram meanings. 
For the settings in telegram 0x165 (Norberts Number 357) there is a setting for temperature sensor (Byte 15/offset9),
```
Betriebsart Raumtemperaturfühler für Heizkreis x
- 0=nicht definiert
- 1=Externer Temperaturfühler
- 2=Interner Temperaturfühler
- 3=Temperatur im Sparmodus
```  
This is set to 3 in your thermostat. I don't know what this setting really means, maybe this indicates that the thermostat is mounted on boiller. @NorbertS1, could you help? 
In telegram 0x16F (Norbers Number 367) the roomtemperature is reported as 32.6 °C (0x0146), i think from internal sensor from inside the boiler. We need to find a setting that indicates that this temperature is not valid for roomtemperature.
Or we can try to set `external temperaturesensor` and see if roomtemperature is than reported as `not attached/open` (0x8000/0x7F00).

norberts1, 2022-09-25:
Years ago a used the FW100 and the FB10/FB100 (remote-controller for hc1).
In this configuration the `Betriebsart Raumtemperaturfühler für Heizkreis x` -value was set to
  `- 1=Externer Temperaturfühler`
My heatersystem-configuration was:
1. FW100 mounted at the wall near by the boiler, but not into the boiler
2. FB10/FB100 mounted in my livingroom

my (0x0165) data:
`01 01 04 19 30 32 07 01 01 01 00 33 14 02 02 15 28 2b 03 01`

Regarding the other values I don't know for which configuration they are really used for:
- 2=Interner Temperaturfühler
- 3=Temperatur im Sparmodus

But anyway, only what you can do is to mount the FW100 (FB10) at least:
1. into the boiler --> then it is perhaps: -->> `3=Temperatur im Sparmodus` or `2=Interner Temperaturfühler`
2. At the wall in any room near by the boiler or in any room used for heating-circuit hc1 : -->> resulting value unknown for me.
3. or FW100 mounted at the wall and FB10/100 mounted in the hc1-room:  -->> `1=Externer Temperaturfühler`
  (my configuration).
4. or FW100 mounted in the boiler and FB10/100 mounted in the hc1-room: -->> resulting value unknown for me.

But if there isn't any remote-thermostat in the system (MsgID:357_1_0 := 0) 
and the `Betriebsart Raumtemperaturfühler für Heizkreis x` is not `1`
and the measured temperature (MsgID:367_4_0) is higher then perhaps 28.0 degree or more 
then the temperature-value come from boiler-internal measurement.

Attached you can find an analyzed plot from my older heater-system using FW100 and FB10:

[image]

moinmoin-sh, 2022-09-26:
After some reading of the manuals I would guess the following:
the settings for  
`Betriebsart Raumtemperaturfühler für Heizkreis x`
- 2=Interner Temperaturfühler
- 3=Temperatur im Sparmodus
are probably related to the `internal `sensor in the FW100 and `external `is FB10/100 and that setting defines which/how the connected roomtemperature sensors shall be used for room dependent calculations 
[image]

MichaelDvP, 2022-09-26:
Yes, this makes sense. I've checked other logs and all thermostats are set to 1. And i think this thermostat is often integrated in boiler. The telegram 16F reports 32.6°C for roomtemp and 6226.0°C for remote temperature, (means not connected) and the lower temp is shown as roomtemp. Your thermostat seems to be configured wrong, you should set Raumeinfluss to 1. If you want to change via ems-esp use terminal (or use mqtt) and type:
`call send "0B 10 FF 09 00 65  01"`
(Mqtt: topic `system`, payload `{"cmd":"send","data":"0B 10 FF 09 00 65 01"}`)
Or use thermostat, configure FB10 remote, then Raumeinfluss to 1, then set back remote to off. 
If it does not work you can simple change back: `call send "0B 10 FF 09 00 65  03"`

moinmoin-sh, 2022-09-26:
1) I tried to make the change at the thermostat (FW100) which doesn't work because when I configure FB10 it complains that the device ID is not found on the bus and doesn't let me change related arameter.
2) I tried the telnet console. Had to add "system" to the call : 
[image]
But that seems to don't change anything. The attached log should show it 

Did I do anything wrong?

[log-0926.txt](https://github.com/emsesp/EMS-ESP32/files/9645484/log-0926.txt)

MichaelDvP, 2022-09-26:
Good that you have logged the change. 
```
001+17:12:20.722 DEBUG 13144: [telegram] Sending write Tx [#242], telegram: 8B 10 FF 09 00 65 01
001+17:12:20.722 TRACE 13145: [emsesp] Me(0x0B) -> Thermostat(0x10), JunkersSet(0x0165), data: 01 (offset 9)
001+17:12:20.722 DEBUG 13146: [emsesp] Last Tx write successful
```
The write is ok and thermostat  have acknowladged the change. Next we check the the change by reading the telegram
```
001+17:12:20.846 DEBUG 13149: [telegram] Sending read Tx [#243], telegram: 8B 90 FF 00 19 00 65
001+17:12:20.864 TRACE 13150: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x0165), length: 0x19
001+17:12:20.941 DEBUG 13151: [emsesp] Last Tx read successful
001+17:12:20.941 INFO 13152: [emsesp] Thermostat(0x10) -> Me(0x0B), JunkersSet(0x0165), data: 01 00 02 19 41 50 03 02 00 03 FF 28 06 01 04 14 24 2A 02 01 00
```
The setting is again 03, the thermostat has set it back and broadcasts the change next:
```
001+17:12:20.999 TRACE 13154: [emsesp] Thermostat(0x10) -> All(0x00), JunkersSet(0x0165), data: 03 (offset 9)
```
So this setting can not be changed. But i don't think anymore that we can use this as a sure sign of boiler-mounted. 

Another option: Go to customizations and disable mqtt publish of roomtemp, then the climate entitiy should switch to seltemp.

moinmoin-sh, 2022-09-26:
I followed your suggestion. The `currtemp ` is removed then from the `thermostat_data ` payload.
The climate card then shows no temperature where the room-tempretaure used to be.
[image]

proddy, 2022-09-26:
perfect!

closed 2022-10-01 by proddy