emsesp/EMS-ESP32 #645: Add aux heater limits

proddy, 2022-09-25:
### Discussed in https://github.com/emsesp/EMS-ESP32/discussions/644

<div type='discussions-op-text'>

<sup>Originally posted by **Cali1205** September 25, 2022</sup>
While working on optimizing the power consuption of my heat pump together with my PV system I thought it is a good idea to be able to control the power to the aux heater.
My idea is to use the limit setting of the aux heater.
[image]

I think it would be very helpful to have these setting available.

The register seems to be 0x492.
Description | Offset
-- | --
aux heater limit with compressor active | 2
aux heater limit heating | 3
aux heater limit hot water | 4

And the values are:
Reg Value | Value
-- | --
0 | 0kW
1 | 2kW
2 | 3kW
3 | 4kW
4 | 6kW
5 | 9kW

</div>

MichaelDvP, 2022-09-28:
I've added this to PR #650, needs to be checked by a user with this heatpump.

proddy, 2022-10-24:
thanks. closing for now.

closed 2022-10-24 by proddy