emsesp/EMS-ESP32 #646: Optimize flash with ESP32

proddy, 2022-09-25:
This is another thing that's been on my mind for a while. A lot of the code was written based on the ESP8266 and ported over to the ESP32, including how we place and extract data from the precious flash memory. On the ESP32 any `const char` is automatically placed in PROGMEM so we can optimize in a few ways
- remove the redundant F()
- remove the redundant PROGMEM and uint32 alignment
- don't use  `__FlashStringHelper`. This is just an empty dummy class to differentiate between SRAM and Flash. Replace with const char *'s

For starters, I changed `device_library.h` which saves a 1kb of flash. I'll use the branch https://github.com/emsesp/EMS-ESP32/tree/flash_optimization for testing

proddy, 2022-09-28:
removed flash. It makes the code cleaner and easier to read too, also I expect some performance gains. The nice thing is that it's saved almost 8KB in Flash. Now I need to test it!

proddy, 2022-09-28:
in my tests performance is good, heap is the same and the application sized is smaller. so this is looking good...

MichaelDvP, 2022-09-30:
I've tried your latest commit and get crashes exactly after one minute. Disabled different services and it seems to be in fetch of ems values (ems disconneced->no crash). If you can not reproduce, i have to search in RC35, MM10 and SM10 fetching.

proddy, 2022-09-30:
yes, known issue. it's in the MQTT code and I know which line to fix. Haven't been home for a while but will resume this weekend.

MichaelDvP, 2022-10-01:
It was the empty fullname for climate entity, that has to be added: [This line](https://github.com/emsesp/EMS-ESP32/blob/90d21445884f1c8c576deb6dcaa70f5a0f898dde/src/locale_common.h#L267) to:
`MAKE_PSTR_LIST(climate, "HA climate config creation", "")`

proddy, 2022-10-01:
yes, i checked the fix in yesterday

proddy, 2022-10-01:
i swapped out std::string to const char * in ha_publish and saved another 760b in flash

proddy, 2022-10-01:
the flash_optimization is up to date with the latest 'dev'. It'll keep it running for a few days and if its stable will merge back into dev.

proddy, 2022-10-08:
will merge into dev

closed 2022-10-08 by proddy