emsesp/EMS-ESP32 #647: Invalid value for number.thermostat_hc1_selected_room_temperature: 30 (range 0.0 - 29.0)

nagyrobi, 2022-09-25:
**Bug description**
Home assistant spams the following message on every thermostat mqtt message when `thermostat_hc1_selected_room_temperature` goes to 30:

`[homeassistant.components.mqtt.number] Invalid value for number.thermostat_hc1_selected_room_temperature: 30 (range 0.0 - 29.0)`

This is on a Generic HT3, with a controller detected as thermostat FW200 (not having an actual EMS thermostat in the system), in a Junkers Ceraclass Excellence ZSC 28-3 MFA E 23.

Observed that this value is tied to climate.thermostat_hc1's `temperature` value. The thermostat entry has `max_temp: 30`, and when it's set to that, `number.thermostat_hc1_selected_room_temperature` tries to go there by itself to the same value.

Similar to https://github.com/emsesp/EMS-ESP32/issues/469, only that the top value is capped the code. Perhaps it would suffice not to hardcode the `max` to `30`, but instead make it equal with thermostat's `max_temp`.

ems-esp32: v3.4.2
home assistant: 2022.9.6

MichaelDvP, 2022-09-25:
@proddy Seems that bosch changes the limits with every device. Should we adapt the limit dynamically for ems-reported values out of range. e.g. add to `emsdevice::generate values`
```
                if ((output_target == OUTPUT_TARGET::MQTT) && (dv.min != 0 || dv.max != 0)) {
                    if (json[name].is<float>() || json[name].is<int>()) {
                        int v = json[name];
                        if (fahrenheit) {
                            v = (v - (32 * (fahrenheit - 1))) / 1.8; // reset to °C
                        }
                        if (v < dv.min) {
                            dv.min = v;
                            dv.remove_state(DeviceValueState::DV_HA_CONFIG_CREATED);
                        } else if (v > dv.max) {
                            dv.max = v;
                            dv.remove_state(DeviceValueState::DV_HA_CONFIG_CREATED);
                        }
                    }
                }
``` 
This should hopefully end all HA complains about min/max values.

MichaelDvP, 2022-09-28:
I thought about user defined min/max limits without wasting a lot of RAM.  I think i've found a solution storing user defined values in the customization in a single string only if set e.g. `00hc2/daytemp|Tagestemp>5<30` if Name and both limits set . The input looks like this:
[image]
and does not show the default min/max, because they are overwritten in RAM on change and ems-esp does not hold the initial values in ram.
When resetting customizations to default or clearing a min or max value the last set value stays until reboot.
But i think customizations are rarly used and mainly set once, so it's not a big disadvantage. 
@proddy Do you think it's usefull? Should i make a pr?

proddy, 2022-09-28:
hehe...I was lying in bed and thinking almost the same solution as you. I had (x,y) but your > and < is much cooler

do the PR!

proddy, 2022-10-01:
Did some testing on the min/max. it works well. two things I'll look into
 - it should recreate the MQTT if the min/max changes, so these get shown immediately in HA without needing a restart
 - the climate control on HA (climate/ems-esp/thermostat_hc1) needs its min_temp and max_temp also adjusting. This is the control dial you see in HA

proddy, 2022-10-02:
I'm working on a PR to address those two points above

MichaelDvP, 2022-10-02:
Take a look at [this approch](https://github.com/MichaelDvP/EMS-ESP32/commit/c414354708058cd6bffb76e90173f92c1dd12351). Not tested yet, For HA checks i need a running mosquitto and reconfigure the ems-connected gateway to it.

proddy, 2022-10-02:
> Take a look at [this approach](https://github.com/MichaelDvP/EMS-ESP32/commit/c414354708058cd6bffb76e90173f92c1dd12351). Not tested yet, For HA checks i need a running mosquitto and reconfigure the ems-connected gateway to it.

that's similar what I had. Let's go with that. I can test locally if you want. I do most of my tests using the standalone/make run way and controlling the output of the MQTT topics being created. But I can also test on HA with it's embedded mosquitto - it's just a container I need to spin up

closed 2022-10-04 by proddy