emsesp/EMS-ESP32 #648: Add GPIO with scheduler

jfilaa, 2022-09-27:
What add possibility to drive some GPIOs, directly by MQTT and by scheduler. For example user defined sheduler plans: to drive circulation pump, start at 6:00 And run 10 min, start at 12:00 and run 15 min, etc.

proddy, 2022-09-27:
it's possible, not an easy feature to add and very dependent on having the correct time set. These kind of actions are much better using an external automation tool like Home Assistant. Have you considered that?

jfilaa, 2022-09-27:
Thank You for responce. Current solution is driven by smart-plug Shelly (sonoff). Shelly is based on ESP32 and have time from NTP. At task of switching circulation pump for heat wather is not needed high presition time... But what reduce only to drive GPIO by MQTT and sheduler will be running at Home Assistant as You offerd?

proddy, 2022-09-27:
On a an EMS-ESP board you need to be very careful what current you're driving to the GPIOs, specially for switches as you could damage both circuits. In your situation, I would just use Home Assistant as the scheduler.

bbqkees, 2022-09-28:
EMS-ESP can already set GPIO. You can use MQTT to set them.
However, as Proddy said you have to be careful what you connect to it as you can seriously damage the boiler.
I explicitly do not support any use of GPIO on EMS Gateways.

I combined some guidelines here: https://bbqkees-electronics.nl/wiki/gateway/using-gateway-gpio.html

proddy, 2022-09-28:
ok, to close this then, no timer or scheduler in EMS-ESP. Use other software (like HA) and interact with EMS-ESP via MQTT or API

jfilaa, 2022-09-29:
Thank you. Can You please write example how to set GPIO by MQTT?

jfilaa, 2022-10-01:
Here Is example how to add GPIO. [image]

proddy, 2022-10-01:
use the command `setvalue` under `analogsensor` to change the GPIO with the Console, API or MQTT

https://emsesp.github.io/docs/#/Command?id=fetching-analog-sensor-information

jfilaa, 2022-10-01:
Thank you much for support.

proddy, 2022-10-29:
and #701 will support the scheduler.

closed 2022-10-29 by proddy