emsesp/EMS-ESP32 #649: Settings not saved

andriuskr, 2022-09-28:
Flashed new ESP32 with newest firmware, connected to boiler. Everything seems fine and working, until reboot. After reboot ems-esp reverst to factory defaults. 
If i try to save configuration, i get empty file

proddy, 2022-09-28:
which version?

andriuskr, 2022-09-28:
3.4.2

MichaelDvP, 2022-09-28:
Same issue as #614? Try dev-build (3.5.0b)  or flash 3.4.1 and update via upload. What flashtool have you used?

oliverm01, 2022-09-30:
Hello! I had exactly the same issue here after flashing 3.4.2 and had my network settings as a suspect first, because the esp32 didn't connect to the network after set and reboot. But the settings file had not been saved.

An firmware upload to 3.5.0b from a local storage (download first) worked and fixed this issue here finally.

MichaelDvP, 2022-09-30:
> Unable to upload 3.5.0b firmware. Getting internal server error with different browsers

See #629

bbqkees, 2022-10-05:
Indeed with 3.4.2 on a new Gateway settings are not persistent. Got quite a number of emails about that already.
I test each Gateway before shipment, but I did not test making changes to settings as i ship them out in default state.

For those who ordered a Gateway in the past few weeks and it has 3.4.2 on it, you can load 3.5.0 via the web interface and then it will work. Other option is to load 3.4.1 via the web and then upload 3.4.2.
If loading 3.5.0 fails with error 500/server error please use the USB flashing method with the new EMS-ESP flash tool to flash the Gateway. You have to open the device to access the USB connector. See:
https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html#uploading-the-firmware-via-ems-esp-flasher-flashtool

Wuehli, 2022-10-05:
> Got quite a number of emails about that already

+1 😄

andriuskr, 2022-10-11:
With usb flashing of 3.5.0b6 sucess, but in the end logs shows erros. And ESP does not boot.

[image]

MichaelDvP, 2022-10-11:
What flasher do you have used? The flashers for tasmota and esphome  use a to small partition for ems-esp v.3.5, 
Use esptool from commandline or try [this](https://github.com/MichaelDvP/EMS-ESP_Flasher/releases)

bbqkees, 2022-10-11:
Which module are you trying to flash?

MichaelDvP, 2022-10-11:
- in v3.4.3 the settings not saved error is fixed.
- have you tried to powercycle the module after flashing?

andriuskr, 2022-10-11:
> Which module are you trying to flash?
[image]
[image]

ESP 32 D1 MINI.

proddy, 2022-10-14:
is it working now?

andriuskr, 2022-10-27:
> is it working now?

No, settings still not saved

MichaelDvP, 2022-10-28:
I've checked the 3.4.3 and can confirm it is NOT working. The fix is placed at wrong place after trying to read from fs and abort, it should be first in writeToFS before _fs->open().

@proddy Please make a fixed 3.4.4. 

@andriuskr up/downgrade (usb or OTA or web upload) to <= 3.4.1 or 3.5.0bx and configure network, then you can change to any version.

MichaelDvP, 2022-10-29:
@andriuskr you can use the new release 3.4.4. https://github.com/emsesp/EMS-ESP32/releases/tag/v3.4.4

andriuskr, 2022-10-30:
> 
Now it is working as expected. Thanks