emsesp/EMS-ESP32 #652: Can't decode payload

nagyrobi, 2022-09-30:
**Bug description**
I get this, occasionally in Home Assistant log:
```
2022-09-30 11:51:19.229 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttClimate._prepare_subscribe_topics.<locals>.handle_current_temperature_received at 0x7f9c640a5b40>>)
2022-09-30 11:51:19.231 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttClimate._prepare_subscribe_topics.<locals>.handle_target_temperature_received at 0x7f9c654ca560>>)
2022-09-30 11:51:19.232 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttClimate._prepare_subscribe_topics.<locals>.handle_current_mode_received at 0x7f9c5cc92cb0>>)
2022-09-30 11:51:19.232 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttNumber._prepare_subscribe_topics.<locals>.message_received at 0x7f9c53e17520>>)
2022-09-30 11:51:19.233 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f9c53e140d0>>)
2022-09-30 11:51:19.234 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttSelect._prepare_subscribe_topics.<locals>.message_received at 0x7f9c3fb22320>>)
2022-09-30 11:51:19.235 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f9c53e148b0>>)
2022-09-30 11:51:19.236 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttNumber._prepare_subscribe_topics.<locals>.message_received at 0x7f9c54d3bbe0>>)
2022-09-30 11:51:19.236 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttNumber._prepare_subscribe_topics.<locals>.message_received at 0x7f9c53e26680>>)
2022-09-30 11:51:19.237 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttNumber._prepare_subscribe_topics.<locals>.message_received at 0x7f9c54326440>>)
2022-09-30 11:51:19.238 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttSelect._prepare_subscribe_topics.<locals>.message_received at 0x7f9c5438b010>>)
2022-09-30 11:51:19.239 WARNING (MainThread) [homeassistant.components.mqtt.client] Can't decode payload b'{"seltemp":29,"currtemp":29.8,"mode":"heat","modetype":"heat","heattemp":0\x9e\x02\x00\x0cems-esp/info{"event":"reconnect","version":"3.4.2","connection":' on ems-esp/thermostat_data_hc1 with encoding utf-8 (for <Job HassJobType.Callback <function MqttSelect._prepare_subscribe_topics.<locals>.message_received at 0x7f9c53d93a30>>)

```

**Steps to reproduce**
Nothing special. About once a day this happens by itself.

**Expected behavior**
No errors in the log.

**Screenshots**
N/A

**Device information**
This is on a Generic HT3, with a controller detected as thermostat FW200 (not having an actual EMS thermostat in the system), in a Junkers Ceraclass Excellence ZSC 28-3 MFA E 23.
ems-esp32: v3.4.2
home assistant: 2022.9.6

**Additional context**

proddy, 2022-09-30:
please attach the debug info. The error is obviously in the topic name '0\x9e\x02\x00\x0cems-esp/' so interested what you set your hostname and mqtt base too

nagyrobi, 2022-09-30:
```json
{
  "System Status": {
    "version": "3.4.2",
    "uptime": "001+18:41:16.321",
    "freemem": 127,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -71,
    "IPv4 address": "redacted",
    "IPv4 gateway": "redacted",
    "IPv4 nameserver": "redacted",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "never",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "connected",
    "enabled": true,
    "server": "redacted",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": false,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 118673,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "gazkazan",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 0,
    "publish time thermostat": 0,
    "publish time solar": 0,
    "publish time mixer": 0,
    "publish time other": 0,
    "publish time sensor": 0,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 485022,
    "bus reads (tx)": 53491,
    "bus writes (tx)": 118599,
    "bus incomplete telegrams": 0,
    "bus reads failed": 84,
    "bus writes failed": 1314,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "MH-ET",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": false,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "15.02",
      "entities": 59,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "FW200",
      "device id": "0x10",
      "product id": 106,
      "version": "12.10",
      "entities": 11,
      "handlers received": "0x06 0x0166 0x23",
      "handlers fetched": "0x016F 0x0165",
      "handlers pending": "0xA3 0xA2 0x12 0x0170 0x0171 0x0172 0x0167 0x0168 0xBB 0x0123",
      "handlers ignored": "0xBE 0x01D3 0x35 0x1A"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "15.02",
      "entities": 0
    }
  ]
}
```

proddy, 2022-09-30:
settings look ok

nagyrobi, 2022-09-30:
I also get ocassionally:
```
2022-09-30 16:08:24.361 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'mode' when rendering '{{value_json.mode}}'
2022-09-30 16:08:24.366 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_mode: '' (valid options: ['nofrost', 'eco', 'heat', 'auto'])
2022-09-30 16:08:24.367 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heattemp' when rendering '{{value_json.heattemp}}'
2022-09-30 16:08:24.368 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"seltemp":29,"currtemp":30.4,"modetype":"heat"}' is not a Number
2022-09-30 16:08:24.369 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ecotemp' when rendering '{{value_json.ecotemp}}'
2022-09-30 16:08:24.370 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"seltemp":29,"currtemp":30.4,"modetype":"heat"}' is not a Number
2022-09-30 16:08:24.370 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'nofrosttemp' when rendering '{{value_json.nofrosttemp}}'
2022-09-30 16:08:24.371 WARNING (MainThread) [homeassistant.components.mqtt.number] Payload '{"seltemp":29,"currtemp":30.4,"modetype":"heat"}' is not a Number
2022-09-30 16:08:24.372 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'control' when rendering '{{value_json.control}}'
2022-09-30 16:08:24.373 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_control_device: '' (valid options: ['off', 'fb10', 'fb100'])
2022-09-30 16:08:24.374 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'program' when rendering '{{value_json.program}}'
2022-09-30 16:08:24.375 ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.thermostat_hc1_program: '' (valid options: ['prog_a', 'prog_b', 'prog_c', 'prog_d', 'prog_e', 'prog_f'])
```

MichaelDvP, 2022-10-01:
Have you [tried this](https://emsesp.github.io/docs/#/Troubleshooting?id=home-assistant)?