emsesp/EMS-ESP32 #654: Temperature artifacts when controlling current flow temperature via MQTT

Lemonn, 2022-10-01:
I use an external program to control the current flow temperature of the boiler via MQTT. EMS-ESP32 [v3.4.2] and my program both use MQTT QoS 0. Sometimes when the temperature changes from a period of x°C to y°C, I get artifacts where the temp jumps back for a short period to the previous value. Resulting in unwanted burner starts/stops. I've verified that my program is not sending these values thru monitoring the corresponding MQTT topic. I've also ruled out warmwater or warmwater refreshs as cause.

<details>
  <summary>Log Data</summary>

  ```
 map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:24.843140+2:00 ems-esp telegram - - - 001+00:00:27.827 T 110145: Rx: 88 00 34 00 28 01 B5 80 00 A1 00 00 01 00 01 DE 04 02 59 41 00 FB facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:25 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:24.861726+2:00 ems-esp mqtt - - - 001+00:00:27.842 D 110146: Publishing topic ems-esp/boiler_data_ww (#62562, retain=0, retry=1, size=484, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:25 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:24.870833+2:00 ems-esp command - - - 001+00:00:27.853 D 110147: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:25 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:24.960940+2:00 ems-esp mqtt - - - 001+00:00:27.943 D 110148: Publishing topic ems-esp/boiler_data (#62563, retain=0, retry=1, size=613, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:25 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:24.969362+2:00 ems-esp command - - - 001+00:00:27.952 D 110149: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:25.600644+2:00 ems-esp telegram - - - 001+00:00:28.585 T 110150: Rx: 88 00 18 00 00 01 CF 2D 00 00 02 24 C0 01 B5 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 0C facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:25.672156+2:00 ems-esp mqtt - - - 001+00:00:28.650 D 110151: Publishing topic ems-esp/boiler_data_ww (#62564, retain=0, retry=1, size=484, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:25.687583+2:00 ems-esp command - - - 001+00:00:28.671 D 110152: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:25.774859+2:00 ems-esp mqtt - - - 001+00:00:28.756 D 110153: Publishing topic ems-esp/boiler_data (#62565, retain=0, retry=1, size=613, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:25.780846+2:00 ems-esp command - - - 001+00:00:28.764 D 110154: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:26.253333+2:00 ems-esp telegram - - - 001+00:00:29.238 D 110155: Sending write Tx [#1], telegram: 8B 08 1A 00 00 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:26.268633+2:00 ems-esp telegram - - - 001+00:00:29.238 T 110156: Rx: 8B 08 1A 00 00 50 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:26 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:26.273388+2:00 ems-esp emsesp - - - 001+00:00:29.238 D 110157: No telegram type handler found for ID 0x1A (src 0x0B) facility:0 hostname:192.168.178.178 priority:7 severity:7 tag: timestamp:2022-10-01 19:29:27 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:26.284156+2:00 ems-esp emsesp - - - 001+00:00:29.269 D 110158: Last Tx write successful facility:0 hostname:192.168.178.178 priority:7 severity:7 tag: timestamp:2022-10-01 19:29:27 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:26.288168+2:00 ems-esp telegram - - - 001+00:00:29.269 D 110159: Sending post validate read, type ID 0x18 to dest 0x08 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:27 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.311986+2:00 ems-esp telegram - - - 001+00:00:31.297 D 110160: Sending read Tx [#2], telegram: 8B 88 18 00 1B facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:28 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.327296+2:00 ems-esp telegram - - - 001+00:00:31.297 T 110161: Rx: 8B 88 18 00 1B 27 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:28 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.381978+2:00 ems-esp emsesp - - - 001+00:00:31.367 D 110162: Last Tx read successful facility:0 hostname:192.168.178.178 priority:7 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.385871+2:00 ems-esp telegram - - - 001+00:00:31.367 T 110163: Rx: 88 0B 18 00 00 01 CB 2D 00 00 02 24 C0 01 B4 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 FE facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.400003+2:00 ems-esp mqtt - - - 001+00:00:31.382 D 110164: Publishing topic ems-esp/boiler_data_ww (#62566, retain=0, retry=1, size=484, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.418493+2:00 ems-esp command - - - 001+00:00:31.401 D 110165: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.467955+2:00 ems-esp telegram - - - 001+00:00:31.452 D 110166: Sending write Tx [#3], telegram: 8B 08 1A 00 00 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.483466+2:00 ems-esp telegram - - - 001+00:00:31.468 T 110167: Rx: 8B 08 1A 00 00 50 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.484350+2:00 ems-esp emsesp - - - 001+00:00:31.468 D 110168: No telegram type handler found for ID 0x1A (src 0x0B) facility:0 hostname:192.168.178.178 priority:7 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.498793+2:00 ems-esp emsesp - - - 001+00:00:31.483 D 110169: Last Tx write successful facility:0 hostname:192.168.178.178 priority:7 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:28.502674+2:00 ems-esp telegram - - - 001+00:00:31.483 D 110170: Sending post validate read, type ID 0x18 to dest 0x08 facility:14 hostname:192.168.178.178 priority:119 severity:7 tag: timestamp:2022-10-01 19:29:29 +0200 CEST tls_peer:]
[... cut ...]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:44.953420+2:00 ems-esp command - - - 001+00:00:47.936 D 110264: Calling command 'boiler/selflowtemp' (selected flow temperature) with value 0 facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:45 +0200 CEST tls_peer:]
map[client:192.168.178.178:60575 content:1 2022-10-01T19:29:45.010183+2:00 ems-esp mqtt - - - 001+00:00:47.992 D 110265: Publishing topic ems-esp/boiler_data (#62589, retain=0, retry=1, size=613, pid=1) facility:3 hostname:192.168.178.178 priority:31 severity:7 tag: timestamp:2022-10-01 19:29:45 +0200 CEST tls_peer:]
  ```
</details>

[image]
[image]
[image]

MichaelDvP, 2022-10-02:
Looks like flooding mqtt queue with commands. Reduce command rate.

Lemonn, 2022-10-02:
I understand, but shouldn't the queue keep the message order? In case messages arrive faster than they could be transmitted, wouldn't it be better to throw out the oldest, untransmitted message? Especially if it is of the same type as the newer one?

At the moment, I respond to each currentFlowTemp whit a new SelFlowTemp. I could reduce the amout of updates. But then the question of the appropriate speed arises, as there is no indication of a full buffer.

MichaelDvP, 2022-10-02:
Mqtt queue is 300 messages, if overflowd the oldes is removed and indicated as publish fail. 
Periodical read requests are set to end of queue, but commands should executed fast, and set to top of queue. If you send to much your first command gets shifted in the queue and is executed later. I think this happens to you (see `Tx [#178]` in your log).
@proddy Should we add queue count to the mqtt-status (warn if >2)?  
[image]
(this also shows reconnect count)

Lemonn, 2022-10-02:
I would suggest grouping commands in the buffer whenever they target the same value. For example, if the setFlowTemp is sent multiple times, and therefore is more than once in the send buffer. I would replace the older value whit the newer one. Because when controlling the boiler older values are no longer relevant as soon as a new update arrives.

> Periodical read requests are set to end of queue, but commands should executed fast, and set to top of queue. If you send to much your first command gets shifted in the queue and is executed later.

To circumvent this problem it would be necessary to put commands on top of the queue, whilst also respecting the order of commands. Therefore a command should only be placed on top of the queue if the first element is not already a command. If the first element is a command, the following command should be placed behind the newest command in the queue.

proddy, 2022-10-02:
@MichaelDvP showing the MQTT queue is a nice touch to add to the webUI

@Lemonn we have a few thousand users of EMS-ESP out there today and this is the first issue we've seen related to message congestion, so I'm not too keen on refactoring the queueing logic without first understanding your use case. What are you trying to do and why? Note MQTT is not the best way to send commands to EMS-ESP. The preferred way is via the API as it's instantaneous

MichaelDvP, 2022-10-02:
Sorry, i've mixed up mqtt queue and ems tx queue. This is the ESM tx-queue that is flooded (and produces 2 overflows).
EMS timing is set by master and we can only write if master is polling the emsesp client.
I think the easiest way to solve is not sending so much useless messages.

Lemonn, 2022-10-02:
> What are you trying to do and why?

Instead of cycling the boiler when it reaches a certain predetermined CurrentFlowTemp. I use a dynamic flow temperture, to force the burner on for a certain amount of time. This timeslot is determined by outside factors such as the current outside temperature. When I would only use a high selFlowTemp, the burner would not modulate. So to force the burner into modulation, I dynamically set the selFlowTemp always a few degrees under the CurrentFlowTemp. This results in the boiler gradually reducing the power whiteout cycling. 

[image]

In the attached screenshot the regulation could be seen. First, the Bruner is forced on by setting the SelFlowTemp a few degrees above the CurrentFlowTemp. Then the SelFlowTemp is dynamically kept a few degrees below the CurrentFlowTemp to force the boiler into modulation down (See CurBurnPower). if The heating cycle is finished, the boiler is forced off by setting the flow temp to 0.

> we have a few thousand users of EMS-ESP out there today and this is the first issue we've seen related to message congestion, so I'm not too keen on refactoring the queueing logic without first understanding your use case.

EMS-ESP32 is a great project. It allows to controll boilers far better than the orginal hardware. You don't have to do this youerself. I am willing to contribute fixing this problem.

> Note MQTT is not the best way to send commands to EMS-ESP. The preferred way is via the API as it's instantaneous

Okay, what causes the delaying of MQTT commands?

Lemonn, 2022-10-02:
> Sorry, i've mixed up mqtt queue and ems tx queue. This is the ESM tx-queue that is flooded (and produces 2 overflows). EMS timing is set by master and we can only write if master is polling the emsesp client. I think the easiest way to solve is not sending so much useless messages.

I think it's up to the owner of the queue, to decide if a message is useless or not. In my program, I react to updates whit a new regulation value. I can't determine if the previous one is already transmitted over EMS or not. Therefore in my option, it's best when the queue decides if a value inside the queue is no longer useful to send. This is the case if the queue contains a new value of the same type. 

There is also reordering which happens if an overflow occurs. Here it would be better to throw out the oldest/newest not-yet transmitted message instead.

MichaelDvP, 2022-10-02:
Heating is a very slow thing, the bus polls are between 100ms and 5 sec, we can only send a single command if we are polled. Also other systems on the bus react slow, If you send the command every 30 sec there will be no difference in heating performance. It is useless to send commands faster then the heating system can react.

Lemonn, 2022-10-02:
I'm not sending the commands in specific intervals. The interval in which updates happen is determined by EMS-ESP:

1. EMS-ESP sends a new value over MQTT
2. I do some computation on the new values
3. I send an updated SelFlowTemp
4. I wait for a new update from EMS-ESP and continue whit step 1.

MichaelDvP, 2022-10-02:
Set the mqtt boiler interval to 10 sec or more. The auto interval publishes boiler topic if one value in the payload changes.

Lemonn, 2022-10-02:
I will implement a check if a relevant value changes and only then send an update. Although this might solve the direct problem, I still think the underlying queue mechanics should be changed. It unnecessarily pollutes the EMS-Bus, if old values are sent, even though there is already an updated value available. 

Have I understood the ordering of commands in the queue correctly?

Starting whit a queue:
[OldestCommand]
When a new command is added:
[NewerCommand, OldestCommand]
When a new command is added:
[NewestCommand, NewerCommand, OldestCommand]

Shouldn't it be?

[OldestCommand]
When a new command is added:
[OldestCommand, NewerCommand]
When a new command is added:
[ OldestCommand, NewerCommand, NewestCommand]

proddy, 2022-10-02:
> Is there a reason why newer commands are always added in front of older commands?

I think there probably was when we wrote it 2 years ago, but I forgot!

MichaelDvP, 2022-10-02:
The queue is not to buffer incomming commands. It is intended to buffer the scheduled fetches, triggert every minute. These are set to end of queue. Writes are set to top of queue and in normal operation (except yours), this command is send out long before the next command is received.  We can change it and use a proirity queue for writes and a second for reads, but if someone sends to much commands, the reads will never be executed and values are not updated. There is no useful way to solve it, only send less command in a time. **The limit is not by emsesp, it is the ems-bus.** And normal operation is to have maximal one write in tx-queue. 
BTW: send_raw is only called by  system/send command, all others use send_read_request, send_write_request.

Lemonn, 2022-10-02:
If someone uses it like me as a response to updates it would be impossible to flood the reads. Because no more reads = no more updates. You're right flooding is then possible but this should be no problem as a program that simply floods the EMS-Bus has no real use case. I don't know if the prioritization of the write operation is necessary. If not one queue should be enough. Simply put the writes at the back of the queue. If the queue is full delete the first element. If I see it correctly this only requires one change.
https://github.com/emsesp/EMS-ESP32/blob/fc057d18c99e9ded6fe34f0bd3aa46480d9d0879/src/emsesp.cpp#L1148

```cpp
// sends write request
void EMSESP::send_write_request(const uint16_t type_id,
                                const uint8_t  dest,
                                const uint8_t  offset,
                                uint8_t *      message_data,
                                const uint8_t  message_length,
                                const uint16_t validate_typeid) {
    txservice_.add(Telegram::Operation::TX_WRITE, dest, type_id, offset, message_data, message_length, validate_typeid, false);
}
```

> The queue is not to buffer incomming commands. It is intended to buffer the scheduled fetches, triggert every minute.

If the queue is only for reads another option would be to only allow one write operation which is placed first in the queue. If another write operation is received then the older not yet transited one gets replaced.

MichaelDvP, 2022-10-02:
> If someone uses it like me as a response to updates it would be impossible to flood the reads.

Your use is unique. 
> Because no more reads = no more updates.

No, most telegrams are broadcasted by the devices, we active read only a  few settings not broadcasted and this once a minute. You should read how the bus works.
> If the queue is only for reads another option would be to only allow one write operation which is placed first in the queue. If another write operation is received then the older not yet transited one gets replaced.

For some settings there is a need to send more telegrams to change different offsets not in a row, resulting in 2 writes.
For these it is not important which is first. 

Also it is useful to send different commands to change settings, like changing the roomthermostat: needs change in control telegram and roominfluence. This works well. I see no reason to change this only because you are not willing to change  publish rate. All changes i thought of and you have suggested have more disadvantages then current implementation. 
I think there is no better solution, because the limit is set by the bus, not ems-esp, but afair i've mentioned this before...

I'm still open to implement a better solution.

Lemonn, 2022-10-02:
> Your use is unique.

I don't doubt that my use case is something special :)

> This works well. I see no reason to change this only because you are not willing to change publish rate.
 
I'm not saying that this is in any way a relevant feature. So I'm okay if it never gets changed. I clearly understand that the bus has a certain throughput. I already changed parts of my program to limit the output. What I want to change is the behavior in case someone sends too many messages. Because I think the queue reordering is quite confusing behavior.

### Suggestion

If the values read once a minute, could this not be in order whit the write commands?

If this is the case, both command types (read and write), could be added at the back of the queue. If the queue reached its capacity, additional new elements should be dropped, until there is space again. Whenever a write is thrown away, a corresponding log message should be displayed. To prevent the starvation of the scheduled read command, I would suggest prioritizing it by allowing it to replace the last element in a full queue.

I would combine this whit the following change, which prevents old values from being sent if there is already an updated version:

```cpp
void TxService::add(const uint8_t  operation,
                    const uint8_t  dest,
                    const uint16_t type_id,
                    const uint8_t  offset,
                    uint8_t *      message_data,
                    const uint8_t  message_length,
                    const uint16_t validateid,
                    const bool     front) {
    auto telegram = std::make_shared<Telegram>(operation, ems_bus_id(), dest, type_id, offset, message_data, message_length);

#ifdef EMSESP_DEBUG
    LOG_DEBUG(F("[DEBUG] New Tx [#%d] telegram, length %d"), tx_telegram_id_, message_length);
#endif

    //#################################
    auto latest_telegram = tx_telegrams_.front().telegram_;
    if (latest_telegram->type_id == telegram->type_id && latest_telegram->offset == telegram->offset) {
    #ifdef EMSESP_DEBUG
            LOG_DEBUG(F("[DEBUG] Replaced telegram [#%d] whit updated version, because it sets the same value"), tx_telegrams_.front().id_, tx_telegram_id_++);
    #endif
        tx_telegrams_.pop_front();
        tx_telegrams_.emplace_front(tx_telegram_id_, std::move(telegram), false, validateid);
        return ;
    }
    //#################################

    // if the queue is full, make room but removing the last one
    if (tx_telegrams_.size() >= MAX_TX_TELEGRAMS) {
        tx_telegrams_.pop_front();
    }

    if (front) {
        tx_telegrams_.emplace_front(tx_telegram_id_++, std::move(telegram), false, validateid); // add to front of queue
    } else {
        tx_telegrams_.emplace_back(tx_telegram_id_++, std::move(telegram), false, validateid); // add to back of queue
    }
    if (validateid != 0) {
        EMSESP::wait_validate(validateid);
    }

}
```

I'm willing to implement this myself and provide a pull-request.

closed 2022-10-24 by Lemonn