emsesp/EMS-ESP32 #657: Actual room temperature should be condisder the offset

tefracky, 2022-10-02:
**Is your feature request related to a problem? Please describe.**
It is possible to set a possible offset to the measured room temperature of the thermostat. The actual temperature (hc1/currtemp) does not consider this offset and therefore, also the resulting MQTT climate entity shows the wrong room temperature.

**Describe the solution you'd like**
Either the current room temperature should directly consider this offset or an additional offset-corrected temperature should be provided.

**Describe alternatives you've considered**
In Home Assistant, custom climates can be defined. Hence, the room temperature can be calculaded through a template. It works, but a direct implementation in the EMS-system would be better.

MichaelDvP, 2022-10-02:
This is a function of the thermostat. There is an setting to correct the internal sensor and the resut on temperature is used by the thermostat for control and is send as roomtemp to ems-esp. If this setting is not implenmented for your thermostat, please determine it by using settings on the thermostat and log the change, then we can implement it.

tefracky, 2022-10-02:
Ahhh, you are right! I used "Raumtemperatur Offset" (room temperature offset?), but I have to use "Fühlerabgleich" (sensor calibration?)... But unfortunaly, I did not found "sensor calibration" in the EMS-ESP. Maybe I'm just incompetent, but does the function exist somewhere or is it not implemented?
I use a RC300 thermnostat from Bosch/Buderus.
Thanks a lot!

MichaelDvP, 2022-10-03:
We have it for RC30, RC35 and RC100H, for RC300 it is unknown. Maybe the same as RC100H, check telegram 0x273.

tefracky, 2022-10-03:
I made some tests and it seems to be 0x0240.

This is the output:

Offset = 0.0       Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: 00 (offset 7)
Offset = 0.1       Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: 01 (offset 7)
Offset = 0.2       Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: 02 (offset 7)
Offset = 0.3       Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: 03 (offset 7)
Offset = -0.1      Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: FF (offset 7)
Offset = -0.2      Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: FE (offset 7)
Offset = -0.3      Thermostat (0x10) -B-> All(0x00), RC300Settings(0x0240), data: FD (offset 7)

MichaelDvP, 2022-10-03:
Thanks, added to [my repo](https://github.com/MichaelDvP/EMS-ESP32/releases), please test, then i make a pr.
In telegram 0x240 we have now offsets 7,8,9,10 for sensorcalibration, damping, bulding and minExternalTemp. Other offsets are unknown, i think something device specific like std.display, language, brightness, etc.

tefracky, 2022-10-03:
In principle, the sensor calibration works, but there are two issues:
1. Not the exact value but a lot of more digits are provided (e.g. 0.30000001192092896 instead of 0.3). This is in principle not a big deal, but unpleasant.
2. Negative values cannot set at all! They are read as negative, if I set it on my thermostat, but they are converted to positive values, if I try to set them over the ESP.

MichaelDvP, 2022-10-03:
Is this on web page, or api, or mqtt? If only webpage try to refresh page to clear the browser cache.

tefracky, 2022-10-03:
It was a browser problem. Overall, it works well, so you can make the PR. Thanks!

MichaelDvP, 2022-10-04:
I can reproduce the long numbers: It's a strange thing with input fields, we have seen this before (#124, #329). Seems there was a change in libs or browser js engine. Edge/Chrome do not accept Intl.NumberFormat, I added a rounding now. But for some values the inputfield sets very strange numbers, A value of +/-1.4 is always set to +/-1.299... and rounded to 1.3. Setting via api/mqtt is all ok, it's only the browser input field.

tefracky, 2022-10-04:
Yes, I can confirm this behavior exactly as you described. I also think, rounding will be the best way to get rid of this.

closed 2022-10-24 by proddy