emsesp/EMS-ESP32 #658: Dev update - internal server error

yazend, 2022-10-02:
I treyd upgrade to new dev Version, get an internal Server error. Don't know how to fix.
[image]

If logfiles needed let me know.
Alex

MichaelDvP, 2022-10-03:
See #629

yazend, 2022-10-03:
MichaelDvP thanks for the information,
I bought the Wifi only about 1 or 1½ months ago, instructions on how to flash it via USB would be helpful, I don't want to open the device, before I have a safe way to update it to the current DEV version.

bbqkees, 2022-10-03:
@MichaelDvP I updated my wiki just now so it points to your tool.
@yazend no worries the update is pretty easy. Just unscrew the lid from the Gateway and connect the internal USB port to a pc.
There is nothing you can damage, in case a firmware update goes wrong you can still try again or load an older version of the firmware.

proddy, 2022-10-03:
@MichaelDvP do you want to upload your EMS-ESP Flasher to a new repo in https://github.com/emsesp ?

yazend, 2022-10-03:
I will try it later on... let you guys know result. 
Many thanks for your quick support.
- FYI, you guys do a realy good job - thanks

yazend, 2022-10-03:
OK, I got it working 
but new DEV version still have this issue.
[image]

can I follow this instuction and use new EMS-ESP-3_5_0b4-ESP32.bin as well?

[image]

proddy, 2022-10-03:
Yes

MichaelDvP, 2022-10-04:
@yazend 
- If you have flashed the v3.5 once via usb, you can web-upload or OTA any version of ems-esp. 
- If you have the new bbQKees gateway with soldered esp32 you can use `partitions.esp32.16MB.bin`  in usb flashing,

yazend, 2022-10-05:
Hey guys, many many thanks for your support, I am realy preciate.