emsesp/EMS-ESP32 #659: MQTT several values missing

Lemonn, 2022-10-03:
I tested [Development Build v3.5.0b4](https://github.com/emsesp/EMS-ESP32/releases/tag/latest). There seems to be some problem regarding the publication of several values over MQTT. Some values like selflowtemp, get published as expected. But others like heatingactive, brungas, tapwateractive, and ww3wayvalve, are not published. The web interface is not affected, as the values are displayed correctly.

```json
  "MQTT": {
    "enabled": true,
    "host": "192.168.178.79",
    "port": 1883,
    "base": "ems-esp",
    "username": "admin",
    "password": "admin",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": true,
    "max_topic_length": 128,
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "ha_enabled": false,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
```

Lemonn, 2022-10-03:
Nothing is blocked on the customizsation page, i just keept the default settings. Testet both versions whit the same settings. 

Here is the output from MQTT whit version 3.4.2:

```json
{"heatingactive":"off","tapwateractive":"off","selflowtemp":0,"selburnpow":45,"heatingpumpmod":0,"curflowtemp":29.2,"exhausttemp":24.8,"burngas":"off","burngas2":"off","heatingpump":"off","fanwork":"on","ignwork":"off","oilpreheat":"off","curburnpow":0,"burnstarts":586942,"burnworkmin":421040,"burn2workmin":0,"heatworkmin":298638,"servicecodenumber":0,"maintenancemessage":"H00","maintenance":"off","maintenancetime":0}
```

```json
{"wwtapactivated":"on","wwsettemp":40,"wwtype":"flow","wwcirc":"off","wwcurtemp":26.6,"wwcurflow":0,"wwstoragetemp1":26.6,"wwonetime":"off","wwdisinfecting":"off","wwcharging":"off","wwrecharging":"off","wwtempok":"off","wwactive":"off","ww3wayvalve":"on","wwstarts":153986,"wwworkm":122402}
```

proddy, 2022-10-03:
I can't reproduce. We would need to look into your particular setup

Lemonn, 2022-10-03:
Okay, that's strange. Let me know what you need and what I should do to find the reason. 

What I know so far is: 
- The corresponding value is updated in the Web-interface
- No values missing in version 3.4.2
- Used the same configuration as in 3.4.2
- No hints from the logs
```
2022-10-03 17:43:07.772   DEBUG 2293:   [mqtt]       Publishing topic ems-esp/boiler_data (#838, retain=0, retry=1, size=439, pid=1)

2022-10-03 17:43:07.875   DEBUG 2294:   [mqtt]       Publishing topic ems-esp/boiler_data_ww (#839, retain=0, retry=1, size=272, pid=1)

2022-10-03 17:43:07.976   DEBUG 2295:   [mqtt]       Publishing topic ems-esp/boiler_data (#840, retain=0, retry=1, size=441, pid=1)

2022-10-03 17:43:09.694   DEBUG 2296:   [mqtt]       Publishing topic ems-esp/heartbeat (#841, retain=0, retry=1, size=29
```

<details>
  <summary>Settings</summary>
  
  ```json
{
  "type": "settings",
  "System": {
    "version": "3.5.0b4"
  },
  "Network": {
    "ssid": "xxxxxxxx",
    "password": "xxxxxxxx",
    "hostname": "ems-esp",
    "static_ip_config": false,
    "enableIPv6": true,
    "bandwidth20": false,
    "tx_power": 20,
    "nosleep": false,
    "enableMDNS": true
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "password": "ems-esp-neo",
    "channel": 1,
    "ssid_hidden": false,
    "max_clients": 4,
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "192.168.178.79",
    "port": 1883,
    "base": "ems-esp",
    "username": "admin",
    "password": "admin",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": true,
    "max_topic_length": 128,
    "publish_time_boiler": 0,
    "publish_time_thermostat": 0,
    "publish_time_solar": 0,
    "publish_time_mixer": 0,
    "publish_time_other": 0,
    "publish_time_sensor": 0,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "ha_enabled": false,
    "nested_format": 1,
    "discovery_prefix": "homeassistant",
    "publish_single": false,
    "publish_single2cmd": false,
    "send_response": false
  },
  "NTP": {
    "enabled": true,
    "server": "time.google.com",
    "tz_label": "Europe/Amsterdam",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": true,
    "port": 8266,
    "password": "ems-esp-neo"
  },
  "Security": {
    "jwt_secret": "ems-esp-neo",
    "users": [
      {
        "username": "admin",
        "password": "admin",
        "admin": true
      },
      {
        "username": "guest",
        "password": "guest",
        "admin": false
      }
    ]
  },
  "Settings": {
    "tx_mode": 2,
    "ems_bus_id": 11,
    "syslog_enabled": true,
    "syslog_level": 9,
    "trace_raw": true,
    "syslog_mark_interval": 0,
    "syslog_host": "192.168.178.79",
    "syslog_port": 9342,
    "shower_timer": false,
    "shower_alert": false,
    "shower_alert_coldshot": 10,
    "shower_alert_trigger": 7,
    "rx_gpio": 5,
    "tx_gpio": 17,
    "dallas_gpio": 4,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "low_clock": false,
    "telnet_enabled": true,
    "notoken_api": false,
    "readonly_mode": false,
    "analog_enabled": true,
    "pbutton_gpio": 33,
    "solar_maxflow": 30,
    "board_profile": "E32",
    "fahrenheit": false,
    "bool_format": 1,
    "bool_dashboard": 1,
    "enum_format": 1,
    "weblog_level": 9,
    "weblog_buffer": 100,
    "weblog_compact": false,
    "phy_type": 1,
    "eth_power": 16,
    "eth_phy_addr": 1,
    "eth_clock_mode": 0
  }
}
  ```

</details>

<details>
  <summary>Customizations</summary>
  
  ```json
{
  "type": "customizations",
  "Customizations": {
    "sensors": [],
    "analogs": [],
    "masked_entities": []
  }
}
  ```

</details>

proddy, 2022-10-03:
Can you see the values from the Telnet Console using `show`? The code that generates the MQTT/Console/API data is different than the one that does the Web. If you're interested the function is `EMSdevice::generate_values()` .

CheeseE, 2022-10-05:
btw, same here with version v3.5.0b2, only bool values are not published.

Lemonn, 2022-10-05:
@proddy I'm willing to investigate this issue further. I'm able to build the application but unable to successfully run or debug it on hardware. It would be nice, if you could provide a short description, of how to test/debug the application. 

Is there some method to simulate the EMS-BUS? Because to test on the physical BUS, I always have to stop my heating controller and flash the new version, which is quite time-consuming.

Lemonn, 2022-10-05:
Thank you for the fast fix, great work. Tested https://github.com/emsesp/EMS-ESP32/issues/659, works as expected.

closed 2022-10-05 by Lemonn

proddy, 2022-10-05:
@Lemonn you asked how to test the code with simulating the EMS bus. It's all described in https://emsesp.github.io/docs/#/Coding?id=standalone-testing 

I also had a version using cmake and realtime debugging in VSCode on a Windows environment but stopped that as I was the only one using it. I use WSL/Ubuntu now and gcc

JasonKaisersmith, 2022-11-03:
I have the same problem, with this version.  When Can I expect a fix?

This is my first device, so not the best starting experience!

proddy, 2022-11-03:
> I have the same problem, with this version. When Can I expect a fix?
> 
> This is my first device, so not the best starting experience!

you could uninstall this free software application that took 4 painstaking years to write and develop your own solution if you're dissapointed?  :)

JasonKaisersmith, 2022-11-03:
There is no need to be rude.  I simply asked when a fix is expected.

I did not disparage or insult you or the software, but simply stated that as this is my first experience that it was not the best start.  

I was amazed and delighted to discover that there was such a solution as this, and I am appreciative of that fact and all the effort that goes into it.  Working in the SW industry myself, then I do know and understand how much effort it really is hidden behind what we see here.

Don't forget that your SW is just one part of a complex eco system that one must learn.  As such, the learning curve is rather high as there is not just this SW, but also the physical device that connects to the EMS-bus, Rpi, Grafana, Influx, Mosquitto MQTT, telegraf, docker, Portainer, etc. on-top of trying to understand the terminology of the EMS devices themselves and intrepret what is meant by the abbreviations.
The majority of these technologies I have little or no experience of using previously, (I have spent 8+ hours just "fighting" with the telegraf config, and longer with grafana, trying to get it all work).   So then I'm sure that you can understand when one expresses a little bit of disappointment when one discovers the reason for something not working is a bug that doesn't yet have a release date.

proddy, 2022-11-03:
> There is no need to be rude. I simply asked when a fix is expected.

If you scroll up you'll see this issue was fixed 29 days ago. If you're still seeing similar issues please open a new GitHub issue and make sure you attached the relevant information (https://emsesp.github.io/docs/#/Support)

commit 974510b2 (2022-10-05): fix #659, mqtt bool values
commit 9f5a5108 (2022-10-05): Merge pull request #662 from MichaelDvP/dev

fix #659, mqtt bool values