emsesp/EMS-ESP32 #666: Junker FW500 wird nicht erkannt (device id: 0x10, product id: 116, version: 17.02)

ollikomo, 2022-10-06:
**s your feature request related to a problem? Please describe.**
Junker Regler FW500 ist noch nicht bekannt

Describe the solution you'd like.
Meine Junkers Cerapur Eco und IPM2 wird korrekt erkannt aber leider nicht der FW500. Bitte um Integration.

{
  "System Status": {
    "version": "3.5.0b4",
    "uptime": "000+02:37:33.059",
    "freemem": 119,
    "free_app": 6259,
    "reset reason": "Power on reset / APP CPU reset by PRO CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -67,
    "IPv4 address": "192.168.178.113/255.255.255.0",
    "IPv4 gateway": "192.168.178.1",
    "IPv4 nameserver": "192.168.178.1",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 447,
    "API fails": 8
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 8277,
    "bus reads (tx)": 434,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 3,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "de",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": true,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 5,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.07",
      "entities": 61,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26",
      "handlers ignored": "0x03 0x04 0x05"
    },
    {
      "type": "Mixer",
      "name": "IPM",
      "device id": "0x20",
      "product id": 102,
      "version": "20.07",
      "entities": 4,
      "handlers received": "0x010C",
      "handlers pending": "0x011E",
      "handlers ignored": "0xBE 0x010B 0x1E 0x1A"
    },
    {
      "type": "Mixer",
      "name": "IPM",
      "device id": "0x21",
      "product id": 102,
      "version": "20.07",
      "entities": 4,
      "handlers received": "0x010C",
      "handlers pending": "0x011E",
      "handlers ignored": "0xBE 0x010B 0x1A"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.07",
      "entities": 0
    },
    {
      "type": "Unbekannt",
      "name": "unknown",
      "device id": "0x10",
      "product id": 116,
      "version": "17.02",
      "entities": 0
    }
  ]
}

[log.txt](https://github.com/emsesp/EMS-ESP32/files/9724971/log.txt)

proddy, 2022-10-06:
@MichaelDvP another thermostat to add!

MichaelDvP, 2022-10-07:
Very nice, all needed infos (system info and log) in the first post. 
The log gives a bit to think about. 
there are some `000+01:08:10.088 ERROR 111: [emsesp] Call failed with error: cannot find values for entity 'flowtemphc' (Error)` 
which normally came from mqtt.publish2command:true with mosquitto (or publish to subscribed in iobroker), but the setting is false. Later there is  
```
000+02:55:11.290 DEBUG 198: [command] Calling command 'system/info' (show system status)
000+02:55:11.495 DEBUG 199: [command] Calling command 'boiler/values' ()
```
looks like Thomas' ioBroker-adapter is starting.

In the log there is only the thermostat monitor message, but i think FW500 will use the same setting-telegram as FW200.

ollikomo, 2022-10-07:
Yes you are right. The IOBroker adapter from Thomas is used. If you need a log again just let me know. And thanks for the quick response.

ollikomo, 2022-10-07:
I have now updated the ESP. The FW500 is recognized but now there are lots of EMS telegrams read (Tx) and EMS telegrams written (Tx) errors. I once attached a new log file this time without the IOBroker adapter.
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/9732546/log.1.txt)

MichaelDvP, 2022-10-07:
Seems the clock of thermostat is wrong in secondsiand dayofweek. A write is acknowlaged, but the clock does not change. Could you please post the values of thermostat.

ollikomo, 2022-10-07:
I hope you mean the data?
[device_entities.csv](https://github.com/emsesp/EMS-ESP32/files/9733222/device_entities.csv)

ollikomo, 2022-10-07:
Got the new build on it. After 15 minutes still no error displayed. Looks great so far. I'm still attaching a log to check. If it looks good from your side too, it can be closed here. I thank you.
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/9733512/log.3.txt)

MichaelDvP, 2022-10-07:
There is a switch module at device-id 0x13 and the boiler sometimes sends a null in telegram 0x5/offset 34, but this device is not registered on the bus, strange. Do you get a response to `read 13 02` 

Still wondering why the clock is not writable, maybe with right day-of-week value, but why it does not show summertime (daylightsavingtime)? If a owner of junkers FW-thermostat check it in terminal if time is setable, it would be helpfull to re-implement it : Do a `read 10  06`, check the `data` and set `call system send "0B 10 06 00 data"` with same data but last byte 01 or 03 instead of 00. e.g. data:16 0A 0D 07 1C 16 05 00  (2022-10-07 14:29:22 dow 5, dst 0)

proddy, 2022-10-07:
> Still wondering why the clock is not writable, maybe with right day-of-week value, but why it does not show summertime (daylightsavingtime)? If a owner of junkers FW-thermostat check it in terminal [...]

@nemesis03 could you help test this on your FW100?

nemesis03, 2022-10-07:
I just tried to set the date/time in the terminal on my FW100 with the last byte 01 or 03 instead of 00. The date/time I get back is always with "00" as the last byte

MichaelDvP, 2022-10-07:
So the Junkers do not set dst in this telegram. Could you also test changing the hour (3rd byte in this telegram, 0D in the above example).

ollikomo, 2022-10-07:
A little note and question. I once had a FB10 attached a few years ago. But it didn't work the way I wanted. Had to remotely control the HC2 upstairs, but that didn't work. the FW500 recognized him but did not process any data. I was able to set what I wanted on the FB10, but nothing changed on the FW500. There was also no temperature displayed by the Fb10 on the FW500. Now my question can it be that the FB10 is somehow still stored in the bus as 0x13
I can also connect the FB10 again as a testp to see which ID it has. I've always assumed that the FB10 is defective, it was bought cheaply on ebay to try it out

MichaelDvP, 2022-10-07:
The device 0x13 is not stored in boiler, but boiler sends a value to it, Afair 0x13 is a kind of switch, i don't know for what. 

A FB10 as remote control has to be configured as remote, and a FW500 hc has to configured as remote controlled, then FB10 will show up as device 0x18 or 0x38 and sends its temperature once a minute to the master thermostat.

nemesis03, 2022-10-07:
yes, I can change the the hour, minute, second, day, month and year through the terminal.
The day of week is always corrected in the response

MichaelDvP, 2022-10-07:
Have you send the whole telegram with last zero? Or only the hour, eg. `call system send "0B 10 06 02 0F"`  for hour 15? Or all date/time without last two bytes.

nemesis03, 2022-10-07:
I did send the whole string (like: `call system send "0B 10 06 00 16 0A 0D 07 13 13 05 00"`) (the "05" is then updated to the correct day of week)
but sending `call system send "0B 10 06 02 0F"` does also work and sets the correct hour

ollikomo, 2022-10-07:
I don't want to ask too much, but... I've now connected my FB10. However, it is recognized in the ESP as FR10 with 0x19. In the FW500 I set "HC2 Remote Control" to FB10. A current room temperature of -327.6°C is displayed in the ESP under FW500 and 22.5°C under FB10.
As soon as the FB10 is connected, my HC2 no longer works either, but you can certainly see that in the log. I'm curious what you say about it.
[ems-esp.local_api_system.txt](https://github.com/emsesp/EMS-ESP32/files/9735582/ems-esp.local_api_system.txt)
[log (4).txt](https://github.com/emsesp/EMS-ESP32/files/9735584/log.4.txt)
[image]

MichaelDvP, 2022-10-08:
I think you have to configure the FB10 to be remote, now it acts like a masterthermostat.

nemesis03, 2022-10-08:
> I think you have to configure the FB10 to be remote, now it acts like a masterthermostat.

I am very interested in doing this with my FB100. Any advise on how to do that is highly appreciated. Currently my FB100 is configured to work as a master thermostat for heating circuit 1, but I want to use it as a remote for my FW100 (actually just to sniff the commands being sent on the bus)

MichaelDvP, 2022-10-08:
I don't know how to configure the Junkers thermostats, check manual and the display in configuration mode. My knowledge is mostly from #336. I expect the remote thermostat at device-id 0x19 (hc2) and sending telegram 0x123 to the master-thermostat.  Don't know what is wrong, this FB10 broadcasts  0x170 instead, the master telegram for hc2.

ollikomo, 2022-10-08:
I'll do a factory reset tonight. and adjust according to the instructions. The only stupid thing in the instructions is that you should set the HC 1-10 (parameter 5A) first, in my case 2 . The instructions then state that 1 should be set for the FW100 and 2 for the FW200 (parameter 5B). The FW500 is not mentioned at all. But I can also select number 3, which should probably be my guess of the FW500. Then there is parameter 6A which is for temperature offset. For the next parameters there is nothing in the instructions for what they stand for.
6B default value 40
6C default value 80
6D default value 75
6E default value nothing
Maybe someone knows exactly what can be set there, I have a guess that it can be the flow temperature for HC and WW and the pump running time. I'll reset everything later and see what happens.

MichaelDvP, 2022-10-08:
Have you tried 5B set to FW200? afaik FW100 is only one hc, FW200 can manage hc1 and hc2, but hc3 needs a new master. But you want to control hc2 as remote, so it should be compatible to FW200.

ollikomo, 2022-10-08:
I've tried everything, no matter what I do it doesn't work. But I think what my problem is. It's not the FB10 but the whole system, and I don't have the FW500 in the gas boiler but outside in the living room. I read the instructions carefully again and now understand that a FB 10 or FB100 only works if the FW500 is connected directly to the boiler. In my case, that means I would have to install the Fw500 in the boiler, the FB10 on the upper floor and an additional FB100 in the living room. Can that be?

MichaelDvP, 2022-10-09:
No, normally you can use the master thermostat as roomthermostst in one hc and a remote thermostat in a different hc. 
But your FB10 acts as masterthermostat and sends commands to the mixer.
Try to set the FB10 to 5A=10, this should give device-id 0x39.

ollikomo, 2022-10-09:
I set 5A to 10. It is still recognized as FR10 and the error message Error 12/178 IPM coding with 10 not recognized comes in the FW500.
That's how it should be, only that with the FR10 surprises me.

ollikomo, 2022-10-21:
a little update..
I have now inserted the FW500 directly into the boiler and reconnected the fb 10. It looks like the controls are working as they should. the only thing that is not displayed in the FW500 is the room temperature of FB10. However, the flow temperature is controlled correctly. Now to the EMS ESP:
The room temperature HC2 is still not displayed correctly (currently -327.6°C). The TB10 is still recognized as an FR10.
Attached is the log file and the Api/System again.
[ems-esp.local_api_system(1).txt](https://github.com/emsesp/EMS-ESP32/files/9839849/ems-esp.local_api_system.1.txt)
[log (5).txt](https://github.com/emsesp/EMS-ESP32/files/9839857/log.5.txt)

proddy, 2022-10-24:
what is a TB10?

ollikomo, 2022-10-27:
I meant FB10.
Such a new twist. I got a new FB10 from the heating engineer and lo and behold, an FB10 is displayed and the temperature on the FW500 is also recognized correctly. Then my old FB10 was actually an FR10.

ollikomo, 2022-11-10:
yes can be closed. runs great

closed 2023-02-19 by proddy