emsesp/EMS-ESP32 #669: Cannot update RX TX gpio pins

Rainson12, 2022-10-07:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

I am using the EMS-ESP32 software. 

**Steps to reproduce**

* Flash the software
* start the esp32 
* login to web interface
* update the custom pins from 23/5 (rx/tx) to something like 16/17 
* when the interface asks for restart, click it
* after restart the pins 23/5 are again set

**Expected behavior**
changed gpio pins would be saved

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

this is taken after updating the pins
```
{
  "System Status": {
    "version": "3.4.2",
    "uptime": "000+00:01:47.426",
    "freemem": 157,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "[time.google.com](http://time.google.com/)",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "disconnected",
    "enabled": false,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": false,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 1,
    "ha enabled": false,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "disconnected"
  },
  "Settings": {
    "board profile": "CUSTOM",
    "tx mode": 1,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 1,
    "bool dashboard": 1,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": []
}
```

these are the logs:

```
000+00:00:00.000 I 0: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
000+00:00:00.000 I 1: [emsesp] No board profile found. Re-setting to CUSTOM
000+00:00:00.000 I 2: [emsesp] Starting EMS-ESP version 3.4.2 (hostname: ems-esp)
000+00:00:00.000 I 3: [dallassensor] Starting Dallas sensor service
000+00:00:00.000 I 4: [analogsensor] Starting Analog sensor service
000+00:00:00.000 I 5: [emsesp] Loaded EMS device library (102 records)
000+00:00:00.000 I 6: [emsesp] Starting Access Point with captive portal on 192.168.4.1
```

MichaelDvP, 2022-10-08:
I think this isthe same as #614. You need to upload v3.4.1 or v3.5.0.b to initialize the file system. After first start you can upload v3.4.2 again.
@proddy I think we should release a 3.4.3 with filesystem bug fixed.

proddy, 2022-10-08:
Yes, I was pretty sure it was added to 3.4.2 (main) but I checked and that fix isn't there. I'll do a v3.4.3.

But the filesystem fix was for new systems where there was no settings file. In this case there is one, so it may be something else. I'm checking now...

proddy, 2022-10-08:
@Rainson12 try [v3.4.3](https://github.com/emsesp/EMS-ESP32/releases/tag/v3.4.3)

closed 2022-10-14 by proddy