emsesp/EMS-ESP32 #670: ENTITIES  Restarts after Update

stmb-100, 2022-10-08:
After updating to Development Build v3.5.0b5, all ENTITIES in the dashboard are set to 0 at regular intervals and a new scan begins.

[log.txt](https://github.com/emsesp/EMS-ESP32/files/9738916/log.txt)

EMS-ESP Version
v3.5.0b5

Device (Platform / SDK)
ESP32 / v4.4.2

System Uptime
000+00:00:14.257

CPU Frequency
240 MHz

Heap (Free / Max Alloc)
91 KB / 39 KB

Flash Chip (Size / Speed)
4.096 KB / 40 MHz

Application (Used / Free)
1.864 KB / 56 KB

File System (Used / Free)
32 KB / 160 KB

Regards
Stefan

proddy, 2022-10-08:
wow, that's not good. I'll look into it. Did you compile yourself or take the github release build?

MichaelDvP, 2022-10-08:
Please send your configuration from help->support-info. 
I think best to clean reflash the module via usb. Filesystem with 196KB is not normal, also heap is very low. 
The log shows a homeassistant configuration, but there are a lot of commands coming from mqtt, could be a publish to command-topic, that does not work with HA/mosquitto broker. (check in mqtt config: disable HA-discovery und uncheck publish single/publish to command, reenabe HA discovery, save) 
The low uptime looks like a crash/restart.

proddy, 2022-10-08:
also did you come from v3.5.0b4?

stmb-100, 2022-10-08:
> wow, that's not good. I'll look into it. Did you compile yourself or take the github release build?

the Github Release Build

stmb-100, 2022-10-08:
> also did you come from v3.5.0b4?

yes

stmb-100, 2022-10-08:
[emsesp_info (3)..txt](https://github.com/emsesp/EMS-ESP32/files/9739508/emsesp_info.3.txt)

The file has Null Information

MichaelDvP, 2022-10-08:
> The file has Null Information

Free heap to low, i think due to mqtt queue and incomming mqtt commands. 
Again: best is to flash with usb and use the right partitions. 
2.nd best is reset to factory defaults, connect to AP and configure from scratch.

stmb-100, 2022-10-08:
[emsesp_info (4)..txt](https://github.com/emsesp/EMS-ESP32/files/9739543/emsesp_info.4.txt)

MichaelDvP, 2022-10-08:
> "Network Status":  "disable sleep": true,

Please disable this, sometimes this can give instabilities
> "MQTT Status": "publish2command": true,

Disable this, your log shows that the mqtt broker sends the publishes as command back, this makes much traffic and cost much ram. This works only on special settings of the mqtt broker, especially ioBroker-mqtt.
> "Devices": [

Funny, no boiler at this time. But uptime is low.

proddy, 2022-10-08:
also I see all the MQTT publishes are failing, perhaps because of the hostname have a ',' ? I'll try it out here

proddy, 2022-10-08:
also looks like all the MQTT settings are on. It's recommended to have both Clean Session and Retain set to disabled, unless you really know what you're doing and how MQTT works.

MichaelDvP, 2022-10-08:
@proddy It's one minute uptime and 1800 mqtt publishes, the queue is constantly overflowing because ems-esp and mqtt broker playing ping-pong with the messages. The fails are only 1300, so 500 are send out, nearly 8 per second.

proddy, 2022-10-08:
that's it!

proddy, 2022-10-08:
also surprised it worked in 3.5.0b4  :flushed:

stmb-100, 2022-10-08:
> also surprised it worked in 3.5.0b4

in 3.5.0b4 everything worked great

proddy, 2022-10-09:
even if i copy your settings I cannot reproduce

stmb-100, 2022-10-09:
Fallback to "version": "3.5.0b4" and it works again.

Flash with USB and ESPTool (Windows)

esptool.exe --port COM7 --chip esp32 erase_flash

esptool.exe --chip esp32 --port COM7 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 EMS ESP.bin

EMS ESP version
v3.5.0b4
Device (Platform / SDK)
ESP32 / v4.4.2
system uptime
000+11:17:23.891
CPUFrequency
240MHz
Heap (Free / Max Alloc)
107KB / 51KB
Flash chip (size / speed)
4096KB / 40MHz
Applications (Used / Free)
1,869KB / 115KB

File System (Used / Free)
24KB / 40KB
Is that too small how can I extend it?

greeting
Stefan
[emsesp_info.72.txt](https://github.com/emsesp/EMS-ESP32/files/9741184/emsesp_info.72.txt)

proddy, 2022-10-09:
I see lots of traffic in API calls and mqtt, about 40 publishes every minute. There's something special about your setup, perhaps the publish single option or an external app like iobroker? If you have time try the latest b5 again but disable mqtt and check memory. Then add mqtt but not HA. And check again. The only way to get to the root cause here is by process of elimination. Very old school!

stmb-100, 2022-10-09:
> I see lots of traffic in API calls and mqtt, about 40 publishes every minute. There's something special about your setup, perhaps the publish single option or an external app like iobroker? If you [...]

Yes, I have IOBroker running via the API and MQTT for IOBroker too

MichaelDvP, 2022-10-09:
When using publish to command topic make sure the ioBroker-mqtt does not publish on subscribe, connect etc. These settings are working:
[image]

proddy, 2022-10-14:
@stmb-100 can we close this?