emsesp/EMS-ESP32 #671: REST-API - strange characters in http://<hostname>/api/<device>/commands output

evroom, 2022-10-09:
Hi,

REST-API calls for `http://<hostname>/api/<device>/commands` have strange characters in the output.
Seen after updating to EMS-ESP Version: v3.5.0b5 (from v.3.4.2b5).

Example: `http://ems-esp.local/api/Boiler/commands`

```
{
  "boilhystoff": "•",
  "boilhyston": "•",
  "burnmaxpower": "¥¦§¨©ª«¬xV­ºL¢ü?$±ý?",
  "burnminperiod": "¥¦§¨©ª«¬xV­ºL¢ü?$±ý?",
  "burnminpower": "¥¦§¨©ª«¬xV­ºL¢ü?$±ý?",
  "commands": "lists all commands",
```

Same for the devices Thermostat and Mixer.

Is this a bug or a local issue ?

Greetings,

E.J.

evroom, 2022-10-09:
Here my system info:

[emsesp_info_v.3.5.0b5.txt](https://github.com/emsesp/EMS-ESP32/files/9741307/emsesp_info_v.3.5.0b5.txt)

proddy, 2022-10-09:
Definitely a bug. Thanks for reporting

evroom, 2022-10-09:
Solved with EMS-ESP Version v3.5.0b6
(EMS-ESP-CI-3_5_0b6.bin)

`http://ems-esp.local/api/Boiler/commands`
```
{
  "boilhystoff": "hysteresis off temperature",
  "boilhyston": "hysteresis on temperature",
  "burnmaxpower": "burner max power",
  "burnminperiod": "burner min period",
  "burnminpower": "burner min power",
  "commands": "lists all commands",

```
Thanks for the quick fix !

evroom, 2022-10-09:
Solved with EMS-ESP Version v3.5.0b6
(EMS-ESP-CI-3_5_0b6.bin)

closed 2023-02-19 by proddy