emsesp/EMS-ESP32 #673: more BusID's

stmb-100, 2022-10-09:
Is it possible to add further BUS-IDs for terminals.
My gateway (Premium II base board) has the address Terminal 0x0A.
ServiceKey 0x0B and Modem 0x0D is occupied
TimeModule 0x0F and AlarmModule 0x12 remain.
I would like to attach a second gateway (EMS interface board V3) to the bus for testing. Can I also attach this to the address 0x0F or 0x12?

proddy, 2022-10-09:
I'd say just try 0xF and 0x12 and see what happends. To be safe enable the "Enable read-only mode (blocks all outgoing EMS Tx Write commands)" option to ensure your 2nd gateway board is only in 'listen' mode.

MichaelDvP, 2022-10-12:
I've tested with ids 0x48 to 0x4F: 0x48 to 0x4D are working good, no unwanted communication like 0x12. 
0x4E and 0x4F not working, the master does not echo to TX, but i have ems1.0, maybe ems+ handles these ids.

proddy, 2022-10-16:
Ok, so remove 0x12 and add 0x48 -> 0x4D ? I think we can just do this easily in the web code and change console.cpp for the command

MichaelDvP, 2022-10-16:
0x12 could be fine for overriing a thermostat and setting the flowtemp, but we need to reply to version request for this. I think remove for now, but keep the option in mind.
Converter 0x0E works good, add it.
Gateway 0x48 is mostly KM200, works good
0x49-0x4D how to call it in UI, only number or  `Gateway 2`...
[image]

proddy, 2022-10-16:
hmm, we need a name. Either Gateway, Special or Custom ?

closed 2022-11-11 by proddy