emsesp/EMS-ESP32 #678: Entities with read only null/blank values can be seen in customisation panel, but are not visible/pushed to MQTT.

rossmck007, 2022-10-10:
**Bug description**
**A clear and concise description of what the bug is.

Not all entities on a given device are visible to EMS-ESP dashboard, but can be seen in User Customisation panel.  Hence these entities are not on the publish schedule for MQTT.

**Mention which EMS-ESP version you're using.*

EMS-ESP v3.5.0b6 (also  on b4 previously) connected in serial via 2 wire to (Junkers HT3) Bus.  

**Steps to reproduce**
*Steps to reproduce the behaviour.*

1. See that the dashboard shows description of a given device with 0 entities listed, or X entities listed when the device is know to have X+Y entities in total.  

[image]

2. For a device with known entities missing, have a look in Settings/Customisation, then select the same device and the true total list of entities then shows. 

[image]

3. Note that for each entity that is not being shown, the value in the User Customisation column is blank/null suggesting it can't be read correctly.

[image]

**Expected behavior**
*A clear and concise description of what you expected to happen.*

I would expect that all entities would be able to be interfaced through EMS-ESP to MQTT.  In my case I wish to compare (in Home Assistant via MQTT) the flow temperature of HC1 and HC2, but I found that HC1-flow is not listed as an entity for publication, despite HC2-flow being available.  Hence my search to find the root cause.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

3 embedded.

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

{
  "System Status": {
    "version": "3.5.0b6",
    "uptime": "000+00:41:20.076",
    "freemem": 111,
    "free_app": 6264,
    "reset reason": "Software reset CPU / Software reset CPU"
  },
  "Network Status": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -65,
    "IPv4 address": "192.168.0.228/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "195.130.130.4",
    "static ip config": false,
    "enable IPv6": false,
    "low bandwidth": false,
    "disable sleep": false,
    "AP provision mode": "disconnected",
    "AP security": "wpa2",
    "AP ssid": "ems-esp"
  },
  "NTP Status": {
    "network time": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tz label": "Europe/Amsterdam"
  },
  "OTA Status": {
    "enabled": true,
    "port": 8266
  },
  "MQTT Status": {
    "MQTT status": "connected",
    "MQTT publishes": 2112,
    "MQTT publish fails": 0,
    "enabled": true,
    "client_id": "ems-esp",
    "keep alive": 60,
    "clean session": true,
    "base": "ems-esp",
    "discovery prefix": "homeassistant",
    "nested format": 2,
    "ha enabled": true,
    "mqtt qos": 0,
    "mqtt retain": false,
    "publish time boiler": 10,
    "publish time thermostat": 10,
    "publish time solar": 10,
    "publish time mixer": 10,
    "publish time other": 10,
    "publish time sensor": 10,
    "publish single": false,
    "publish2command": false,
    "send response": false
  },
  "Syslog Status": {
    "enabled": false
  },
  "Sensor Status": {
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0
  },
  "API Status": {
    "API calls": 0,
    "API fails": 0
  },
  "Bus Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "bus telegrams received (rx)": 3844,
    "bus reads (tx)": 235,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 0,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100
  },
  "Settings": {
    "board profile": "S32",
    "locale": "en",
    "tx mode": 3,
    "ems bus id": 11,
    "shower timer": false,
    "shower alert": false,
    "rx gpio": 23,
    "tx gpio": 5,
    "dallas gpio": 18,
    "pbutton gpio": 0,
    "led gpio": 2,
    "hide led": false,
    "notoken api": false,
    "readonly mode": false,
    "fahrenheit": false,
    "dallas parasite": false,
    "bool format": 6,
    "bool dashboard": 5,
    "enum format": 1,
    "analog enabled": true,
    "telnet enabled": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3",
      "device id": "0x08",
      "product id": 95,
      "version": "18.07",
      "entities": 60,
      "handlers received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34",
      "handlers fetched": "0x14 0x16 0x33",
      "handlers pending": "0xBF 0xC2 0x1A 0x35 0x26 0x2A",
      "handlers ignored": "0x05"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "device id": "0x10",
      "product id": 158,
      "version": "74.03",
      "entities": 81,
      "handlers received": "0x06 0xA2 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers fetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02A6 0x02BA 0x02B0 0x029C 0x02CE 0x02F5 0x031B 0x023A 0x0240",
      "handlers pending": "0xA3 0x12 0x13 0x0471 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x02D0 0x02D2 0x031E",
      "handlers ignored": "0x35 0x02E1 0x02E2 0x02EB 0x02EC 0xBF 0x02C4"
    },
    {
      "type": "Thermostat",
      "name": "RC200/CW100",
      "device id": "0x39",
      "product id": 157,
      "version": "41.08",
      "entities": 0,
      "handlers pending": "0x042B 0x047B 0x0273",
      "handlers ignored": "0x042C 0xBF"
    },
    {
      "type": "Mixer",
      "name": "MM200",
      "device id": "0x20",
      "product id": 161,
      "version": "29.05",
      "entities": 3,
      "handlers received": "0x02D7",
      "handlers ignored": "0x1E 0x1A 0x0255 0xBF"
    },
    {
      "type": "Mixer",
      "name": "MM200",
      "device id": "0x21",
      "product id": 161,
      "version": "29.05",
      "entities": 4,
      "handlers received": "0x02D8",
      "handlers ignored": "0x1A 0x0255 0xBF"
    },
    {
      "type": "Controller",
      "name": "HT3",
      "device id": "0x09",
      "product id": 95,
      "version": "18.07",
      "entities": 0
    }
  ]
}

**Additional context**
*Add any other context about the problem here.*

On my EMS (HT3) Bus is the Junkers Boiler feeding a dual zone (CH and Underfloor) Bosch MM200 2 Zone Mixer, with respective Bosch CW400 wired thermostat (CH) and CW100 wired thermostat (underfloor).  

Profile BBQKS32, Tx HT3.

THANKS

rossmck007, 2022-10-10:
Clicking through on image 3 above I see the following:

[image]

This suggest that any entity with a value "not set" may be part of the issue.

proddy, 2022-10-11:
yes, this is by design. The entities listed on the dashboard is just a count of what you can see, so entities that have valid values. You have 3 on your MM200 and most probably 0 on your RC200.

rossmck007, 2022-10-11:
understood from the EKM-EMS perspective, and from the boiler perspective I guess some values will no longer be relevant as other parts of the extended hardware setup are delegated the task.  But hc1 flow temp (on MM200) and last_error_message (on main boiler) are two entities I would have thought could/should be readable at least?  Any insights from the wide variety of systems you support.  Great product, software and support options btw.

proddy, 2022-10-11:
> @proddy The utf-8 uom looks odd, should we add charset to the json-output?

yes, I'm using a JSON browser plugin so never saw that strange decoding character in the UOM. But I do see it from the Dashboard when you Export the page to a CSV file. Can we add the charset there too?

rossmck007, 2022-10-11:
@MichaelDvP newbie-alert can you elaborate on how I can query/poll telegram traffic/logs direct via Terminal.  I have however located the all-logs system log in the EMS-ESP UI.  

You are correct it turns out there was no heat demand at the time I pasted the system log so the MM200 circuits were not active.  I can confirm via the main thermostat UI in service mode, that under demand, the UI reports accurate non-zero flow temp, set point, and pump status for both hc1 and hc2.

In my brief test I also saw hc1 flow temp raise to >60deg under demand, and post demand is now reporting 20deg, I monitored EMS-ESP throughout this heat cycle, EMS-ESP reporting  "value": "not set" throughout (including multiple refresh of browser to force and update).

Thanks for your patience and attention.

MichaelDvP, 2022-10-12:
See [https://emsesp.github.io/docs/#/Console?id=console](https://emsesp.github.io/docs/#/Console?id=console), you'll find the password for `su` command in web under security settings. F1 or `help` shows all commands, you can also use every mqtt/api command via `call`. 

Refresh browser is normally not needed, the dashboard refreshs every 30 sec, Most values on the ems-bus will update only every minute, only a few monitor telegrams are send every 10 sec.

rossmck007, 2022-10-18:
@MichaelDvP thanks already, now all clear and after a few hoops:

Condition 1 no call for heat (mixer off, both zones)
`read 10 12 > Thermostat(0x10) -W-> Me(0x0B), RCErrorMessage(0x12), **data: <empty>**`
read 20 2D7 > Mixer(0x20) -W-> Me(0x0B), MMPLUSStatusMessage_HC(0x02D7), data: 00 00 07 80 00 00 00 00 03

Condition 2 call for heat in both zones (mixer on, both zones, visually confirmed with both zone pumps on)
`read 10 12 > Thermostat(0x10) -W-> Me(0x0B), RCErrorMessage(0x12), **data: <empty>**`
read 20 2D7 > Mixer(0x20) -W-> Me(0x0B), MMPLUSStatusMessage_HC(0x02D7), data: 01 00 07 80 00 43 00 00 08

In both cases the Bosch main thermostat (Service mode/diagnosis/data/'monitored values') correct 'set flow temp' and 'actual flow temp' for both HC1 and HC2, suggesting that the data is in fact available through HT3 bus?

Thanks in advance.

rossmck007, 2022-10-18:
@MichaelDvP @proddy 

>  80 00 flowtemp sensor not connected or broken.

Solved my issue, thanks a million.

Taking the MM200 apart and the protective covers off the entire manifold assembly - it became clear only the HC2 underfloor in/out flow temp detectors are fitted.  I assume as they are required for the underfloor circuit mixer - when this is not the case for the normal rad heating.

My confusion diagnosing this : the (quote) "HC 1 actual flow temperature" reported at the thermostat (I conclude) is not the actual (measured via sensor) but rather is the "HC 1 demanded/calculated/required flow temp".  This amounts to the same thing, but only the sensor values (the actual actual temps!) are being reported on the Bus.  And to compound my confusion, the HC2 is configured with sensors for both in/out flows (due to the nature of underfloor control/balance and need for the mixer).  I suspect whenever a mixer is fitted, actual temperature sensors are added as the control method.

For completeness I'll fit the HC1 circuit with an outflow temp sensor to measure the actual, if nothing else it will enable me to fault diagnose delta - but more importantly I'm interested in economising actual flow temps.

Thanks for your time both, hope this is useful for any other Junkers/Bosch MM200 users out there in the future.

Now I can get to actually understanding the data....!

EMS-ESP Fantastic product that defies even my installer who said it could not be done.  

I have also learned however that the old school Bosch thermostat is likely the best solution as it modulates, has climate compensation and a heating curve.  My intended DIY Tado interfacing (not natively compatible as can not be set as a master thermostat on the MM200) may not be the ultimate solution after all.

rossmck007, 2022-10-18:
Thanks again both.  Marked as closed and solved.

closed 2023-02-19 by proddy