emsesp/EMS-ESP32 #681: EMS-ESP management with "HeatingTemp"

MarkRoks, 2022-10-13:
Hello,

I have had an EMS-ESP32 gateway with (the latest) firmware v3.4.3 for several years now. I control the boiler via the Rest API, by adjusting the "HeatingTemp", which is then taken over by the "SelectedFlowTemp". 

If the "CurrentFlowTemperature" is low enough, relative to the "HeatingTemp", with the associated hysteresis, the boiler will start heating.

However, if I do not have a thermostat connected to the EMS, the SelectedFlowTemp falls back to 5.0 degrees. Regardless of what I (continue to) send to the "HeatingTemp". Subsequently, the boiler will therefore not start heating.

What could be the cause of this happening, that it falls back to 5.0 degrees at "SelectedFlowTemp"? The EMS-ESP gateway should also only (and only) be controlled via the Rest API.

I'd like to hear about it.

Kind regards,

Mark

proddy, 2022-10-24:
could be related to https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-works-at-first-but-is-then-reset-to-it39s-original-value ?

MarkRoks, 2022-10-25:
Hi Proddy,

Thanks for pointing me to a solution to my problem.

However, is this a temporary solution to the problem of the restfull API to the EMS-ESP gateway?

It's only recently that I've been experiencing this problem. Will this be fixed in an upcoming update?

Regards,

Mark

MarkRoks, 2022-10-25:
Hello proddy,

If I send the same value (5 degrees Celcius) to both the HeatingTemp and the SelFlowTemp, the HeatingTemp goes to 30 degrees Celcius and the SelFlowTemp to 15 degrees Celcius.

Is this coming from the EMS-ESP gateway?

Previously, the EMS-ESP gateway took exactly the values ​​that were sent. After receiving the advice, I notice this.

Regards,

Mark

proddy, 2022-10-25:
> Hi Proddy,
> 
> Thanks for pointing me to a solution to my problem.
> 
> However, is this a temporary solution to the problem of the restfull API to the EMS-ESP gateway?
> 
> It's only recently that I've been experiencing this problem. Will this be fixed in an upcoming update?
> 
> Regards,
> 
> Mark

I think this is a one off fixing left overs from previous versions. If it happens again let me know and we'll deep dive into it

proddy, 2022-10-25:
I don't know to be honest, there is are some default resetting and overrides from other EMS devices. @MichaelDvP knows this area better than I do. What devices are you using (sending your system debug profile will help)

MarkRoks, 2022-10-25:
Hi Proddy,

I understand. I have a Nefit Trendline boiler and a Bosch EasyControl thermostat.

Apart from the fact that the EMS-ESP gateway cannot read/write the data from the Bosch EasyControl, I do this in a roundabout way.

I read the status from HomeBridge, manipulate it via NodeRed and then send it to the boiler via the EMS-ESP gateway. I don't think it's anything weird really.

Despite this, the boiler still assumes different values ​​than what is sent from the EMS-ESP gateway.

At first I assumed that the Bosch EasyControl was causing the problem, to also send data directly to the boiler. I managed to rule out this cause by only allowing the Bosch EasyControl to communicate via a Bosch EasyControl Adapter. As a result, no data from the thermostat can reach the boiler (directly). Only through editing and control from Node-Red, from a Restfull API.

Regards,

Mark

MarkRoks, 2022-10-25:
Here my logs;

_2022-10-25 20:57:05.265 TRACE 7977: [emsesp] Me(0x0B) -> Boiler(0x08), UBASetPoints(0x1A), data: 05
2022-10-25 20:57:05.265 DEBUG 7978: [emsesp] No telegram type handler found for ID 0x1A (src 0x0B)
2022-10-25 20:57:05.285 DEBUG 7979: [emsesp] Last Tx write successful
2022-10-25 20:57:05.285 DEBUG 7980: [telegram] Sending post validate read, type ID 0xE4 to dest 0x08
2022-10-25 20:57:06.023 DEBUG 7981: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1369, retain=0, retry=1, size=499, pid=1)
2022-10-25 20:57:06.124 DEBUG 7982: [mqtt] Publishing topic ems-esp/boiler_data (#1370, retain=0, retry=1, size=790, pid=1)
2022-10-25 20:57:06.225 DEBUG 7983: [mqtt] Publishing topic ems-esp/thermostat_data (#1371, retain=0, retry=1, size=64, pid=1)
2022-10-25 20:57:07.417 DEBUG 7984: [telegram] Sending read Tx [#70], telegram: 0B 88 E4 00 1B
2022-10-25 20:57:07.433 TRACE 7985: [emsesp] Me(0x0B) <- Boiler(0x08), UBAMonitorFastPlus(0xE4), length: 0x1B
2022-10-25 20:57:07.455 DEBUG 7986: [emsesp] Last Tx read successful
2022-10-25 20:57:07.455 TRACE 7987: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty>
2022-10-25 20:57:07.455 DEBUG 7988: [emsesp] This telegram (UBAMonitorFastPlus) is not recognized by the EMS bus
2022-10-25 20:57:07.455 DEBUG 7989: [emsesp] No telegram type handler found for ID 0xE4 (src 0x08)
2022-10-25 20:57:07.750 DEBUG 7990: [telegram] Sending write Tx [#68], telegram: 0B 08 16 01 05
2022-10-25 20:57:07.750 DEBUG 7991: [emsesp] Fetching values for deviceID 0x08
2022-10-25 20:57:07.750 DEBUG 7992: [telegram] Tx read request to device 0x08 for type ID 0x14
2022-10-25 20:57:07.750 DEBUG 7993: [telegram] Tx read request to device 0x08 for type ID 0x16
2022-10-25 20:57:07.750 DEBUG 7994: [telegram] Tx read request to device 0x08 for type ID 0x33
2022-10-25 20:57:07.750 DEBUG 7995: [telegram] Tx read request to device 0x08 for type ID 0x26
2022-10-25 20:57:07.764 TRACE 7996: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameters(0x16), data: 05 (offset 1)
2022-10-25 20:57:07.764 DEBUG 7997: [emsesp] No telegram type handler found for ID 0x16 (src 0x0B)
2022-10-25 20:57:07.785 DEBUG 7998: [emsesp] Last Tx write successful
2022-10-25 20:57:07.785 DEBUG 7999: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
2022-10-25 20:57:07.852 TRACE 8000: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 1E 17 00 05 F1 0A 01 05 46 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 1E
2022-10-25 20:57:08.122 TRACE 8001: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: 00 (offset 27)
2022-10-25 20:57:08.417 DEBUG 8002: [telegram] Sending read Tx [#75], telegram: 0B 88 16 00 1B
2022-10-25 20:57:08.433 TRACE 8003: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), length: 0x1B
2022-10-25 20:57:08.447 DEBUG 8004: [mqtt] Publishing topic ems-esp/heartbeat (#1372, retain=0, retry=1, size=267, pid=1)
2022-10-25 20:57:08.509 DEBUG 8005: [emsesp] Last Tx read successful
2022-10-25 20:57:08.513 TRACE 8006: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 1E 17 00 05 F1 0A 01 05 46 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 1E
2022-10-25 20:57:08.548 DEBUG 8007: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1373, retain=0, retry=1, size=499, pid=1)
2022-10-25 20:57:08.649 DEBUG 8008: [mqtt] Publishing topic ems-esp/boiler_data (#1374, retain=0, retry=1, size=790, pid=1)
2022-10-25 20:57:08.817 DEBUG 8009: [telegram] Sending read Tx [#76], telegram: 0B 88 16 1B 20
2022-10-25 20:57:08.836 TRACE 8010: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), length: 0x20 (offset 27)
2022-10-25 20:57:08.860 DEBUG 8011: [emsesp] Last Tx read successful
2022-10-25 20:57:08.860 TRACE 8012: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 (offset 27)
2022-10-25 20:57:09.116 DEBUG 8013: [telegram] Sending read Tx [#71], telegram: 0B 88 14 00 1B
2022-10-25 20:57:09.132 TRACE 8014: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), length: 0x1B
2022-10-25 20:57:09.159 DEBUG 8015: [emsesp] Last Tx read successful
2022-10-25 20:57:09.159 TRACE 8016: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 14 D3 28
2022-10-25 20:57:09.485 DEBUG 8017: [telegram] Sending read Tx [#72], telegram: 0B 88 16 00 1B
2022-10-25 20:57:09.572 TRACE 8018: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), length: 0x1B
2022-10-25 20:57:09.638 DEBUG 8019: [emsesp] Last Tx read successful
2022-10-25 20:57:09.638 TRACE 8020: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 1E 17 00 05 F1 0A 01 05 46 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 1E 00 1E
2022-10-25 20:57:09.850 DEBUG 8021: [telegram] Sending read Tx [#77], telegram: 0B 88 16 1B 20
2022-10-25 20:57:09.866 TRACE 8022: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), length: 0x20 (offset 27)
2022-10-25 20:57:09.889 DEBUG 8023: [emsesp] Last Tx read successful
2022-10-25 20:57:09.894 TRACE 8024: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 (offset 27)
2022-10-25 20:57:10.283 DEBUG 8025: [telegram] Sending read Tx [#73], telegram: 0B 88 33 00 1B
2022-10-25 20:57:10.297 TRACE 8026: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), length: 0x1B
2022-10-25 20:57:10.350 DEBUG 8027: [emsesp] Last Tx read successful
2022-10-25 20:57:10.350 TRACE 8028: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 3C F1 00 28 00 00 46 D8 FF FF 00
[... cut ...]
2022-10-25 20:57:47.410 DEBUG 8076: [emsesp] No telegram type handler found for ID 0x23 (src 0x18)
2022-10-25 20:57:52.744 TRACE 8077: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 0F 00 F4 17 00 01 01 00 40 00 F1 80 00 00 F1 00 00 15 30 59 00 CC 00 00 00
2022-10-25 20:57:53.017 TRACE 8078: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00
2022-10-25 20:57:53.271 TRACE 8079: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 00 F1 80 00 20 00 00 01 00 00 49 71 00 51 E3 00 80 00_

MichaelDvP, 2022-10-26:
It is very confusing:
> However, if I do not have a thermostat connected to the EMS,
> I have a Nefit Trendline boiler and a Bosch EasyControl thermostat.
> As a result, no data from the thermostat can reach the boiler (directly).
>`Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00`

MarkRoks, 2022-10-26:
Hello,

I understand that the thermostat itself also sends a HeatingTemp to the boiler and the boiler takes over. However, if I now do this via the EMS-ESP gateway, both the HeatingTemp and the SelFlowTemp assume a different value.

This was never before. Therefore I assumed that this was influenced by the thermostat. However, by placing an EasyControl adapter in between, the signal to the boiler is provided entirely by the EMS-ESP gateway. It is therefore strange that these values ​​are adjusted, compared to the controlled one.

Regards,

closed 2023-02-19 by proddy