emsesp/EMS-ESP32 #682: EMS-ESP can't control my Bosch Condens 3000W

davbarhu, 2022-10-13:
Dear All,

 connected it to my Bosch Condens 3000W and my home network and evething seems to be okay. I can read the parameters, forward them via mqtt and so on.

Even so I can not control my boiler from the Gateway. I've tried it via the GUI of the GW and from Console too, but without any success. I do not have any other controller or thermostat on the BUS, only the boiler and your gateway.
I've dived into the forums and wiki, but I feels that I got stuck.

ems-esp-bosch-condens-3000:# show
EMS-ESP version 3.5.0b6

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:23.12) (57)
  heating active: off
  tapwater active: off
  selected flow temperature:  °C
  burner selected max power:  %
  heating pump modulation:  %
  current flow temperature:  °C
  gas: off
  gas stage 2: off
  heating pump: on
  fan: off
  ignition: off
  oil preheating: off
  heating activated: on
  heating temperature:  °C
  boiler pump max power:  %
  boiler pump min power:  %
  pump delay:
  burner min period:
  burner min power:  %
  burner max power:  %
  hysteresis on temperature:  °C
  hysteresis off temperature:  °C
  burner current power:  %
  burner starts:
  total burner operating time: 519 days 4 hours 7 minutes
  burner stage 2 operating time: 0 days 0 hours 0 minutes
  total heat operating time: 492 days 2 hours 7 minutes
  total UBA operating time: 1820 days 22 hours 10 minutes
  service code number:
  maintenance message: H00
  maintenance scheduled: off
  time to next maintenance:  hours
  dhw set temperature:  °C
  dhw selected temperature:  °C
  dhw type: buffer
  dhw comfort: hot
  dhw flow temperature offset:  °C
  dhw circulation pump available: off
  dhw charging type: 3-way valve
  dhw hysteresis on temperature:  °C
  dhw hysteresis off temperature:  °C
  dhw disinfection temperature:  °C
  dhw circulation pump mode: 2x3min
  dhw circulation active: off
  dhw current intern temperature:  °C
  dhw current extern temperature:  °C
  dhw current tap water flow:  l/min
  dhw activated: on
  dhw one time charging: off
  dhw disinfecting: off
  dhw charging: off
  dhw recharging: off
  dhw temperature ok: on
  dhw active: off
  dhw 3-way valve active: off
  dhw starts:
  dhw active time: 27 days 2 hours 0 minutes

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:23.12) (0)

I can't see the number values in CLI as above, but I can see it via web GUI.

ems-esp-bosch-condens-3000:# show ems
EMS Bus is connected.

EMS Bus info:
  Tx mode: 2
  Bus protocol: HT3
  #recognized EMS devices: 2
  #telegrams received: 711
  #read requests sent: 52
  #write requests sent: 3
  #incomplete telegrams: 0
  #read fails (after 3 retries): 0
  #write fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%

Rx Queue is empty

Tx Queue is empty

ems-esp-bosch-condens-3000:# show devices
These EMS devices are currently active:

Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:23.12)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x34
 Fetched telegram type IDs: 0x14 0x16 0x33
 Pending telegram type IDs: 0xBF 0xC2 0x1A 0x35 0x26 0x2A
 Ignored telegram type IDs:

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:23.12)

ems-esp-bosch-condens-3000:#
ems-esp-bosch-condens-3000:# show commands
Available commands (*=do not need authorization):
 system:
  commands              lists all commands *
  fetch                 refresh all EMS values
  info                  show system status *
  publish               publish all to MQTT *
  restart               restart EMS-ESP
  send                  send a telegram
  watch                 watch incoming telegrams *

 boiler:
  boilhystoff           hysteresis off temperature
  boilhyston            hysteresis on temperature
  burnmaxpower          burner max power
  burnminperiod         burner min period
  burnminpower          burner min power
  commands              lists all commands *
  emergencyops          emergency operation
  emergencytemp         emergency temperature
  entities              lists all entities *
  heatingactivated      heating activated
  heatingtemp           heating temperature
  info                  lists all values *
  maintenance           maintenance scheduled
  maintenancedate       next maintenance date
  maintenancetime       time to next maintenance
  pumpdelay             pump delay
  pumpmodmax            boiler pump max power
  pumpmodmin            boiler pump min power
  reset                 Reset
  selburnpow            burner selected max power
  selflowtemp           selected flow temperature
  wwactivated           dhw activated
  wwchargeoptimization  dhw charge optimization
  wwcirc                dhw circulation active
  wwcircmode            dhw circulation pump mode
  wwcircpump            dhw circulation pump available
  wwcomfort             dhw comfort
  wwcomfort1            dhw comfort mode
  wwdisinfecting        dhw disinfecting
  wwdisinfectiontemp    dhw disinfection temperature
  wwflowtempoffset      dhw flow temperature offset
  wwhystoff             dhw hysteresis off temperature
  wwhyston              dhw hysteresis on temperature
  wwmaxpower            dhw max power
  wwmaxtemp             dhw maximum temperature
  wwonetime             dhw one time charging
  wwseltemp             dhw selected temperature
  wwseltemplow          dhw selected lower temperature
  wwseltempsingle       dhw single charge temperature
  wwtapactivated        dhw turn on/off

ems-esp-bosch-condens-3000:#

ems-esp-bosch-condens-3000:# log trace
Log level: trace
000+00:32:05.461 T 46: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 23 01 54 64 00 01 12 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:32:05.670 T 47: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 8E 02 8E A0 00 00 03 00 00 98 58 00 0F 24 00
000+00:32:05.811 D 48: [mqtt] Publishing topic ems-esp/heartbeat (#436, retain=0, retry=1, size=290, pid=1)
000+00:32:10.132 D 49: [mqtt] Publishing topic ems-esp/boiler_data_ww (#437, retain=0, retry=1, size=461, pid=1)
000+00:32:10.236 D 50: [mqtt] Publishing topic ems-esp/boiler_data (#438, retain=0, retry=1, size=602, pid=1)
000+00:32:15.485 T 51: [emsesp] Boiler(0x08) -B-> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:32:15.710 T 52: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorFast(0x18), data: 23 01 52 64 00 01 12 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:32:15.920 T 53: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorWW(0x34), data: 0F 02 8E 02 8E A0 00 00 03 00 00 98 58 00 0F 24 00
000+00:32:16.150 T 54: [emsesp] Boiler(0x08) -B-> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 59 04 03 D8 0B 68 57 00 00 00 0A CF FF 03 F4 B4 80 00
ems-esp-bosch-condens-3000:# call boiler heatingtemp 45
000+00:32:19.218 D 55: [command] Calling command 'boiler/heatingtemp' (heating temperature) with value 45
000+00:32:19.271 D 56: [telegram] Sending write Tx [#124], telegram: 8B 08 16 01 2D
000+00:32:19.271 T 57: [emsesp] Me(0x0B) -W-> Boiler(0x08), UBAParameters(0x16), data: 2D (offset 1)
000+00:32:19.271 D 58: [emsesp] No telegram type handler found for ID 0x16 (src 0x0B)
000+00:32:19.302 D 59: [emsesp] Last Tx write successful
000+00:32:19.302 D 60: [telegram] Sending post validate read, type ID 0x16 to dest 0x08
000+00:32:19.383 T 61: [emsesp] Boiler(0x08) -B-> All(0x00), UBAParameters(0x16), data: FF 23 64 00 00 FB 03 01 03 64 0A 02
000+00:32:19.614 D 62: [telegram] Sending read Tx [#125], telegram: 8B 88 16 00 1B
000+00:32:19.614 T 63: [emsesp] Me(0x0B) -R-> Boiler(0x08), UBAParameters(0x16), length: 0x1B
000+00:32:19.664 D 64: [emsesp] Last Tx read successful
000+00:32:19.664 T 65: [emsesp] Boiler(0x08) -W-> Me(0x0B), UBAParameters(0x16), data: FF 23 64 00 00 FB 03 01 03 64 0A 02

The heating temp changing was just an example. I can not change any parameters.

Thank you in advance,
Best regards,
David

MichaelDvP, 2022-10-13:
The missing numbers is a bug in b6, the serialized json numbers have to be read `as<std::string>`  I'll add a commit to my PR.

The tx write look ok, the data is sent and the boiler acknowledged, but value is not set. Depending on boiler type and controller some values can not be changed or set back after a short while. But there is no know list of these types, you have to test which values can be changed.

MichaelDvP, 2022-10-13:
Yes, it's difficult. I have this controller: 
[image]
 and changing a temperature only works if the rotarys are set AUT. Switching on/off is not possible, also set by the knops.
I've read that on other controllers it is possible to send temperaturs lower than the controller setting, but not higher. Some completly electronic setable controllers allow to change all values.
A new idea from a recent discussion and test: device-ID 0x12 is normally a alarm module that allows setting the flowtemp by a analog input. Try to set ems-esp to id 0x12 and then try changing selflowtemp and selburnpow. Maybe this works.

davbarhu, 2022-10-14:
MichaelDvP you are a genius. I have the same issue what you have mentioned. I can lower the values, but I can not set higher values than the controller settings. We solved it! Thanks a lot!

closed 2023-02-19 by proddy

LordTompa, 2023-10-17:
Hi, I think I have exactly the same problem. Could someone explain to me how to set the device ID to device ID 0x12?
I'm new with the EMS-ESP and I'm at a loss.
Thanks for your help

proddy, 2023-10-19:
we removed the Alarm Module 0x12, can't remember why though.

MichaelDvP, 2023-10-20:
> we removed the Alarm Module 0x12, can't remember why though.

Because the thermostat is flooding the bus with version requests to alarm module. This stops if we reply wth a valid version, but we don't know what is reconfigured because of this information. 

The working solution for this issue was to set the flowtemp high on the control and only send lower values, this works from any device, no need to emulate a alarm module. The alarm module solution is a guess, not validated that it works. Not tested to side effects.