emsesp/EMS-ESP32 #686: Add min / max values on top of customization screen

tp1de, 2022-10-16:
Min / Max values can be changed / added to entities by using the customization screen. This is done by entering within pop-up window.
It's not where convenient to have no overview about entered (or original) min max values.

Let me recommend to add the 2 fields for min / max on the overview screen of customization settings.

proddy, 2022-10-16:
We knew of these trade-offs when the first version of adding min & max was implemented. I agree it would make sense at some point to fetch the default min/max from the server and show the range in the Value column for numeric values, like "22.5 (5/30)" and also populate the pop-up window.

MichaelDvP, 2022-10-17:
The empty popup was intended to show if customization is set, but can be changed to show actual min/max.
The default values are not stored in memory to save ram, we can only show the acual used min/max values. 
I think min/max columns are better than brackets and value should be last right.

tp1de, 2022-10-17:
The user interface for customization is not very convenient to adjust multiple values. Whenever a popup showed up the customization screen returns to original screen (starting with line 1). With multiple pages like for me, it's not very convenient to scroll down after each change. And for min/max values it's difficult to get the overview of changes since it's not on the overview screen of customizations - just within the  json-file. That's why I thought it might be good to see the min/max values on the first screen.

Nevertheless it's a nice feature to get the min/max changes within seconds updated in HA.

proddy, 2022-10-17:
I agree @tp1de . There's still plenty of work to do.

@MichaelDvP instead of adding a Min and Max column to the table, which really is only useful for numeric values that are writeable, we could instead just add a single column called 'Parameters' that for example show the type and range, e.g.
 'text'
 'enum'
 'number 5-30'
 'boolean'

?

tp1de, 2022-10-17:
I think it's not priority. Any chance that returning from pop-up window the cursor stays on selected line?

MichaelDvP, 2022-10-17:
It's for customization and this is done normally once. The json for all shortnames, longnames, customnames, mask and min max is very large. Blow it up with settings (bool, enum, etc)  that can not be customized? I think this is not needed. Min max shows only for entities that have a command and are numbers. For other entities and those without value the min max is not shown. This looks like this:
[image]

I agree that best is to set the right min/max while registering, but we don't know the right values and they are different for different boilers and thermostats. (e.g. maxburnpower 100%, 135%, 165%).

tp1de, 2022-10-30:
> but for my eyes the columns are better readable.

@MichaelDvP  I agree on the columns are better

MichaelDvP, 2022-10-31:
>  and if possible the cursor should return to the changed line when updating customizations.

I don't know how to do this the easy way (remember and restore scroll position).
I think @proddy knows better how to do it.

closed 2022-11-10 by proddy