emsesp/EMS-ESP32 #687: Reading solar sensor temp

ajvdw, 2022-10-16:
When connecting a solar sensor to a boiler (for reading the buffer temp) additional data is added to the 0x34 telegram.
For example: 
rx: 08 00 34 00 3E 02 50 80 00 21 00 00 01 00 00 40 82 00 24 62 00 02 2C 6F  -> 02 2C = 55.6 deg 
rx: 08 00 34 00 3E 02 29 80 00 21 00 00 01 00 00 40 82 00 24 62 00 00 BF 93 -> 00 BF = 19.1 deg
Please add this 'solar sensor temperature' - entity to emsesp

MichaelDvP, 2022-10-16:
This is easy to add, but i don't know what this really measures. Could you suggest names for mqtt-shortname, and the fullnames en, de, nl, se, pl, no  (or some of them) This a dhw value, so we have prefix dhw (WW in german, ww in shortname).

`"wwsolarboiler", "dhw solar boiler temperature", "WW  Solarboiler Temperatur", "dhw Zonneboiler temperatuur", "", "", ""`

ajvdw, 2022-10-16:
The name might be misleading: the typical use case is that a solar collector setup stores heat in a storage buffer. The energy is used for heating dhw. To ensure a proper dwh temperature, the water is also going through the boiler. The solar sensor measure the temperature of the storage buffer in order to decide if the water needs addtional heating to a legionella-safe temperature. This setup can also be used in a heat pump cascade.

ajvdw, 2022-10-17:
@MichaelDvP I've tested the entity. It shows up, but the value is factor 10 to high. 201.0 deg instead of 20.1

ajvdw, 2022-10-17:
The change was tested, issue solved.

closed 2022-10-17 by ajvdw