emsesp/EMS-ESP32 #692: Cant access Device Typ "Switch" via API

apos-t, 2022-10-17:
Hi,

When I try to access http://ems-esp/api/switch/ the output Message is: "unknown device"

The Log file says:

2022-10-17 18:04:58.888 D 1076: [command] Command failed: unknown device 'switch'
2022-10-17 18:04:58.888 E 1077: [emsesp] Call failed with error: unknown device (Error)

but you can see the device in the dashboard with the correct values:
[image]

**Device information**
`<html><body>
<div id="content"><div class="tabs "><nav class="tabs-navigation"><ul class="tabs-menu" role="tablist"><li class="tabs-menu-item json is-active" role="presentation"><span class="devtools-tab-line"></span></li></ul></nav></div></div><span class="devtools-tab-line"></span><span class="devtools-tab-line"></span><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"></div></div></div></div><div id="json-panel" style="visibility: visible; height: 100%;" class="tab-panel-box" role="tabpanel" aria-labelledby="json-tab"><div class="tab-panel json"><div class="jsonPanelBox tab-panel-inner"><div class="toolbar"><div class="devtools-separator"></div><div class="devtools-searchbox"><input class="searchBox devtools-filterinput" placeholder="JSON durchsuchen" value=""></div></div><div class="panelContent">

  |  
-- | --
System Status |  
version | "3.5.0b7"
platform | "ESP32"
uptime | "000+03:35:38.862"
freemem | 111
free_app | 6249
reset reason | "Software reset CPU / Software reset CPU"
Network Status |  
connection | "WiFi"
hostname | "ems-esp"
RSSI | -81
IPv4 address | "192.168.2.82/255.255.255.0"
IPv4 gateway | "192.168.2.1"
IPv4 nameserver | "192.168.2.3"
static ip config | false
enable IPv6 | false
low bandwidth | false
disable sleep | false
AP provision mode | "disconnected"
AP security | "wpa2"
AP ssid | "ems-esp"
NTP Status |  
network time | "connected"
enabled | true
server | "192.168.2.1"
tz label | "Europe/Berlin"
OTA Status |  
enabled | true
port | 8266
MQTT Status |  
MQTT status | "connected"
MQTT publishes | 4075
MQTT queued | 0
MQTT publish fails | 0
MQTT connects | 1
enabled | true
client_id | "ems-esp"
keep alive | 60
clean session | false
base | "ems-esp"
discovery prefix | "homeassistant"
nested format | 2
ha enabled | true
mqtt qos | 0
mqtt retain | false
publish time boiler | 15
publish time thermostat | 15
publish time solar | 120
publish time mixer | 15
publish time other | 120
publish time sensor | 15
publish single | false
publish2command | false
send response | false
Syslog Status |  
enabled | false
Sensor Status |  
temperature sensors | 0
temperature sensor reads | 0
temperature sensor fails | 0
API Status |  
API calls | 2932
API fails | 653
Bus Status |  
bus status | "connected"
bus protocol | "Buderus"
bus telegrams received (rx) | 12400
bus reads (tx) | 2632
bus writes (tx) | 2
bus incomplete telegrams | 4
bus reads failed | 0
bus writes failed | 0
bus rx line quality | 100
bus tx line quality | 100
Settings |  
board profile | "S32"
locale | "en"
tx mode | 1
ems bus id | 11
shower timer | false
shower alert | false
rx gpio | 23
tx gpio | 5
dallas gpio | 18
pbutton gpio | 0
led gpio | 2
hide led | true
notoken api | true
readonly mode | false
fahrenheit | false
dallas parasite | false
bool format | 6
bool dashboard | 1
enum format | 1
analog enabled | false
telnet enabled | false
Devices |  
0 |  
type | "Boiler"
name | "Topline/GB162"
device id | "0x08"
product id | 115
version | "03.06"
entities | 67
handlers received | "0x10 0x11 0x15 0x1C 0x18 0x19 0x34"
handlers fetched | "0x14 0x16 0x33"
handlers pending | "0xBF 0xC2 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
1 |  
type | "Thermostat"
name | "RC35"
device id | "0x10"
product id | 86
version | "01.08"
entities | 57
handlers received | "0xA3 0x06 0xA2 0x12 0x42 0x48 0x47 0x52 0x51 0x5C 0x5B 0x38 0x39"
handlers fetched | "0x3E 0x3D 0x3F 0xA5 0x37"
handlers pending | "0x13 0x49 0x4C 0x53 0x56 0x5D 0x60"
handlers ignored | "0x9D 0x1A 0x35"
2 |  
type | "Switch"
name | "WM10"
device id | "0x11"
product id | 71
version | "02.00"
entities | 2
handlers received | "0x9C 0x1E"
handlers pending | "0x9D"
3 |  
type | "Controller"
name | "BC10/RFM20"
device id | "0x09"
product id | 68
version | "02.03"
entities | 0

</div></div></div></div>
</body>
</html>`

apos-t, 2022-10-18:
Thank you in the new dev Build it`s working like excpected.

closed 2022-10-18 by apos-t

MHummie, 2022-11-24:
I'm having an issue with WM10 as well. Like in this screenshot, I can see "flow temperature (TC1)" and "status". However, the status is not published via MQQT while "flow temperature" is. In the customization tab, I don't see a name assigned to "status" either. Is this the same issue? I'm running v3.4.4 now, when I try to install the latest dev version I get an internal server error.

proddy, 2022-11-25:
> I'm having an issue with WM10 as well. Like in this screenshot, I can see "flow temperature (TC1)" and "status". However, the status is not published via MQQT while "flow temperature" is. In the [...]

going to v3.5.x on a BBQKees board needs a different method. There are notes on the wiki and also plenty of issues on this. What you need to do is make a backup of your configuration, then use the EMS-ESP Flasher from the Github project folder to upload the firmware. You only need to do this once, switching from v3.4 to v3.5