emsesp/EMS-ESP32 #695: Improvements to Polish translations

proddy, 2022-10-17:
Not at all. I am happy to help (as much as I can) because EMS-ESP is great and very usefull.
Unfortunately, Polish, as they say, is one of the most complicated languages ​​in the world and it is practically impossible to translate into Polish correctly without knowing the context. Most of the parameters that I have translated so far appear in my Buderus heat pump and I tried to translate them as best as possible based on the messages displayed by the pump controller or from the Polish manual of the pump. I will try to translate something more into Polish in the future.
However, there are two global problems with getting a correct Polish translation.
1. In the translation of the web interface you are breaking English sentences into words, the words are then translated into a specific language and already translated are "put together" back. It's probably ok for German but for Polish it will almost never give you a correct sentence! The order and "variant" of words usually also changes. Example: "Network" -> "Sieć", "Settings" -> "Ustawienia", but "Network Settings" != "Sieć Ustawienia" but "Ustawienia sieci" (word order changed, no capital letter in middle and one word changed)! The only 100% solution is to translate whole sentences (like for entity names). 
2. The translations of the values' names ​​also differ depending on the name of the entity. In English, for example, there is a simple OFF, but in Polish, one of the following may be correct: WŁĄCZ, WŁĄCZONO, WŁĄCZONE, WŁĄCZONA, WŁĄCZONY and many others. It would be great if few alternative translations (even only for Polish and only nesessary ones) could be added to the code. Do you think it's possible?
For cases with up to three variants of the word, PLURAL syntax: {{zero | one | other}} seems to be enough. Do you want to see my attempt? I can prepare test PR...

_Originally posted by @pswid in https://github.com/emsesp/EMS-ESP32/issues/683#issuecomment-1281303616_

proddy, 2022-10-17:
@pswid yes, we can make code changes to get this right.

pswid, 2022-10-18:
My idea is to make it as simple as possible to add several translations for a specific words/phrases (if needed) and select them later using the index. **plural (zero, one, other)** seems to be the simplest way if we need up to 3 translations of the single word/phrase. See my example of using alternative translations of `SYSTEM_CHECK`. Notice that only PL index file contains alternative translations. Other language index files are intact but everything still works (other languages just work as before). See modifications: https://github.com/emsesp/EMS-ESP32/commit/bb98f13b196c49a08ac12b77e49bb9b9dab2e902

after above changes we achieve this
[image]

What do you think about such approach? Do you have another/better idea?

pswid, 2022-10-18:
and when it comes to the reverse order of the words (e.g. STATUS or SETTINGS) - why not just pass the second word as a parameter?

we could define e.g.:
```
//** for EN and other */
STATUS: '{0} Status',
NETWORK: 'Network',
```

```
//** for PL */
STATUS: 'Status {0}',
NETWORK: {{Sieć | sieci | sieciowy}},
```

and later instead of
`{LL.NETWORK() + ' ' + LL.STATUS()}`

just use:
`{LL.STATUS( LL.NETWORK(1) )}`

for EN it will give us:
**'Network Status'**

and for PL:
**'Status sieci'**

This is exactly what we need! The same can also be done with SETTINGS ...

proddy, 2022-10-18:
yes, that's a good approach.

pswid, 2022-10-19:
OK. I'll try to make proper corrections in Polish translation using this approach. Hopefully later this week.

pswid, 2022-10-25:
I'm still working on it (it takes so long because I don't have too much free time). Web interface looks very good in Polish already (no changes for other languages). I have corrected all the places where the order of words or the variant of the words were incorrect. I still want to do some tests, but the fix will be ready for publication soon. Maybe even this evening...

proddy, 2022-10-31:
all done right?

pswid, 2022-11-02:
Now the web interface in Polish looks much better. However, there are still some phrases to be corrected/completed. Later this week I will try to prepare another update. When it comes to translating entity names, it is a more difficult matter, because I do not know the meaning of many of them. Nevertheless, I will try to do something else on this topic too...

pswid, 2022-11-02:
BTW, could you explain what exactly are doing options _Shower timer_, _Shower alert_ and especially _Cold shot duration_. I don't understand it, so it's hard to translate it correctly...

proddy, 2022-11-03:
there's a brief description here: https://emsesp.github.io/docs/#/Configure-firmware?id=general-options

proddy, 2022-11-03:
The polish translations break the other languages because those translations don't have the arguments to support the different phrasing. e.g. DEFAULT(1) will not work in other language as the string arguments are missing. I see a few options
 - adjust all the other languages to include the additional argument. This is a waste since they'll never be used and it will increase the file size and RAM needed
 - create a custom formatter (https://github.com/ivanhofer/typesafe-i18n/tree/main/packages/formatters) that has a special case if the locale is PO. This is my preferred option but tricky to implement. And it will slow down client-side rendering in the UI.
 - remove PO and make a special build only for Polish

For now since we have hundreds of users on the dev build I suggest to roll back the PO changes in the .tsx code until we can implement the custom formatter.

@MichaelDvP what do you think?

MichaelDvP, 2022-11-04:
> DEFAULT(1) will not work in other language as the string arguments are missing

Only the baseTranslation, Other languages working. 
> adjust all the other languages to include the additional argument. This is a waste since they'll never be used and it will increase the file size and RAM needed

- It's only a few words, We can reduce the cases to one character: `DEFAULT: 'Defaul{{t|t}}',`
- Or use "pl" again as baseTranslation and set in en: `DEFAULT: 'Default',`

I prefer to use the one char option in en baseTranslation. I'm supprised that you suggest a special build ;-)

proddy, 2022-11-06:
Think we can close this now.

closed 2022-11-06 by proddy

pswid, 2022-11-07:
> there's a brief description here: https://emsesp.github.io/docs/#/Configure-firmware?id=general-options

Thanks. I overlooked this information. As I understand, the functionality applies to a situation where hot water is stored in the reservoir and the boiler reports the current flow (in my heat pump, wwcurflow is always =0). But how can you change the outlet temperature quickly to generate cold bursts? In a flow heater it's easy, but in the case of a reservoir a posiibility do cut off the hot water would be required (e.g. quick three-way valve), right? Could you write something more about how it works?

pswid, 2022-11-07:
> Only the baseTranslation, Other languages working.
>
>
> * It's only a few words, We can reduce the cases to one character: `DEFAULT: 'Defaul{{t|t}}',`
> * Or use "pl" again as baseTranslation and set in [...]

Yes I also noticed that English translation was broken. It was because baseTranslation was changed back to EN and wrong plural notation `WORD: {{word}}` was used. For English (cardinal) it should be `{{one|other}}`.
As @MichaelDvP  said, you can fix it by setting baseTranslation to PL again or fixing the notation in EN file (it's only a few words), e.g. to `WORD: {{word|word}}`. 
BTW, special buid for PL would be a bad decision.