emsesp/EMS-ESP32 #699: Error/Warnung Logfile Home Assistant

Stefan28L, 2022-10-24:
I use the S32 Gateway from Kees on a Junkers ZBS 14/100 S-3 MA 21 S000 gas heating and a FW120 Thermostat.
EMS-ESP Version
v3.5.0b4

In my home assistant i get every 10 seconds an error in the log file:

>>>
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:164
Integration: MQTT (documentation, issues)
First occurred: 18:11:18 (126 occurrences)
Last logged: 18:32:08

Invalid option for select.thermostat_hc1_room_sensor: '' (valid options: ['extern', 'intern', 'auto'])
<<<

together with a warning:

>>>
Logger: homeassistant.helpers.template
Source: helpers/template.py:583
First occurred: 18:11:18 (126 occurrences)
Last logged: 18:32:08

Template variable warning: 'dict object' has no attribute 'roomsensor' when rendering '{{value_json.hc1.roomsensor}}'
<<<

Can you help me to solve the problem ?
Is it a problem of the home assistant mqtt discovery ?

I use HA on a rpi3-64.

Home Assistant 2022.10.5
Supervisor 2022.10.0
Operating System 9.2
Frontend 20221010.0 - latest

As mqtt broker i use the Mosquitto broker integration of HA.

proddy, 2022-10-24:
Can you try https://emsesp.github.io/docs/#/Troubleshooting?id=home-assistant ?

Stefan28L, 2022-10-25:
Thanks for the quick response.
Yes, I tried that.
Since I'm using version 3.5.0b4, I deactivated and activated the MQTT Discovery function for >5 minutes.
The error messages still pop up every 10 seconds..

proddy, 2022-10-25:
ah ok. before we try and reproduce this by simulating your setup can you try one more thing
 - make sure in EMS-ESP, MQTT Settings you have the Retain Flag and Clean Session both set to **off**
 - download MQTTExplorer
 - delete everything under homeassistant. if you other sensors installed, search and remove the ones that have ems-esp in the path
 - close your eyes
 - restart EMS-ESP

Stefan28L, 2022-10-26:
thanks for the detailed instructions. 
i followed your instructions. after restart, everything is working like before. 
even the two messages in the log are still there :-(

Stefan28L, 2022-10-26:
i double checked everything and found some ems-esp settings under homeassistant topic which i didn't delete before. i will check it again. sorry

Stefan28L, 2022-10-26:
unfortunately it still doesn't work
just to be shure not to make an mistake:
i have to delete the topics under "homeassistant" not the "ems-esp" topic itself ?

proddy, 2022-10-26:
> unfortunately it still doesn't work just to be shure not to make an mistake: i have to delete the subtopics under "homeassistant" not the "ems-esp" topic itself ?

doesn't matter, you can delete everything. It'll be recreated every few seconds anyway.

But looking at the problem again I think it may be something in the code that needs fixing. 

Can you do two quick tests
- in WebUI or Console do you see an entity called 'room sensor' (or the translated equivalent)
- can you go to Console and do a `watch 165` and post the output from a few telegrams
- can you check the 'enum format MQTT/API' in the Settings to see if its Value or Index. If it's Index then I suspect the bug is in `Mqtt::publish_ha_sensor_config()` only uses the string enum names and not the index number

Stefan28L, 2022-10-27:
"hc1 room sensor (hc1/roomsensor)" is the only entity with this name.

i'm not shure how / where to check the 'enum format MQTT/API". can you explain this to me ?

i have trouble to get a telnet connection to the gateway :-(

2022-10-27 22:05:15	Looking up host "192.168.178.59" for Telnet connection
2022-10-27 22:05:15	Connecting to 192.168.178.59 port 23
2022-10-27 22:05:15	client negotiation: WILL NAWS
2022-10-27 22:05:15	client negotiation: WILL TSPEED
2022-10-27 22:05:15	client negotiation: WILL TTYPE
2022-10-27 22:05:15	client negotiation: WILL NEW_ENVIRON
2022-10-27 22:05:15	client negotiation: DO ECHO
2022-10-27 22:05:15	client negotiation: WILL SGA
2022-10-27 22:05:15	client negotiation: DO SGA
2022-10-27 22:05:18	Failed to connect to 192.168.178.59: Network error: Connection refused
2022-10-27 22:05:18	Network error: Connection refused

unfortunately i run out of time for now.
I have to try later....

proddy, 2022-10-28:
go the webUI, Help, and click on that button that talks about getting support to download the support info file. Then attach it to this Github issue

Stefan28L, 2022-10-28:
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/9887340/emsesp_info.txt)

proddy, 2022-10-28:
Everything looks ok. Let me think about this and I'll get back to you

MichaelDvP, 2022-10-30:
The roomsensor was introduced in #641 and has 3 enum options 1,2,3. Maybe there is another otion we do not know. Please check in terminal `read 10 165` and post the output here.
As workaround you can go to customizations and exclude the roomsensor from mqtt.

Stefan28L, 2022-10-31:
Here is the output of the command `read 10 165`

`000+18:50:38.725 I 0: [emsesp] Thermostat(0x10) -W-> Me(0x0B), JunkersSet(0x0165), data: 01 00 02 19 3D 42 02 01 00 03 FF 28 06 01 02 20 27 29 01 01 00`

MichaelDvP, 2022-10-31:
Looks ok, the roomsensor has value 3 (min from extern/intern) and shoud also be published in thermostat_data. I don't know why HA complains.

Stefan28L, 2022-10-31:
Thank you @proddy and @MichaelDvP  for taking care on it.

As i don't need the information of this mode in HA, i have disabled the hc1 room sensor in mqtt. Now i have no more errors in the log.

One last information:

- can you check the 'enum format MQTT/API' in the Settings to see if its Value or Index. If it's Index then I suspect the bug is in Mqtt::publish_ha_sensor_config() only uses the string enum names and not the index number

The setting is set to "value".

MichaelDvP, 2022-10-31:
Please post the payload of mqtt topic `thermostat_data `and `homeassistant/select/%base%/thermostat/roomsensor`  
The settings according to enum-format are generated for all entities the same, i don't think roomsensor makes a difference, but let's see.

Stefan28L, 2022-10-31:
payload `thermostat_data`:
`{"wwcharge":"off","hc1":{"seltemp":20.5,"currtemp":20.6,"mode":"eco","modetype":"eco","heattemp":20.5,"ecotemp":20.5,"nofrosttemp":16,"control":"off","program":"prog a"}}`

payload `homeassistant/select/ems-esp/thermostat_hc1_roomsensor/config`
`{"~":"ems-esp","uniq_id":"thermostat_hc1_roomsensor","command_topic":"~/thermostat/hc1/roomsensor","options":["extern","intern","auto"],"stat_t":"~/thermostat_data","name":"hc1 Room sensor","object_id":"thermostat_hc1_room sensor","val_tpl":"{{value_json.hc1.roomsensor}}","dev":{"ids":["ems-esp-thermostat"]}}`

proddy, 2022-10-31:
I added the roomsensor to the test code in test.cpp (just the 1 line) and with `test cmd` can simulate the FW120. I don't get the error either in HA.

MichaelDvP, 2022-11-03:
That's strange: 
- your telegram 0x165 shows a valid value (3) for roomsensor
- the `thermoatat_data` does not contain roomsensor (is it excluded?)
- the ha-discovery exisits, (should be removed on exclude or value not set).
- roomsensor seems to be the only value, for all others it works

I have no idea what could cause this, sorry.

proddy, 2022-11-11:
closing, can't reproduce.

closed 2023-02-19 by proddy