emsesp/EMS-ESP32 #701: Change State/Mode on schedule

HerrRiebmann, 2022-10-25:
First of all: Thank you very much for this incredible project!

### Idea
A Scheduler to set a specific value at several predefined times.

### Implementation
- [x] Scheduler Active

- [x] Monday
- [x] Tuesday
- [x] Wednesday
- [x] Thursday
- [x] Friday
- [ ] Saturday
- [ ] Sunday

- [x] 07:30 `thermostat` `{"cmd":"hc1/mode", "data":"day"}`
- [x] 17:30 `thermostat` `{"cmd":"hc1/mode", "data":"night"}`

Or some kind of script-configuration: weekday;minutesOfDay;DeviceType;Cmd,Data
```
1;450;1;hc1/mode;day
1;1050;1;hc1/mode;night
2;450;1;hc1/mode;day
2;1050;1;hc1/mode;night
5;450;1;hc1/mode;day
5;960;1;hc1/mode;night
```

### Background
My Buderus Thermostat has about 6 different automatic-programs. None of them is suitable for a workingspace.
I´d just like to change the thermostat `hc1/mode` from **_day_** to **_night_** and vice versa, on five days a week.
Since the ESP is running all day long and I´m currently not planning to additionally introduce another (_constantly running_) device, I thought this could be realized within this great project.
As saied, the EMS-ESP is used in my office, where no Home Assistant will take place.

### Symbiosis
With this approach, it would also be possible to lower or raise the temperature. Change the state of the heatpump. Disable the warmwater cycle, and much more.

### Alternatives
Installing [Mosquitto](https://mosquitto.org/), [NodeRed](https://nodered.org/) and a [time scheduler](https://flows.nodered.org/node/node-red-contrib-ui-time-scheduler/in/590bc13ff3a5f005c7d2189bbb563976), which needs a Computer, Raspberry PI, Synology NAS or whatever.

Using any home automation.

Or manually changing the state with [HTTP Shortcuts](https://http-shortcuts.rmy.ch/) on Android.
(_My current approach_)

### Similar Request
Adding a scheduler for setting GPIOs was requested here:
https://github.com/emsesp/EMS-ESP32/issues/648
But I´m still hoping for a simple solution on a single device.

I´d love to see something like that, adjustable from the web interface!
In this huge project and without any experience in react, it´s hard to contribute this feature myself.

proddy, 2022-10-25:
As you've probably read from #648 we really don't want to build in a scheduler, although it is feasible. There are other automation systems that do this like you pointed out but your case you don't have one or want to install one, which I understand. 

I do like the idea of having a scheduler where you set a period (day, time, frequency) that calls a command. It would be fun to program. Let me think about it and discuss with Michael.

proddy, 2022-10-27:
shall I do the web front end and the C++ class stubs and you do the rest @MichaelDvP ? First we need to get 3.5.x out

proddy, 2022-10-31:
for the timing we could re-use some of https://github.com/emsesp/EMS-ESP32/blob/ft_aux/src/poolpump.cpp

MichaelDvP, 2023-02-24:
@proddy 
> Any thoughts on how the UI should look?

I thought of same look and feel as customizations, A list with:
DayofWeek  | Time | command | value
 M T W T F S S
[x][-][x][-][x][x][x]  | 10:30 | thermostat/hc2/seltemp | 20

Click on a line opens a dialog to change the values.
Maybe add a column periodical [x]  that disables dayofweek and time is than time between commands, not time of day.

proddy, 2023-02-24:
Ok, I'll start working on the WebUI prototype. What do you think about these requirements?
 - where to add the Scheduler? As a new section/menu item next to 'Settings' or as a new Tab in 'Settings' next to 'Customization' ?
 - As you described it'll show a table with dow, time, command and value with the optional 'periodic' in minutes
 - If the periodic is 0, then the command will be executed when EMS-ESP boots. This satisfies an earlier request for a 'boot loader' but only sends one command,, unless we support cascading later on
 - add a scheduler enabled/disabled option
 - should we disable if NTP is disabled or not running?

MichaelDvP, 2023-02-24:
1.Yes i think a settings tab. For ownEntities there could be another tab.
2. Yes, i think if perodic is active  dow is grayed, if dow havea setting periodic s gray. Time can be used for both.
3. Periodic 0 is a good idea. But difficult: when to activate this command, a defined time after boot, or poll if this command is already registered? Users can add more commands with periodic 0, all executed at startup.
4. good idea, as first checkbox to each line, set in list, the other options in dialog.
5. disable dow setting, periodic can be done without NTP.

We should have an extra setting file, not as section in customization.

proddy, 2023-02-25:
I'll start on the web front end today with the mock API do we can review the logic and json structure. Then next step is to create the WebScheduler C class

proddy, 2023-02-25:
@MichaelDvP did the first implementation of the web front-end...

[image]

The json format is 

```
ScheduleItem {
  id: string; // unique index
  active: boolean;
  flags: number;
  time: string;
  cmd: string;
  value: string;
  description?: string; // optional
}
```

I'll work next on the C code adding the end-points to load/save to a new config file on LittleFS. Then leave the heavy work for you in the loop :-)

MichaelDvP, 2023-02-25:
Looks nice, but i'm missing a button to add a new line.

proddy, 2023-02-25:
haha! completely forgot about that!!

HerrRiebmann, 2023-02-25:
Guys, you are gorgeous!
Looks exactly like I hoped for 😍

Thanks a lot 👍🏻 Looking forward to Beta-Test this feature.

proddy, 2023-02-26:
I add the ADD button and some of the C code. I still need to add the NTP check and tidy some stuff up which I'll try and finish today.

MichaelDvP, 2023-02-26:
I've tested, nice. Here is the first implementation of loop. Maybe the on boot needs more attention and checks. Also maybe trigger with uptime.
```
void WebSchedulerService::loop() {
    time_t          now       = time(nullptr);
    tm *            tm        = localtime(&now);
    static int      last_min  = 0;
    static uint32_t starttime = now / 60;

    if (tm->tm_min == last_min) {
        return;
    }
    last_min            = tm->tm_min;
    bool     has_NTP    = tm->tm_year > 120;
    uint8_t  dow        = 1 << ((tm->tm_wday + 6) % 7); // start with monday
    uint16_t min        = tm->tm_hour * 60 + tm->tm_min;
    uint32_t check_time = now / 60 - starttime;
    EMSESP::webSchedulerService.read([&](WebScheduler & webScheduler) {
        for (const ScheduleItem & scheduleItem : webScheduler.scheduleItems) {
            if (scheduleItem.active
                && ((has_NTP && (dow & scheduleItem.flags) && min == scheduleItem.elapsed_min)          // day of week scheduling
                    || (scheduleItem.flags == 0x80 && scheduleItem.elapsed_min == 0 && check_time == 0) // only on boot
                    || (scheduleItem.flags == 0x80 && scheduleItem.elapsed_min > 0 && (check_time % scheduleItem.elapsed_min == 0)))) { // periodic scheduling
                StaticJsonDocument<EMSESP_JSON_SIZE_SMALL> docin;
                StaticJsonDocument<EMSESP_JSON_SIZE_SMALL> docout; // only for commands without output
                JsonObject                                 in  = docin.to<JsonObject>();
                JsonObject                                 out = docout.to<JsonObject>();
                in["data"]                                     = scheduleItem.value;
                if (CommandRet::OK != Command::process(scheduleItem.cmd.c_str(), true, in, out)) {
                    EMSESP::logger().warning("Scheduled command %s with data %s failed", scheduleItem.cmd.c_str(), scheduleItem.value.c_str());
                } else {
                    EMSESP::logger().debug("Scheduled command %s with data %s executed", scheduleItem.cmd.c_str(), scheduleItem.value.c_str());
                }
            }
        }
    });
}
```

proddy, 2023-02-26:
I had already implemented something similar. I'll check it now so you can take a look

MichaelDvP, 2023-02-26:
BTW: it's not really need NTP active, if the RTC is set manually it's ok. And RTC could also set by thermostat clock if there is valid time from thermostat and no in EMS-ESP.

proddy, 2023-02-26:
yes, I had already removed it.

proddy, 2023-02-26:
@MichaelDvP I checked in my latest. Can you take a look? i used your code as your logical brain is smarter than mine

MichaelDvP, 2023-02-26:
The dayofweek in struct.tm starts wth sunday = 0, the flags with monday, so use :     
`uint8_t  dow        = 1 << ((tm->tm_wday + 6) % 7); // start with monday`

proddy, 2023-02-26:
i had changed the web code so it's started with Sunday. So flags 1st bit = 1 = Sunday

MichaelDvP, 2023-02-26:
- Calender commands working, timer with one minute also Other options not tested.
- With two commands and editing the first, the second disappear from the list. 
- selecting timer, there should be an option to set own time interval (e.g. 9 minutes). (it's possible in list, but complicated:  disable timer, set calender, edit time, set back to timer in list)

proddy, 2023-02-26:
> With two commands and editing the first, the second disappear from the list

I noticed that too. Is on my list to fix.

> selecting timer, there should be an option to set own time interval (e.g. 9 minutes). (it's possible in list, but complicated:  disable timer, set calender, edit time, set back to timer in list)

that's exactly why I went for that approach! By having a fixed selection it reduces a lot of the validation complexity in the web code. And I couldn't think of a use case where anyone would want something other than standard 1m or 1hr. Perhaps add 10 or 15 min then if really needed. Anything above 4hrs you can do with a weekly event.

proddy, 2023-02-26:
question: should we auto-sort the table on TIME ? Reason is, if you change an entry, it'll come up at the bottom of the list.

MichaelDvP, 2023-02-26:
imo not really important, i hope noone uses such a long list, that he did not find his own commands

proddy, 2023-02-26:
it was just one line to add, and it makes it look nicer so I add it. I think it's ok so far? I'd better stop adding things it now before I get carried away.

Do we still need to look at the commands for boot-up?

MichaelDvP, 2023-02-26:
> Do we still need to look at the commands for boot-up?

I don't know what commands these are, i have really no idea. But on startup it takes a while to have something like a heatingcircuit in thermostat registered. So this command will give error. With the minute tirgger also the command go to full realtime-minute, notdirect or 1 minute after startup. Maybe we need to add logic to retry on fail a few time in a to defined time intervall. But to define it to a usefull value, we have to know what should be send. So i think best to leave as is and wait for usage complains.

MichaelDvP, 2023-02-28:
Especially for timer it's maybe  good to have a api command to enable/disable the scheduls. (e.g. force heating off)
1. But for this we need a shown, unchanged id to use something like `api/system/schedule: {"id":"1","data":"on"}` 
2. or use id:1/0 as on/off: `api/system/schedule: {"id":"1","data":<description>}`, 
3. or use the description as command `api/schedule/<description> {"data":"on"}` (ignore empty description and doublettes).  

What do you think?

proddy, 2023-02-28:
I can't see an immediate use for enabling/disabling a schedule via an API. Is this something IOBroker would use?

In any case it's trivial to add so why not. It would mean the C code would need to generate a fixed and unique ID (uint8_t) when the Scheduler list is saved. When it finds new IDs (from the random string generator) it replaces with the new ones, which is just a sequential count.

MichaelDvP, 2023-02-28:
> I can't see an immediate use for enabling/disabling a schedule via an API. Is this something IOBroker would use?

No, i'm thinking of #951, the `force heating off` in patch branch repeats selflowtemp=0 every minute an can be done in scheduls now, but the user like to switch heating on (disable schedule) or off (enable schedule) by command.

proddy, 2023-02-28:
ok, got it. Let's add it. Need help?

MichaelDvP, 2023-02-28:
> Need help?

If we use option 1: Yes, the construction of id is confusing me, It's counted in `WebScheduler::read(` and saved to file. In `WebScheduler::update(` it's read from file as count or get from web as random 4char string and set to schedulItem. So it's different when reading from file or get it from web.  Why not count it up in web (or use lowest free number when generating)?

Option 3  can be done without showing the id to web. Also if adding a command it's usefull also to publish the state, that's more clear with publish topic `schedule_data: {<descripion>:<on/off>, ...}` and command topic `schedule/<description> {"data":<on/off>}` I'll try to go with this if you agree. By adding/removing a description, users can decide it the state is published and a command is added. For calender scheduls noone will need the publish/command.

proddy, 2023-02-28:
oh I missed that post with the 3 options. Now I understand.

The web table renderer component requires a column/key called `id` of type string. It's a real pain and the reason some a lot of the table data has this unique id in it (mainly converted from integer to string). In the current scheduler implementation, I assign an `id` from the C+ code before POSTing to the WebUI. It doesn't mean anything as long as it's unique for the table to render. 

In the new design that you're proposing we have to add the numeric ID and persist it in the C code and make sure it's unique.. And have it show in the Web table as well which can be used as a reference ID in the API calls. The IDs can be added/calculated when the Web form is saved as a sequential count.

I would leave the description as optional and whatever people want to put in it. It's more for a user comment than anything else.

hopefully that's clearer?

proddy, 2023-03-01:
@MichaelDvP I've been thinking how to implement this and keep a unique ID that the user can reference. We can't count since when scheduler items are removed it will leave a gap. I suggest adding `id` to the class/json, make it a string but call it 'name' in the Web table and have the user enter a value when creating a new entry in the dialog box. A javascript validation rule can make sure it's unique. Then this can be referenced via the API. Shall I code it?

proddy, 2023-03-01:
@MichaelDvP I have some time tonight - shall I implement it?

HerrRiebmann, 2023-03-04:
Thank you so much!
I´ve just updated to the latest dev-version and setup my scheduler:
[image]
Straight forward to ensure the name is unique and without spaces.

A little bit confusing to choose between "weekly" and "timer", but not having the day-selection in that dialog view.
But after storing it for the first time, I understood how that works!

My tiny suggestion would be to change the name "weekly" to "daily", to make this more clear.
And I think it´s possible to mess this up, when adding a weekly entry and push on "timer" on the list-view.
[image]
This would require a selection "on startup, every minute, every hour", which I haven´t selected.
Maybe disabling the weekdays or timer, based on the selection from the dialog window would make this more solid.

Just my first impression and the so called cherry on the cake. This is so much more, then I ever expected!!!
If you want me to test something, just let me know.

proddy, 2023-03-04:
Thanks for the feedback. Very valid. I'll look into
 - adding the day names to the Add dialog box
 - only being able to change the schedule (days) from the dialog box, not directly in the table list. This gives more control and prevents the issue you mentioned when switching to Timer and back
 - renaming Weekly to Daily

MichaelDvP, 2023-03-04:
Could we leave the name optional, a daily program with a lot of entries does not need commands/publishes, only the ones someone want to control via api/mqtt. Each command needs some heap. I suggest to add an explanation to the headline:
[image]
```
ems-esp-eth:# show commands
Available commands (*=do not need authorization):
 system:
  publish               publish all to MQTT *
  send                  send a telegram
  fetch                 refresh all EMS values
  restart               restart EMS-ESP
  watch                 watch incoming telegrams *
  info                  show system status *
  commands              lists all commands *

 scheduler:
  TestStartup           enable schedule item
  get_adc               enable schedule item
```

(btw. testing to use a read command as value: read adc and parse the resulting value to seltemp.)

proddy, 2023-03-04:
Ok. It means adding back ID and making it unique with the random string generator. I'll work on that this weekend

MichaelDvP, 2023-03-04:
Yes, i'm on the old random id, just renamed the description to entity-name and add mqtt-publish/command if not empty. That's working, only need t o add a check for double names (command stores the first, mqtt-publish the last due to overwriting).

proddy, 2023-03-04:
good stuff. if its ready do a PR and Publish it, then I'll refresh my branch and work on the changes to the Dialog box.

proddy, 2023-03-05:
half-done, I'll do the rest on the plane. It's a 9hr flight ;-)

proddy, 2023-03-05:
@HerrRiebmann another version is in dev with slightly more intuitive UI. Feedback welcome.

HerrRiebmann, 2023-03-06:
@proddy Perfect!
This seems much more intuitive / self explained for me.
The upgraded name was "null" for both of my entries. I don´t know if this would be an issue, using MQTT or API. New entries just have a blank name.
Translations seems fine, for german and english.

As far as I can see, the API uses `http://{{Hostname}}/rest/schedule/` rather than `http://{{Hostname}}/api/schedules/` or something. At least I could not find any working combination of api/schedules/ ScheduleName|info|commands
But I don´t see any usecase for me to read or modify those schedules via API. Just playing around with postman :)

From my point of view a pretty amazing feature! All in all, the functional range of the GUI, API, MQTT, Logs, overall performance, are really incredible! You should sell this as a template for other devices ;)

MichaelDvP, 2023-03-06:
The `/rest/` path is used by the webinterface to update the data. Api is always `/api/`. For now we have only `/api/scheduler/<name>`as command to activate/deactivate the scheduls. I'll add the `info`/`commands`/`values` for reading via api.
> But I don´t see any usecase for me to read or modify those schedules via API

As mentioned, it is only usefull to activate/deactivate the timer scheduls if it is used to force a setting like boiler off, see #951. 
It is not planed to set/read the complete schedule structure via api.

HerrRiebmann, 2023-03-06:
Does not work for me by calling the schedule:
[image]

http://_{{Hostname}}_/api/scheduler/**WorkStart**
```
{
    "message": "cannot find values for entity 'WorkStart'"
}
```

Also nothing by using the ID 01 or 02.
Or submitting the json schedule within the body, like the webinterface does it.

Sorry for bothering you with that. Just want to help you out testing that!

MichaelDvP, 2023-03-06:
Sure, the reading commands are not implemented, only the setting, if you use http-post and send the value it works.
(Or try terminal `call scheduler workstart 0` to disable)
You can check my implementation of the reads here: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

HerrRiebmann, 2023-03-06:
My bad! You are right, I haven´t changed the type from GET to POST, during my tests.

The scheduler kicked in, on time! But understandable, it failed because it does not know which entity to control:

> 2023-03-06 18:00:00.054 E 27: [emsesp] Scheduled command hc1/mode failed with error: unknown device (Error)

So I guess we need to setup the device **thermostat** or **boiler**, as well.

MichaelDvP, 2023-03-06:
Yes, you have to add the device, here are some examples for working scheduls. (working but not usefull ;-))
[image]

HerrRiebmann, 2023-03-07:
That was easy, to prefix the command with the entity!
The scheduler set the mode to "day" at exactly 7 o´clock and to "night" at 18:00. Works as desired!

Now, I´m cold and still working. But happy ;-)

closed 2023-03-11 by proddy