emsesp/EMS-ESP32 #703: Support Matrix additional to Discord

JokerGermany, 2022-10-26:
Would be great if you would let the user the choice if the want to use a free or proprietary chat protocol and support Matrix (element.io), too.

Would be cool if you you would bridge your discord Server with Matrix
https://t2bot.io/discord/

proddy, 2022-10-26:
I hadn't heard of Matrix before, had to google it and found https://joinmatrix.org/guide/matrix-vs-discord/

what's the reason behind this request?

JokerGermany, 2022-10-28:
The reason is to give the people a freedom of choice.
I don't want to use discord.

proddy, 2022-10-30:
It looks complicated to implement, for every channel? Do you have an example discord that has done this so I can compare. Anyone else wants this done?

JokerGermany, 2022-10-30:
> It looks complicated to implement, for every channel? Do you have an example discord that has done this so I can compare. 

t2bot.io bridges 10 Million Discord and Telgram channels/groups with 2 Million Discord users. ( https://matrix.org/blog/2022/08/26/this-week-in-matrix-2022-08-26 )
On bigger Group which use the group and i know is this one:
https://discord.gg/SponsorBlock

Another Discord where you 

If you need help i can assist you, then you just need to accept it in every channel.

proddy, 2022-10-30:
ok, lets do it

JokerGermany, 2022-10-30:
I am working on it:
[image]
Need to change some room permission, invite an anti-spam-bot and so on.
I will write back when i am done.

JokerGermany, 2022-11-03:
Okay I am ready.
Now you need to add this bot to your discord server:
https://discordapp.com/api/oauth2/authorize?client_id=309408702530846730&scope=bot&permissions=607251456

Then we need a appointment because someone on the discord server with enough rights has to approve the bridge within a certain timeframe.

proddy, 2022-11-06:
bot has been added

JokerGermany, 2022-11-07:
> bot has been added

okay, i tested it in the off-topic channel, welcome-and-rules and announcement.
1. We need to meet at sometime, because
>  Someone with permission to manage webhooks please reply with !matrix approve or !matrix deny in the next 5 minutes.
(See in off-topic ( BTW: message can be deleted when approved))
2. Looks like the bridge need permission to write in rooms like announcement and welcome-and-rules when this rooms should be bridged, too.
> There was a problem bridging that channel - has the guild owner approved the bridge?

JokerGermany, 2023-02-25:
This is not completed.

proddy, 2023-07-03:
because I don't know what to do next,

JokerGermany, 2023-08-26:
> okay, i tested it in the off-topic channel, welcome-and-rules and announcement.
>
>     1. We need to meet at sometime, because
>
>
>
>     2. Looks like the bridge need permission to write in rooms like [...]

This has to be done next

JokerGermany, 2023-08-30:
@MichaelDvP Is @mdvp:tchncs.de your matrix account and do you have enough rights in the discord to accept the bridge?

MichaelDvP, 2023-08-30:
> Is @mdvp:tchncs.de your matrix account 

Maybe, its a while since i created a account, but never used.

> and do you have enough rights in the discord to accept the bridge?

Don't think so, i think i'm normal user, i don't like discord and use is very rarely.

JokerGermany, 2023-08-30:
> Don't think so, i think i'm normal user, i don't like discord and use is very rarely.

Okay, thanks then i need @proddy 
Yeah I don't like discord, too
https://www.youtube.com/watch?v=uvNkdAggUGU

proddy, 2023-08-30:
> > Don't think so, i think i'm normal user, i don't like discord and use is very rarely.
> 
> Okay, thanks then i need @proddy Yeah I don't like discord, too https://www.youtube.com/watch?v=uvNkdAggUGU

I don't know Matrix, and don't trust Discord (this is just what most of the open source community seemed to use) so tell me what to do.

JokerGermany, 2023-08-31:
> I don't know Matrix, and don't trust Discord (this is just what most of the open source community seemed to use) so tell me what to do.

1. Looks like the bridge need permission to write in rooms like announcement and welcome-and-rules when this rooms should be bridged, too.

2. For the other room, when I ask to bridge the room you have 5 Minutes to approve the request with  !matrix approve
Thats the reason why we need to make an appointment to bridge the rooms.
(I made an example request in off-topic)

proddy, 2023-09-02:
we just need to agree on a date/time when I'm online so can bash that button to accept within the 5 min time window

JokerGermany, 2023-09-04:
All channels are bridged now.
For the Announcement channel I had to create an Discord Server, follow the announcement Channel and then bridge this channel to matrix... 
( https://matrix.to/#/!MBrxZRUoApYYjmyion:t2bot.io/$3oaznIZpoHdVKoj5Djc4uO1fo7SfW8hc8uZgDEbO7zw?via=t2bot.io&via=matrix.org&via=envs.net )

The github channels aren't public right now, because I need to wait for the first message from discord, give the github hook enough rights and then make it for all others read only.

Difference between Discord and Matrix:
https://matrix.to/#/#ems-esp:matrix.org contains two more public rooms:
1. [#EMS-ESP_community:gitter.im](https://matrix.to/#/#EMS-ESP_community:gitter.im)
2. [EMS-ESP32 Release](https://matrix.to/#/#ems-esp32-release:matrix.org) which only contains this RSS Feed: https://github.com/emsesp/EMS-ESP32/releases.atom

JokerGermany, 2024-01-17:
@proddy  FYI Sadly the channels are unbridged now:
https://matrix.to/#/!eTLgPMtcqcAGwtXDRF:t2bot.io/$j1Y7lxa3aRdDGl8w9gF64E8AvP8bYz2tgXTtRtN81To?via=t2bot.io&via=matrix.org&via=envs.net
> Hi @room - Unfortunately the Discord bridge deleted all of its bridge associations and cannot be recovered. Manual re-bridging is required. I've already started putting infrastructure changes in place to avoid this being an issue in the future, but cannot fix the past ☹️
> 
> If you run into issues re-bridging, please head over to 
> [t2bot.io](https://matrix.to/#/#help:t2bot.io)
> 
> A more technical description of what happened is available on the blog: https://t2bot.io/blog/2024/01/14/discord-bridge-reset/

proddy, 2024-01-17:
oh, ok, then same trick as last time then. maybe this weekend?

JokerGermany, 2024-01-29:
Sorry, don't get the message