emsesp/EMS-ESP32 #704: Error in cur_temp

ajvdw, 2022-10-27:
I'm receiving the following data through mqtt:

Message 0 received on ems-esp/thermostat_data at 8:47 AM:
{
    "hc1": {
        "seltemp": 20.5,
        "currtemp": 20.5
    }
}

but in EMS_ESP the Thermostat->hc1 current room temperature, reads 20,05
It's a Thermostat: Logamatic TC100/Moduline Easy

Sometimes the proper values come through.

proddy, 2022-10-27:
Where is the incorrect value displayed? And which version?

MichaelDvP, 2022-10-27:
@proddy This thermostat uses DIV100 for temperature, i think this issue is here in line300: https://github.com/emsesp/EMS-ESP32/blob/c3cfed5ac36f728d65917b8ac146b93ecfe52739/src/helpers.cpp#L297-L300

If you want to render to one decimal line300 should be (with rounding) 
`strlcat(result, itoa((new_value % format + format / 2) * 10 / format, s, 10), sizeof(s));`
for 2 decimals:
`strlcat(result, format > 10 ? smallitoa( s, (uint8_t)(new_value % format)) : itoa(new_value % format, s, 10), sizeof(s));`

ajvdw, 2022-10-27:
The incorrect values only appear in the mqtt message. The UI of emsesp
shows the correct values....

MichaelDvP, 2022-10-27:
Do you prefer in mqtt values with one or two decimals? The accuracy of the used ntc sensors is in range of 1 degree, decimals are only usefull to show trends. But the thermostats reports the high resolution of 0.01 degree, we can show it. I suggest to round to 0.1 degree.

ajvdw, 2022-10-27:
1 decimal is ok.

ajvdw, 2022-10-27:
Maybe the problem is not entirely clear:
The thermostat report a temperature of 20.05.
EMSESP reads the value correctly and shows it on the dashboard correctly.
But once the value is sent through MQTT it is rounded to 1 decimal, but the rounding is wrong 20.05 becomes 20.5.
20.15 also becomes 20.5

proddy, 2022-10-28:
I thought we used 1 decimal place everywhere in EMS-ESP? I'd need to check

MichaelDvP, 2022-10-28:
> But once the value is sent through MQTT it is rounded to 1 decimal, but the rounding is wrong 20.05 becomes 20.5.
20.15 also becomes 20.5

Are you sure? I think the rendering skips leading zeros of deciimals, 20.15 should be ok, but 20.07 becomes 20.7. 
If 20.15 also is wrong rendered i don't know why.
> I thought we used 1 decimal place everywhere in EMS-ESP? I'd need to check

Transformnumberfloat rounds 2 digits, renderValue(float) to precision 0-8, renderValue(int) one digit for DIV10, for DIV100 this bug. In dashboard for temperatures minimumFractionDigits is one, but no rounding to one digit.

ajvdw, 2022-10-28:
> Are you sure? I think the rendering skips leading zeros of deciimals, 20.15 should be ok, but 20.07 becomes 20.7.
> If 20.15 also is wrong rendered i don't know why.
> 

I did check some received data, you are right only x.0y is incorrectly rounded to x.y
20.15 stays 20.15 (I messed up the test code).

proddy, 2022-10-29:
merged changes, please check @ajvdw

commit 6a478eec (2022-10-29): fix #704 round div100 temperatures to one decimal
commit 76734b77 (2022-10-29): Merge pull request #706 from MichaelDvP/dev

fixes #704 and #698
closed 2022-11-01 by proddy