emsesp/EMS-ESP32 #707: IP address and MAC address sensors in auto discovery

ajvdw, 2022-10-28:
Is it possible to add one or two sensors that deliver the EMS-ESP ip address and mac address?
Similar to the "WiFi strength".
I'm using DHCP and need to login to my modem to find out the IP address that was handed out. MDNS messages are killed by the router, so accessing by hostname does not work either.

proddy, 2022-10-29:
The sensors are really used for moving values, not static like an IP/MAC address. When EMS-ESP starts it sends the data you need in a topic called `info`. You can use the MQTT platform in HA to read that?

```
ems-esp/info = {"event":"start","version":"3.5.0b7","connection":"Ethernet","hostname":"ems-esp","MAC":"A8:03:2A:62:64:CF","IPv4 address":"192.168.1.134/255.255.255.0","IPv4 gateway":"192.168.1.1","IPv4 nameserver":"192.168.1.1"}
```

ajvdw, 2022-11-07:
Is this info message sent with the retain flag set? 
On the other hand, an IP address can change when the dhcp lease time forces a renew.

closed 2022-11-10 by proddy