emsesp/EMS-ESP32 #709: Cannot upload version 3.4.4

mtbsteve, 2022-10-30:
**Bug description**
I downloaded version 3.4.4 from this github. When I try to upload the .bin file, on my EMS-ESP32, I get after a few seconds an "upload failed" error.

**Steps to reproduce**
Try to upload the .bin file

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*
EMS-ESP Version 3.4.1
Device ESP32/v3.3.5-1-g85c43024c

I previously upgraded to the v3.4.1 version a few weeks back where everything was working as expected.

proddy, 2022-10-30:
node idea why it's not working, a lot has changed in the last 6 months. I would wait until 3.5.0 is release (coming very soon) and upgrade to that version

MichaelDvP, 2022-10-31:
@mtbsteve There are some possible reasons why upload failed:
- the file you try to upload is corrupted: reload bin again and retry
- the network transfer is disturbed: retry
- full queues and not enough ram for update: make sure ems-esp is running stable for a few minutes and mqtt is working or disabled, then try again.
- You have flashed the old software with wrong partition size (e.g. esphome-flasher, tasmota-flasher) and new software does not fit in partition: flash via USB with right partition size. (look at the other issues mentioned that)

bbqkees, 2022-10-31:
If other bin files work fine, it's the bin file.
If all fail you should try uploading 3.5.0 via USB.
See here for the instructions: https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html#uploading-the-firmware-via-ems-esp-flasher-flashtool

mtbsteve, 2022-11-02:
Thanks for your help. I re-downloaded the bin file from another computer and then the update to 3.4.4 went fine.