emsesp/EMS-ESP32 #710: Buderus RC100 remote control not properly displayed

mtbsteve, 2022-10-30:
**Question**
I connected a Buderus RC100 remote controller to the EMS bus.
The RC100 is listed in the EMS-ESP dashboard device overview as Thermostat "RC100/Moduline 1000/1010" but with zero entities and values.

There should be at least 2 values, the actual room temperature and the selected target room temperature for the predefined heat circuit.

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api/system*

**Additional context**
Interestingly, when I initially connected the RC100 to the bus while it was factory-preconfigured to operate as a room thermostat and not as remote-control, I got the entities for room actual temp and room target temp listed.
I have to operate the RC100 as remote controller as required by my MC310 control unit.

proddy, 2022-10-30:
Please update to v3.5.0. Make sure you make a copy of your settings and customizations first. If you have a S32 or E32 board then consult BBQKees' instructions. The latest dev build has many fixes and changes to the thermostat remotes.

bbqkees, 2022-10-31:
Updating to 3.5.0 from 3.4.1 needs to be done via USB.
See here for the instructions: https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html#uploading-the-firmware-via-ems-esp-flasher-flashtool

proddy, 2022-11-10:
closing this due to no activity. please re-open if its still an issue

closed 2023-02-19 by proddy