emsesp/EMS-ESP32 #714: Source for EMS spec

Timmmm, 2022-10-31:
Hi, I was just wondering how you guys figured out the EMS spec? I skimmed the code and it doesn't have the feel of something that was reverse engineered. Is the spec actually available somewhere? Or do you have insider information somehow?

MichaelDvP, 2022-10-31:
Some reading stuff from people who has mainly made the reverse engineering (not complete, not in order of relevance):
https://www.mikrocontroller.net/topic/141831
https://www.mikrocontroller.net/topic/210031
https://www.mikrocontroller.net/topic/309075
https://emswiki.thefischer.net/doku.php
https://github.com/norberts1/hometop_HT3
http://www.kabza.de/MyHome/EMSBus/EMSbus.php
https://github.com/Th3M3/buderus_ems-wiki
https://domoticproject.com/ems-bus-buderus-nefit-boiler/
and
https://bbqkees-electronics.nl/wiki/
https://emsesp.github.io/docs/#/

Timmmm, 2022-10-31:
Awesome information, thanks!

closed 2023-02-19 by proddy