emsesp/EMS-ESP32 #716: Budersu GB135 wrong identified / 2 stage burner modulation and hysteresis missing

pudermueller, 2022-11-01:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/emsesp/EMS-ESP32/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api/system*

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2022-11-01:
> My boiler is identified as GB125 - but it is an GB135 with a 2 stage burner

They share the same product-id, we can rename it to "GB125/GB135/MC10)"
> In the extended settings there is also Hysterese 2 ON/OFF with 8°C/-8°C show.

Please set to log all and change the values to 7/-7 and back to 8/-8. Send the log. Then we can identify the telegram/offset of these values.
>  No telegram type handler found

These are command telegrams send from thermostat to mixer (0xAC) and boiler (0x35, 0cx1A), The values are also broadcasted by mixer(0xAB) and boiler(0x18), so there is no need to read them. @proddy Maybe we should only mention broadcasted telegrams as "No handler". (emsesp.cpp line 850: `if (!found && telegram->dest == 0)`). 

telegram 0x29 is a single value read every 10 minutes
``` 
2022-11-01 18:53:12.172 N 21: [emsesp] Thermostat(0x10) -R-> Controller(0x09), ?(0x29), length: 0x01
2022-11-01 18:53:12.191 N 22: [emsesp] Controller(0x09) -W-> Thermostat(0x10), ?(0x29), data: 6B
```
Don't know what it is, maybe button states on MC10?

proddy, 2022-11-06:
> @proddy Maybe we should only mention broadcasted telegrams as "No handler". (emsesp.cpp line 850: if (!found && telegram->dest == 0)).

this is now in b9, thanks Michael.

pudermueller, 2022-11-07:
Dear all, enclosed the LOG with changes of hysteresis2 from +8 ->+7 / -8 -> -7 and back. offset seems to be 12 and 13
[log(4).txt](https://github.com/emsesp/EMS-ESP32/files/9954585/log.4.txt)
Best regards pudermueller

MichaelDvP, 2022-11-08:
Ok, it's in the latest dev. 
For 2.stage modulation: Is there really a value for that? I expect that it's only burngas2 on and a difference in currburnpower.
In the known monitor telegrams there is no other field for this.

pudermueller, 2022-11-11:
Thx. Works fine now. I also identified the 2 stage modulation i was looking for. It is already in and just named "burner current power ", but is at the end of the day the info i need.

proddy, 2022-11-11:
can we close this @pudermueller  ?

pudermueller, 2022-11-12:
Yes thx

closed 2022-11-12 by proddy