emsesp/EMS-ESP32 #720: Support for Buderus Modul EM10

tobi-nmx, 2022-11-03:
**Is your feature request related to a problem? Please describe.**
The **Buderus EM10** module is not recognized. The log says: "Unrecognized EMS device (**deviceID 0x15, productID 243**). Please report on GitHub.". EM10 offers alarm output (relais) and 0-10V input for external burner modulation.

**Describe the solution you'd like**
Support for the EM10 device. I'd be especially interested in the events from the 0-10V input.

**Additional context**
Let me know if I can help, e.g. by providing logs/traces. Our system is still not working properly... so I'm not sure if I can trigger all messages, though.

MichaelDvP, 2022-11-03:
The EM10 is supported since #575, but there are no values from the module, it only sets the boiler paramters and listens to error messages.
But EM10 is device-id 0x12, device-ID 0x15 should be a PM10 pump module. This is unknown for now.
Please go to help page and download your support info, attach it here. 
Then go to system log and set loglevel all, wait 2 minutes, download the log, attach it here.

tobi-nmx, 2022-11-04:
Thanks for the fast response. According to the invoice I've paid for EM10, but I'll double check and add a picture of the device, in addition to the logs and support info. Unfortunaltey this has to wait a few days because I don't have remote access to the system, yet.

proddy, 2023-01-22:
in dev-16

closed 2023-01-22 by proddy