emsesp/EMS-ESP32 #727: dhw circulation pump mode (setting "off" or "continuous" always result in "1x3min")

W0w3, 2022-11-06:
EMS-ESP Version: v3.5.0b9
Interface: BBQKees Gateway S32
Boiler: Buderus Logamax plus GB152T

I have the issue that the setting for the warm water circulation pump is not possible to be set to "continuous" or "off". 
Both settings always result in "1x3min".

The issue persists both in the release version 3.4 as well as in the latest beta.

proddy, 2022-11-06:
Can you get us some more information like the debug trace log when you set the value

https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-on-an-ems-device-doesn39t-work

W0w3, 2022-11-06:
Hi proddy,
thank you for your help! :)

Here is the log:
[image]

MichaelDvP, 2022-11-07:
Indeed, there is an issue, the command only checks for numbers 0-7, not for texts, `call boiler wwcircmode 7` set to continuous. Option 0-off does not work, valid values are only 1-7. 
I'll change the option to enum type to accept text options (enum starts always at 0, so it is 0- 1x3min, 6 - continuous then).

W0w3, 2022-11-08:
I installed the latest beta and can confirm the issue is solved as described by fixing continuous setting and removing the off setting. 
Thank you 👍

closed 2023-02-19 by proddy

matzelkoenig, 2025-10-25:
EMS-ESP Version: v3.7.1
Interface: BBQKees Gateway S32
Device: Bosch XCU_THH/CS*800i, Logatherm WLW* (Boiler/HP)

For my Bosch Compress CS6800iAW 12 MB it is also not working to switch it totaly "off". It always results in "1x3min". The mode "continues" does work and also the other "x3min" modes.

proddy, 2025-10-25:
> EMS-ESP Version: v3.7.1 Interface: BBQKees Gateway S32 Device: Bosch XCU_THH/CS_800i, Logatherm WLW_ (Boiler/HP)
> 
> For my Bosch Compress CS6800iAW 12 MB it is also not working to switch it totaly "off". It always results in "1x3min". The mode "continues" does work and also the other "x3min" modes. The Software menu of my device only offers the values 1-6 and continues. So I assume "off" is generally not possible but EMS-ESP is still showing it in my version v3.7.1

I would upgrade to the latest dev release (3.7.3) and try again. It may be resolved, otherwise we'll need to investigate further. Version 3.7.1 is a year old.

matzelkoenig, 2025-10-25:
Thank's for your hint! I am a little bit cautious about updates. The last time my BBQKees crashed and I had to reinstall via cable. I will give it a try again 😉

commit 723662a6 (2022-11-07): fix #727 boiler wwcircmode
commit b91e4743 (2022-11-07): Merge pull request #734 from MichaelDvP/dev

fix #727 boiler wwcircmode