emsesp/EMS-ESP32 #731: Buderus GB125 missing values

Marcwa19197, 2022-11-06:
Hello,

first of all: Thank you for the great product!
I installed it a few days ago and im able to get some data using MQTT, Telegraf and Grafana with InfluxDB.

But im missing a few values. I have a Buderus GB125 with Logano LT160 as water storage.

Im missing the following values, which should be available in the MQTT Topic "boiler_data":
- reTemp (Im able to see the curFlowTemp but not the Returning Temp)
- sysPress
- valveStatus (i have one Heating Cycle, so i subscribed to topic "mixer_data_hc1" and "mixer_data_ww1"

I subscribed to the following topics via telegraf.conf:
```
"ems-esp/shower_data",
"ems-esp/mixer_data_wwc1",
"ems-esp/mixer_data_hc1",
"ems-esp/thermostat_data",
"ems-esp/boiler_data_info",
"ems-esp/boiler_data_ww",
"ems-esp/boiler_data",
"ems-esp/heating_active",
"ems-esp/tapwater_active",
"ems-esp/heartbeat",

```

proddy, 2022-11-06:
Firstly we need some system information (see https://emsesp.github.io/docs/#/Support?id=creating-an-new-request). Can you include that please? Also always upgrade to the latest version (now 3.5.0b9) first as perhaps some of the issues have been already resolved.

When you say missing, missing from what? Can you see these data points somewhere else we can use for reference?

The retTemp (return temperature) should be there if your boiler supports it, but I see it's missing so we'll need some telegrams to double-check. But first do the firmware upgrade. Same with SysPress (system pressure). 

PumpStatus is part of the Mixer device and not the Boiler but I didn't see that listed above?

What is valveStatus ?

Marcwa19197, 2022-11-06:
I updated my first comment and updated to the latest version.

With missing i mean im not able to fetch the date via telegraf. So i also guess mqtt is not providing it.
Or is there a way to check the values directly on the ems-esp via telnet?

If i run "show mqtt" on the ems-esp i see the following:
```
MQTT is connected
MQTT publish errors: 0

MQTT topic subscriptions:
 ems-esp/system/#
 ems-esp/dallasssensor/#
 ems-esp/analogsensor/#
 ems-esp/boiler/#
 ems-esp/thermostat/#

MQTT queue is empty
```

So after the upgrade still no retTemp or SysPress available.
Yes, pumpStatus and valveStatus is provided by the mixer. But also im not able to see this data.
(Im trying to get the Grafana Dashboard working exactly as in the example).

Thanks.

proddy, 2022-11-06:
I really need the support details. In the Web UI go to help and read the last bullet. I think your mixer is not being detected.

If you want to see what's being sent to your MQTT broker installed MQTTExplorer

Marcwa19197, 2022-11-06:
I uploaded the info in the first post. thanks. Will check MQTTExplorer.

proddy, 2022-11-06:
sorry, missed that. Is the LT160 a Mixer? The web and console both show all the data. If it's not there, it wouldn't be in the MQTT messages either.

Marcwa19197, 2022-11-06:
No, LT160 is only the warmwater storage. I checked the it right now in the basement, but im not able to tell you what it is....
There is no label at all.
[image]

MichaelDvP, 2022-11-07:
I have nearly the same system (GB125, MC10, RC35, MM10, SM10).
>    reTemp (Im able to see the curFlowTemp but not the Returning Temp)

The GB125 does not have sensor for returntemp, just connect a dallassensor. 
>    sysPress

Also no sensor, i use a analog pressure sensor. Make sure the analog signal does not exceed 1.5V.
>    pumpStatus

What pump? You have heattingpump and dhw charge pump, both are listed.
>    valveStatus (i have one Heating Cycle, so i subscribed to topic "mixer_data_hc1" and "mixer_data_ww1")

What valve? You do not have a mixer and dhw type is chargepump, so no 3-way-valve.
The picture shows the housing of heatingpump/header. Same housing is used for mixers, but with the lower window opend to see the mixing position.

Marcwa19197, 2022-11-07:
Thank you for the information.
I also have a analog pressure sensor on the left side. What do you mean with 1.5V? And what do you meant exactly with an dallassensor? How can i install one on the system? I think at least the ReturnTemperature would be great to monitor.

Thank you for the explaination regarding the valveStatus! Its clear to me now! :-)

MichaelDvP, 2022-11-07:
Do you have a electronic sensor or a gauge? 
I meant a cheap sensor like [this](https://www.ebay.de/itm/313886871476?hash=item49151d8fb4:g:9ikAAOSwHa1iFQXu&amdata=enc%3AAQAHAAAA4HsxU36%2Bz0F3AYKGionSUNWSsGHlzCpON4j64dJLT4cDhdqTJK8y%2Fo7wr2Nj4%2BCUl5C2yi%2By2gs8DCUOyahiFb3q8L9nFzoZUza0EnRBVNzS3SkajKx65cWXYdJZ4O8oK9Gz7IXiekuZADk%2B9wkdZnFHQD0zhrO%2Buh88KwQkP9qd%2Fu7wItfKzAvTdUPPHggJEaZQdWeWZuo5%2BUODuyONc5iHySeTyeaBdOWkaMCw9wiEUX1JNzvPLD4Rr0ukY7h3LKQ4PS6z2YkY1tjU62V58C5dRiVLscWmTXOlxMWT83rp%7Ctkp%3ABFBM6unHloph) avaible with different ranges 0.8/1.0/1.2 MPa. 5V can taken from the gateway, but 4.5V full range is to much for the esp.  You can selsct the sensor for your expected full range and use a voltage divider 3:1. Or us a sensor with higher range connected directly. With normal pressure in heating of 1,5-2 bar a 1MPa sensor is ok, the output is 1.5 V at 2,5 bar pressure.
Dallassensor means [DS18B20](https://www.ebay.de/itm/255283194278?hash=item3b701069a6:g:dcEAAOSwXatgskBd&amdata=enc%3AAQAHAAAA4F7RZ3FQ1bhHIdTmghvjDmXs0KCW3LJEhq9yu9vmSljCEgIjVDio%2F1udOP7GNNxeLGzX%2Bh8lGOj%2B2hrsvcqfpur%2FAOqp%2FnQJ1qNfz6QIF79c8sOMQ27M5Ts94kI0gHGogNhWFb4%2Bl20dKdiIOG6bXF4%2Fq4cWnjZ8U4CyMlhxSZXnP7NTZircTMbe1rFYGvmv1K5x%2B91rsqp94%2BUdcLkKUzzPBIGD7NL7o7NnC4sa9Iy%2BvMUuoVnjtsWAjacEwO7%2Fowe2Agn%2FNhmd9%2FySkkXPHRcHazqD1%2BYrorvPSudFFuhm%7Ctkp%3ABk9SR8KRtJeKYQ), that can be connected directly to the gateway.

Marcwa19197, 2022-11-07:
I have only a gauge, nothing electronic. But i think i will let this as it is. (See attached picture)
And the Dallassensor should be connected to the gateway and directly to the heat cycle return pipe?
[image]

pudermueller, 2022-11-11:
> No, LT160 is only the warmwater storage. I checked the it right now in the basement, but im not able to tell you what it is.... There is no label at all. [image]

This is the mixer of your heating circuit. This includes also your heat circuit pump you are missing in MQTT. 

Concerning MQTT you should sniff with MQTT Explorer. It should show you a "ems-esp/mixer_data_hc2" <-- 2 and not 1. My GB135 has the heating circuit 1 only for a configuration without mixer and separate pump. If you have a mixer this can in RC35 only be linked to HC2. Same is valid for the valvestatus (also in mixer_data_hc2). 

Regards
Pudermueller

Marcwa19197, 2022-11-13:
Hi @pudermueller ,
thanks for the reply.

I sniffed with MQTT Explorer and im not able to see something mixer related. Im a bit confused now. Is it because of missing hardware? And if so, what im missing? I guess it woul be better to have the returntemp available, so the GB125 "knows" how much it must reheat the heating cycle, or? 

[image]

pudermueller, 2022-11-15:
Hi @Marcwa19197 , 
@MichaelDvP  already identified correctly that your unit is not a mixer unit. You can remove the cover by opening the black plastic screws and lift it. For comparison you might check [Buderus flyer](https://productsde.buderus.com/buderus/productsde.buderus.com/broschueren/broschuere-buderus-heizkreisset-25-6-msl-de-052020.pdf). 
Coming back to your original request: you are able to see the pump status (heating circuit and warm water production) 
and for return temperature you need an additional dallassensor. 
Best regards Pudermueller

closed 2023-02-19 by proddy