emsesp/EMS-ESP32 #732: Thermostat Moduline 400 Holiday setting by MQTT

FireWizard52, 2022-11-06:
Hello all,

I own a Nefit Moduline 400 thermostat, in which I want to set the holiday (or vacation) period by MQTT.

I see several posts, where is stated that Nefit Moduline 400 is equal to Buderus RC35. However I saw also some that stated that Moduline 400 is equal to RC30.

1. Question 1: What is true and is it 100 % equal or are there minor difference?
2. Question 2: Does Moduline 400 support holiday setting by MQTT
3. Question 3: If so, what is the topic and what is the payload.

I already tested the topic ems-esp/thermostat with the payload {"cmd": "holiday", "data": 03.03.2023-20.03.2023}
A check on the thermostat itself learned that there was no holiday period configured.

Who can shine a light on this issue?

Thank in advance,

Regards

MichaelDvP, 2022-11-07:
Please add support info from help page and from dashboard export (button) the thermostat values and attach here.
Modulline  400 is not equal to RC35 and RC30, simlilar, there are some differences. 
The customization page should show the holiday entity, is there a value?

FireWizard52, 2022-11-07:
Hi, @MichaelDvP 

Thanks very much for your quick response.

You answered:

> Modulline 400 is not equal to RC35 and RC30, simlilar, there are some differences.

I had already that feeling, but was unsure. Thanks for the confirmation

> Please add support info from help page and from dashboard export (button) the thermostat values

Not clear to me, what you exactly mean (support info from help page) 

Dashboard export (button), I'm not sure what you exactly mean, but I assume:

`{
  "System": {
    "version": "2.2.2",
    "uptime": "000+13:34:23.309",
    "freemem": 44,
    "fragmem": 21
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 32130,
    "#read requests sent": 8165,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Moduline 400 (DeviceID:0x10, ProductID:78, Version:03.07)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x41 0xA7"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.04)",
      "handlers": ""
    }
  ]
}

> The customization page should show the holiday entity, is there a value?

Not sure, what you mean with "customization page" and "holiday entity" 

Below a screenshot of the Desktop:

[image]

I have to add, that I still use version 2.2.2. (so the old ESP 8286), but the Github page for this version is Read Only.
In my opinion the fact, whether Moduline 400 support holiday setting by MQTT, or not, is independent of the used version.
At least, I saw a thread in the ESP8286 issue, that reported success. (https://github.com/emsesp/EMS-ESP/issues/671)

Regards

MichaelDvP, 2022-11-07:
Huh, v2 is very old and there is much added and fixed in v3, also holiday setting be there. 
You should update to v3,xx A compatibe esp32 is not expensive. 
Plugin replacements for the 8266 D1-mini is MH-ET32 [(see here)](https://www.ebay.de/itm/234630354380?hash=item36a10f15cc:g:JZwAAOSwN35i120z&amdata=enc%3AAQAHAAAAoMnx9WFviTslnDSbxwDvZ%2BRp3Ps3RNyFpwUrnvesV1xJF%2Bk60bfimAUWY1k5w5PlXoKgVdceidlVdA4qTv3xt8wz1Ic3SK9TJFBjoGMcvP2hm9XKXOIC71THlIaj7Go6zyHws8jj4%2Fdv23t0Oh9hKrxgXZAMEkRO0%2FSA2B7qs4wIeq%2FGuT6LG07J4EHJkfjBetSpnVrH%2BUNFn1lybU%2Bd1pA%3D%7Ctkp%3ABk9SR9rPsZ6KYQ). 
ESP32 does not work with buspower, only with jack-power or an usb supply.
If you have no other power option than buspower, you can use the Lolin C3 mini, but there is no precompiled binary yet.

FireWizard52, 2022-11-07:
Hello @MichaelDvP 

You said:

> v2 is very old

I bought this unit from BBQKees, March 2020 and in that time I even used some version 1.9xb.
Of course, I read that you moved early 2021 to the ESP32 platform, but I do not consider equipment of 2.5 years old  as "very old".
I still use Raspberry Pi's, such as model 3, that has been introduced February 2016!

However thanks for the link to the MH-ET32. I will surely consider to upgrade to the new platform.
I have powered it through the service jack, so this should not be an issue

But

As I understood from the link the holiday mode worked for a RC35 thermostat.
I tried it, and it did not work for a Moduline 400, unless I did something wrong.
So therefore my question: 
Does somebody know, whether "holiday" mode is supported by the Moduline 400?

If it is unsupported, upgrading to the new ESP32 will not solve it, in spite of the fact that in v3, holiday settings were implemented.

Or am I wrong?

MichaelDvP, 2022-11-07:
You can not try a rc35 command without a rc35 connected. 
"Old" does not mean time or hardware, the esp8266 is simply to limited for the software enhancements in v3. 
There are ~1000 commits since v2.2 with a lot of enhancements and fixes.
For Moduline 400 enhancements see #449 and #467

closed 2023-02-19 by proddy