emsesp/EMS-ESP32 #738: Heatingtemp and Selflowtemp staying on 30 degrees

MarkRoks, 2022-11-09:
Hello,

When I send a value of 5 degrees to "HeatingTemp" via the EMS-ESP gateway (v3.4.4), it jumps and the "SelFlowTemp" to 30 degrees.

This means that the boiler, at a sufficiently low supply temperature, continues to switch on to heat the central heating water until it exceeds 30 degrees and switches off again. This could be overcome by sending the "HeatingTemp" to 5 degrees.

Is this a known issue?

I had already read something at: https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-works-at-first-but-is-then-reset-to-it39s-original-value

But this is not a cause and/or solution. I tried this, but both the "HeatingTemp" and the "SelFlowTemp" remain at 30 degrees. With a boiler that switches on/off every time.

Do I need to change something in the settings of the EMS-ESP gateway, or change something elsewhere?

I like to hear it.

Regards,

glitter-ball, 2022-11-12:
Your thermostat may be overriding the change. What boiler and thermostat are you using?

proddy, 2022-11-12:
also need your system details. https://emsesp.github.io/docs/#/Support

MarkRoks, 2022-11-14:
> Your thermostat may be overriding the change. What boiler and thermostat are you using?

Hallo Glitter-Ball,

I'm using a Nefit Trendline HRC30 CW5 boiler and a Nefit Easy thermostat.

MarkRoks, 2022-11-14:
> also need your system details. https://emsesp.github.io/docs/#/Support

Hello Proddy,

It seems that the adjustment of the SelFlowTemp and HeatingTemp arise from the Nefit Easy Thermostat. This can be overcome by sending the Heating Activated to "off" via the EMS-ESP gateway, then the HeatingTemp will remain at 30 degrees, but the SelFlowTemp will go to 5 degrees.

This prevents the boiler from switching on unnecessarily. Otherwise, he commutes all day long.

I don't know if this is beneficial for the boiler, just to set it Heating Activated to "off" for a certain period. But it does solve the problem for the time being.

proddy, 2022-12-09:
thanks, I'll add this to the wiki

closed 2023-02-19 by proddy