emsesp/EMS-ESP32 #747: Weird things since upgrade to 3.4.4

glitter-ball, 2022-11-14:
[Sorry - don't get the issue template and can't assign any tags, for some reason. Hopefully someone can add for me please?]

I'm seeing two weird things since I upgraded to 3.4.4...

- NTP is disconnecting and reconnecting regularly, shown both in the system log and in HA
[image]
[image]
[image]
The gap in the bottom image on Saturday is when I did the upgrade, so the regular disconnects are only happening since then.

- Two new entities have appeared in HA...
[image]
[image]
This is curious, as I don't have a heat pump! There are no MQTT Explorer entries with the string 'heatpump' and no entities in HA with the string 'heatpump' either. So where the heck are these coming from? The logs include...
```
2022-11-14 07:51:37.758 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'summermode' when rendering '{{value_json.hc1.summermode}}'
2022-11-14 08:00:19.147 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'hpoperatingstate' when rendering '{{value_json.hc1.hpoperatingstate}}'
2022-11-14 08:05:48.332 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'summermode' when rendering '{{value_json.hc1.summermode}}'
2022-11-14 08:16:11.731 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'hpoperatingstate' when rendering '{{value_json.hc1.hpoperatingstate}}'
```

MichaelDvP, 2022-11-14:
That's a timeout after 2 hours, the new arduino framework syncs rarely, before it was every hour. I thought it is fixed in v3.4.4, but seems only in v3.5.0 here: https://github.com/emsesp/EMS-ESP32/blob/06004ce47820a1f616a6a023cb4278a8cf9b3e51/lib/framework/NTPSettingsService.cpp#L53
Just upgrade to 3.5, it will be released next time.

glitter-ball, 2022-11-14:
Why is EMS-ESP suddenly transmitting heat pump data points when I don't have one?? It's only done this since the upgrade the 3.4.4. but why? I'll disable discovery for 5 minutes and see what happens.

MichaelDvP, 2022-11-15:
> Why is EMS-ESP suddenly transmitting heat pump data points when I don't have one?

Don't know, something in the logic we have since #503 and #550 seems not to work on your system, 
Maybe summermode and hpoperatingstate shows up alternating.
Send bus-reply of `read 10 2A5` and `read 10 471`
And try a `watch 2A5` and wait for the next ha complain.

MichaelDvP, 2022-11-15:
I expect that with boilers the 471 telegram is empty. Than this fix will help: https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

glitter-ball, 2022-11-15:
I’ll try that later. Also boiler name (Worcester Greenstar ErP) no longer appears in device info. Can’t remember if this is different. I’ll do captures later.

glitter-ball, 2022-11-15:
Putty log file attached.  
[putty.log](https://github.com/emsesp/EMS-ESP32/files/10016336/putty.log)

Telegram 471 is empty in my system, as you suspected.

Watching 10 2A5 and waiting for HA is tricky. In 6 hours, there were only 6 log book entries for summermode and hpoperatingstate, so difficult to capture appropriate packets when HA complains.

glitter-ball, 2022-11-15:
My thermostat is Worcester Sense II which is identical to CW400, so other people with boilers may be having this problem too but just haven't spotted it yet.

```
ems-esp:$ show devices
These EMS devices are currently active:

Boiler: Worcester Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08, ProductID:208, Version:01.01)
 Received telegram type IDs: 0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A
 Fetched telegram type IDs: 0x14 0x16 0x33 0x26
 Pending telegram type IDs: 0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA
 Ignored telegram type IDs:

Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.03)
 Received telegram type IDs: 0x06 0xA2 0x2BA 0x2BB 0x2BC 0x2BD 0x2BE 0x2BF 0x2C0 0x31D 0x267
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x2F5 0x31B 0x23A 0x240
 Pending telegram type IDs: 0xA3 0x12 0x471 0x2A6 0x2B0 0x29C 0x472 0x2A7 0x2B1 0x29D 0x473 0x2A8 0x2B2 0x29E 0x474 0x2A9 0x2B3 0x29F 0x475 0x2AA 0x2B4 0x2A0 0x476 0x2AB 0x2B5 0x2A1 0x477 0x2AC 0x2B6 0x2A2 0x478 0x2CC 0x2CE 0x2D0 0x2D2 0x31E
 Ignored telegram type IDs: 0x1A 0x35 0xBF

Controller: ErP (DeviceID:0x09, ProductID:209, Version:01.03)
```

Odd that the boiler type isn't decoded any more.

MichaelDvP, 2022-11-16:
> Telegram 471 is empty in my system, as you suspected.

Ok, than the issue is solved in the above linked dev. Please check.
> Odd that the boiler type isn't decoded any more.

There was no change in database. But i'll rename this product ID to `Logamax Plus/GB192/Condens GC9000/Greenstar ErP`

commit 1f43bb20 (2022-11-18): Merge pull request #753 from MichaelDvP/dev

fixes #747, #748
commit a53d54a1 (2022-11-18): don't use hpoperatingmode on boilers, fix #747
closed 2022-11-26 by proddy