emsesp/EMS-ESP32 #748: Compressor Activity is empty during defrost run

HansRemmerswaal, 2022-11-15:
The compressor activity normally has the values none, heating, hot water, (and I guess cooling).
But I start to see examples where the compressor activity is empty. When I check it looks like it's always at the same time that there is a defrost run. For 2 min my circulation pump speed is becoming 100% and I see the temperature dropping.

Is this value retrieved from the EMS bus, so a software issue in the firmware of my Nefit heat pump? Or is this a values determined in the EMS-ESP32 and needs to be reported as an issue?

[image]

[image]

Devices I use:
BBQKees Gateway E32 with firmware 3.5
Nefit Eviline A/W Split 7.0 E-S

MichaelDvP, 2022-11-15:
I think this is a 6th option in hpactivity. If it happens again try in terminal `read 8 48D` or set in web system/log to all, wait until message 48D is received, save the log, post it here.
You can also test https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest, there i have added a option to hpactivity (5=noforst), but if the value goes to 6, 7, .. you still get empty text.

HansRemmerswaal, 2022-11-15:
I have set in web system/log to all. Will post the log as soon as I see the issue again.

HansRemmerswaal, 2022-11-15:
That system log is not very big, only +/- 200 lines. So I need to react within minutes when it happens...? I put the buffer size to max.

MichaelDvP, 2022-11-15:
As long as you leave the log page open the log is only limited by the memory the browser can allocate. Buffer is emsesp internal and only used if you switch to a different emsesp-page, this is limited to max. 100 lines.
But it's easier to use syslog or telnet with logging to file (like putty) and in telnet commands `set timeout 0` and `watch 8 48d`
And/or check with my lates commit metioned above if your compressor activity log shows empty or "defrost".

HansRemmerswaal, 2022-11-16:
Yesterday I have installed v3.5.0b9, today I had the issue again. See screenshots. So the latest development version didn't solve the problem.

Since I installed the v3.5.0b9 my initial language is each time PL, but ok that is easy to change to EN each time. An anyhow I'm using Home Assistent at the end to read all the data. More problematic is that when I click on the system log button that my screen becomes completely empty. That was not the case before. I tried with Chrome and Edge and both show the same. 

So I was not able to capture the log file.

I'm now installing Putty and will try your alternative log solution.

The issue with viewing the log file using EMS-ESP, any idea what that can be? My previous installed version was 3.50.b7 (last screenshot).

[image]

[image]

[image]

[image]

[image]

[image]

MichaelDvP, 2022-11-16:
> Yesterday I have installed v3.5.0b9, today I had the issue again

From my link or from here?
> Since I installed the v3.5.0b9 my initial language is each time PL, but ok that is easy to change to EN each time. 

Yes, there were some discussions about that. Setting the base to PL saves a bit memory.
> More problematic is that when I click on the system log button that my screen becomes completely empty

That's strange, could you please go to help page, download the support info and attach it here.

HansRemmerswaal, 2022-11-16:
Not sure if the support info is useful... it's empty.

Let me download v3.5.0b9 again via your link and install. 
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10019015/emsesp_info.txt)

HansRemmerswaal, 2022-11-16:
after installing the latest v3.5.0b9 I was able to get a more meaningfull support file
[emsesp_info (2).txt](https://github.com/emsesp/EMS-ESP32/files/10019088/emsesp_info.2.txt)

MichaelDvP, 2022-11-16:
Thanks. 
The free heap is a bit low, this could make some problems, like the empty info.
Please set system log buffer to lowest setting (25) and disable OTA, this saves some ram.
(if log-page does not load, restart em-esp and access it before buffer is filled. Also after reducing buffer a restart help to free the memory)
The E32 need some ram for the interface, with wifi enabled it is a bit more.

HansRemmerswaal, 2022-11-16:
I read you message to late and did already a factory reset. So I took the long way but at the end the result is the same, buffer is reduced and data is displayed again. Maybe an idea to limited the size of the log file in case there is not enough memory anymore for the basic functions.

Your latest build is installed, putty is running with telnet commands set timeout 0 and watch 8 48d.
So waiting now till it happens again, typically in the morning when it's cold.
I keep you updated.

HansRemmerswaal, 2022-11-17:
It happend again, it looks like a defrost to me. I hope you can find the reason the the log.

[putty3.txt](https://github.com/emsesp/EMS-ESP32/files/10028590/putty3.txt)

[image]

[image]

MichaelDvP, 2022-11-17:
Thanks, yes the activity state goes to 6, i'll name state 5 to `unknown`. Can you check your hp-manual if there is another state? Maybe disinfecting? I'll update my github.

HansRemmerswaal, 2022-11-17:
> Can you check your hp-manual if there is another state? Maybe disinfecting?

Sorry, what do you mean? Where do I need to check?
Disinfecting is Saturday at 12:00. Currently that is shown as "hot water"

HansRemmerswaal, 2022-11-17:
There is a "pool" option. The heatpump can be used to heat a swimming pool. Maybe that is 5?

HansRemmerswaal, 2022-11-17:
I do not have a pool, but maybe this is related to that compressor status 5.

[image]

MichaelDvP, 2022-11-17:
Pool is already in as option 4 We have now as actions:
0: none, 1:heating, 2:cooling, 3:hot_water, 4:pool, 5:unknown, 6:defrost

HansRemmerswaal, 2022-11-18:
It's working 👍 
Thanks for making the software everyday a bit better!

[image]

closed 2022-11-18 by HansRemmerswaal