emsesp/EMS-ESP32 #751: Change EMS-ESP device sensor types in Home Assistant

laszlojakab, 2022-11-16:
**Is your feature request related to a problem? Please describe.**
I'm using EMS-ESP32 with Home Assistant. My problem is that some of EMS-ESP device sensors have "wrong type". I know I could use template sensors to fix the problem but it would nice if EMS-ESP could register these sensors with the "correct type". The following sensors are problematic:

- `status`
- `ntp_status`
- `uptime`

**Describe the solution you'd like**
I think the following modifications needed:
- `status`: it should be registered as a `binary_sensor` with `connectivity` [device class](https://www.home-assistant.io/integrations/binary_sensor/#device-class) and not as a `sensor`. By doing this Home Assistant will localize the value to `connected` or `disconnected` or whatever based on the selected language. Also have to mention that Home Assistant dashboard can change color of binary sensor's icon natively based on it state. With the current string value `connected` it is only possible if I introduce a template binary sensor or if I'm using a third party ui component.
- `ntp_status`: same as `status` sensor.
- `uptime`: the natively supported [uptime sensor](https://www.home-assistant.io/integrations/uptime/) in Home Assistant returns the `datetime` of booting in ISO 8601 format (eg: `2022-11-15 20:46:38.049117`). If NTP is enabled EMS-ESP could also provide this value. The [device class](https://www.home-assistant.io/integrations/sensor/#device-class) of the sensor should be `datetime`. By this modification Home Assistant can natively convert the elapsed time to more easily readable text on the dashboards (for example: `2 days ago`). One more note: the devices in Home Assistant which reports uptime and has an RTC or NTP connection reports the booting time as uptime.

proddy, 2022-11-17:
Good points. Which version are you using?

laszlojakab, 2022-11-17:
v3.4.4

proddy, 2022-11-18:
regarding the 'uptime' ISO 8601 is not good for showing the elapsed time. It's better to add it to the MQTT info topic which looks like:

proddy, 2022-11-19:
and `status` is used as a last will message (https://www.home-assistant.io/integrations/mqtt/#birth-and-last-will-messages). Do you want to see if the EMS-ESP is online or just that it's connected to the EMS bus?

proddy, 2022-11-20:
I made the changes to status. For uptime, it needs more thinking and some help from your end

laszlojakab, 2022-11-23:
Thanks for the updates! What do you need from me about uptime?

proddy, 2022-11-26:
> Thanks for the updates! What do you need from me about uptime?

see my earlier response. uptime is elapsed time and shouldn't change. I think you're after the time when the system was started so HA can calculate the uptime (e.g. "2 days ago")

laszlojakab, 2022-11-26:
You are right it is enough to set it in birth mqtt message.

proddy, 2022-11-28:
added boot_time

closed 2022-11-28 by proddy