emsesp/EMS-ESP32 #754: Default webUI langauge to local of browser

proddy, 2022-11-18:
@HansRemmerswaal in https://github.com/emsesp/EMS-ESP32/issues/748#issuecomment-1316413110_ mentioned:
        
"Since I installed the v3.5.0b9 my initial language is each time PL, but ok that is easy to change to EN each time. "

I think this is because of the base language has to be PL (long story). What we can do is auto detect the language from the browser if there is no cookie set.

MichaelDvP, 2022-11-18:
It's also possible to change after autogeneration of i18n files in `i18n-util.ts` the `baseLocale: Locales = 'en'`
But auto detect the browser language is much better.

MichaelDvP, 2022-11-24:
@proddy I have tested my workaround for local compile (build_interfece.py) and GH-action. Works, see: https://github.com/MichaelDvP/EMS-ESP32/commit/88135d27509c159cb24ff184fc99329fc8da6545
But browser detection will be better.

proddy, 2022-11-25:
that's a good solution. Always surprised on how powerful python can be. I'll play around with the browser locale auto detect this weekend

proddy, 2022-11-27:
I tried with adding

```ts
  import { detectLocale, navigatorDetector } from 'typesafe-i18n/detectors';
  const fallbackLocale = 'en';
  const availableLocales = ['nl', 'de', 'en', 'se', 'pl', 'no'];
  const detectedLocale = detectLocale(fallbackLocale, availableLocales, navigatorDetector);
```

but couldn't find a way to override any previous browser setting

@MichaelDvP I think your solution is better for now

proddy, 2022-11-29:
@MichaelDvP If you can add your workaround I think we can get v3.5 out....

MichaelDvP, 2022-11-30:
I've already added it to PR #778 after your:
> I think your solution is better for now

But for now the GH action is only for pre_release, not integrated in tagged_release.

proddy, 2022-11-30:
then we should add it too, until I can get the browser detection working correctly.

closed 2022-12-05 by proddy