emsesp/EMS-ESP32 #756: Scheduled reboot facility

jefft4, 2022-11-20:
Would it be possible to add an automatic scheduled reboot to EMS-ESP?

My smart home controller reboots once each week overnight, restarting the MQTT broker with it.  My EMS-ESP sometimes fails to re-establish the broker connection after this and sits there doing nothing until I wake up and see the alerts from the controller than EMS hasn't reported in for a few hours.  There are also (rarely) problems with it reconnecting EMS, which a reboot soon fixes.

A scheduled reboot option would mean that any temporary problems with EMS-ESP would have a good chance of being resolved without intervention.

proddy, 2022-11-20:
Which version are you on? And are you ipv4 or 6? We made some improvements to the MQTT reconnection in v3.5.x. You can also reboot EMS-ESP remotely with an API call via an external scheduler (eg. in HA or ioBroker)

jefft4, 2022-11-20:
3.4.4 at present, IPv4 network.  I'll try 3.5 when it's out of beta, sounds positive.
Remote reboot is an interesting point; will look into that.  It wouldn't help if the ESP lost wi-fi though.  I like things to do a local reboot to clean up periodically.

proddy, 2022-11-20:
This would be easy to do in #701 - which we're planning to work on after v3.5 is out (before xmas hopefully)

jefft4, 2022-11-20:
:-)

tefracky, 2022-12-19:
> Which version are you on? And are you ipv4 or 6? We made some improvements to the MQTT reconnection in v3.5.x. You can also reboot EMS-ESP remotely with an API call via an external scheduler (eg. in HA or ioBroker)

This is a nice idea. What is the right API-call for a reboot?

proddy, 2022-12-19:
It's in the doc. Search for restart

proddy, 2023-02-24:
closing. will be addressed in #701

closed 2023-02-24 by proddy