emsesp/EMS-ESP32 #758: Entity id for sensors within HA changes when updating name in web-ui

tp1de, 2022-11-20:
I recognized that while changing names for sensors (analog & dallas) within web-ui the entity id changes as well for HA.
I think only full name should change. Entity id should be kept.

proddy, 2022-11-20:
I did a quick test with Analog sensors, and you're correct, when you create one (say GPIO 31, name is Analog1) it will create the HA entity called `sensor.analog_sensor_analog1`. I think it's best to use the GPIO and not the name, so `sensor.analog_sensor_31`

tp1de, 2022-11-20:
I think it is important to keep the unique id in ha when updating names within ems-esp to keep history in statistics as well the dashboard configuration. I haven't recognized issues with dashes within unique id. I would keep it. For analog sensor I would recommend to add gpio to the name ... e.g.  sensor.analog_gpio_31  .... two times "sensor" ist not needed.

tp1de, 2022-11-20:
... and for dallas the "value" in api seems to be rounded to full integer.

proddy, 2022-11-20:
> ... and for dallas the "value" in api seems to be rounded to full integer.

can you create a new issue for this please. It's a new bug, and not related to HA

tp1de, 2022-11-20:
... the whole issue is related to ha as well to api. Api responses for dallas and analog sensors are different anyhow.

MichaelDvP, 2022-11-20:
Yes dallas and anlog api is not in line with devices. 
I suggest to use the info/values command only for a list
```
call dallassensor values
{
  "cylinder_top": 45.3,
  "solar_flow": 23.2,
  "cylinder_middle": 33.9,
  "cylinder_bottom": 21.8,
  "circ_backflow": 34.1,
  "solar_backflow": 22.6,
  "boiler_backflow": 40.6
}
```
and add the value info with name,
```
call dallassensor cylinder_top
{
  "id": "28-7284-131F-1901",
  "name": "cylinder_top",
  "value": 45.3,
  "type": "number",
  "uom": "°C",
  "writeable": false
}
```
For HA the info should change the object_id to from name to id also matching the topic.
( actual:  "object_id": "temperature_sensor_cylinder_top",)
```
homeassistent/sensor/ems-esp/dallassensor_28-7284-131F-1901:
{
  "dev_cla": "temperature",
  "stat_t": "ems/dallassensor_data",
  "unit_of_meas": "°C",
  "val_tpl": "{{value_json['cylinder_top']}}",
  "object_id": "temperature_sensor_28-7284-131F-1901",
  "name": "cylinder_top",
  "uniq_id": "dallasensor_28-7284-131F-1901",
  "dev": {
    "ids": [
      "ems-esp"
    ]
  }
}
```

same scheme for analogs, the id is there the gpio. 
```
call analogsensor values
{
  "pressure": 1.185,
  "test": 0.01
}
```
```
call analogsensor pressure
{
  "gpio": 36,
  "name": "pressure",
  "type": "number",
  "analog": "adc",
  "uom": "bar",
  "offset": 500,
  "factor": 0.003,
  "value": 1.185,
  "writeable": false
}
```
HA config with object_id containing gpio:
```
homeassistent/sensor/ems-esp/analogsensor_36:
{
  "stat_t": "ems/analogsensor_data",
  "val_tpl": "{{value_json['pressure']}",
  "object_id": "analog_36",
  "name": "pressure",
  "uniq_id": "analogsensor_36",
  "unit_of_meas": "bar",
  "dev": {
    "ids": [
      "ems-esp"
    ]
  }
}
```
For digital_out we should also add a command, now we have `analogsensor/setvalue {"value": true, "id": 38}` for setting gpio 38, we should also register command with name, e.g. extra_pump
`analogsensor/extra_pump {"value": true}`

proddy, 2022-11-28:
tested it and works great in HA

closed 2022-11-28 by proddy