emsesp/EMS-ESP32 #760: Dallas temp is rounded for API response

tp1de, 2022-11-20:
For dallas sensors the "value" in api seems to be rounded to full integer.

**/api/dallassensor:/Dallas 1**
{
"id": "28-FF04-CFC4-1604",
"name": "Dallas 1",
**"value": 24,**
"type": "number",
"min": -55,
"max": 125,
"uom": "°C",
"writeable": false
}

**/api/dallassensor:**
{
"Dallas 1": {
"id": "28-FF04-CFC4-1604",
**"temp": 24.1**
},
"28-FF3B-04C5-1604": {
"id": "28-FF3B-04C5-1604",
"temp": 25.6
}
}

MichaelDvP, 2022-11-20:
No!
In v.3.4.4 the value does not show decimals if decimal is 0.
https://github.com/emsesp/EMS-ESP32/blob/d0ac0b7804d8085451954ee7cf457f12e9271085/src/dallassensor.cpp#L395
In v3.5.0 it is always shown with one decimal. 
https://github.com/emsesp/EMS-ESP32/blob/1f43bb201c3d35c651787a189bc2292cdc7446ac/src/dallassensor.cpp#L398

proddy, 2022-11-20:
I thought the problems was between the values displayed as 24 and 24.1 but I just tested it and it works. I think the temp dropped to 24.0 just as you were testing it. closing.

closed 2022-11-20 by proddy