emsesp/EMS-ESP32 #762: Boiler does not respond to custom value EMS-ESP gateway

MarkRoks, 2022-11-21:
Hello,

I have a Nefit Trendline HRC30 CW5 boiler, with an EMS-ESP gateway with firmware v3.4.4. It is possible to adjust the "pump delay" and the "burner min period" through the EMS-ESP gateway. However, it is suspected that nothing will happen with the adjusted values.

For example, I set the "pump delay" to 10 minutes and the "burner min period" to 12 minutes. However, the pump switches off after a few minutes after the boiler has stopped heating for a short time.

When applying other values, nothing actually changes what the boiler actually does.

Greetings,

[image]
[image]
[image]

proddy, 2022-11-21:
Can you provide more details? Like
 - a trace from the console when you're setting the values so we can see the Boiler's response
 - do the values get reset in the EMS-ESP UI after you set them?

MarkRoks, 2022-11-21:
Hi Prody,

I have a Nefit Easy (Logamatic TC100) thermostat.

Just set the "pump delay" back to 2 minutes. He adjusts this neatly in the overview on the web page.

I have added a log of the system as an export below. Perhaps this will provide more clarity.

Regards,

[EMS-ESP-export.txt](https://github.com/emsesp/EMS-ESP32/files/10054644/EMS-ESP-export.txt)