emsesp/EMS-ESP32 #765: Unknown device on bus [Bosch RT800] [Value for set temperature]

Flothoger, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
The EMS-ESP finds unknown devices on the bus.

**0X38** is the indoor unit RT800 (SW Versions match)
**0x10** I could not figure out what this device might be. (Could this be an EMS enabled pump? I figured it can't be the outdoor sensor, because it does not seem to be connected on the EMS2-Bus)

**Describe the solution you'd like**
It would be greate if the RT800 device could be included in the database. Maybe it is similar to a previous model.

I also played around with changing the set tempeartur (Log_Change_Set_Temp.txt), because there currently is no value to set it using the EMS-ESP. While doing this I noticed that when I changed the (room) set-temperature (using the boiler, not the RT800) from 21 to 21.5 degrees the values 0x02A5 and 0x02B9 change. The decimal value of the data is exactly half of the set temperature. E.g. 2A = 42 -> 42/2 = 21 [Degrees] which matches the set value. Maybe these values could also be implemented for the boiler. But I don't know the difference between 0x02A and 0x02B9.

**Additional context**
My setup consists of an **Bosch GC9800IW** 20 P 23 with an UI800 Controller
**Bosch RT800** indoor Unit with SW-Version: 21.02
[emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10069178/emsesp_info.txt)

**Log for 1 minute:**
[Log_1_Min.txt](https://github.com/emsesp/EMS-ESP32/files/10069196/Log_1_Min.txt)

**Log when changing the set temperature from the boiler:**
[Log_Change_Set_Temp.txt](https://github.com/emsesp/EMS-ESP32/files/10069200/Log_Change_Set_Temp.txt)

MichaelDvP, 2022-11-22:
Wow, complete documentation in first post.
Device 0x10 is the main thermostat, in this case i think called UI800, sending telegrams 0x02A5 for measured values and 0x02B9 for setpoints, seems mainly RC300 compatible. The 0x38 is a room thermostat as remote for hc1, compatible RC100H sending telegram 0x042B with setpoint, measured roomtemp and humidity, RT800
I'll add to database. Please test from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Flothoger, 2022-11-22:
Just uploaded it to the EMS-ESP. Works perfectly and the values seem to be correct!

[(new)emsesp_info.txt](https://github.com/emsesp/EMS-ESP32/files/10069952/new.emsesp_info.txt)

Flothoger, 2022-11-22:
In your dev build the default language of the web-ui is polish. I think normally it was english. Not a big deal since you can change it with two clicks, but something I noticed.

MichaelDvP, 2022-11-22:
Yes, it's a known issue, see #745

MichaelDvP, 2022-11-22:
#754

closed 2022-11-22 by Flothoger